esp-rs/espup #14: update subcomand should check if latest version is installed, if not, installs it.

SergioGasquez, 2022-10-11:
## Motivations
At the moment, `espup update`, does not check if the latest version is updated. It will just uninstall the current toolchain and install the default (hardcoded) version.

SergioGasquez, 2022-10-18:
We should avoid having the [latest version hardcoded](https://github.com/esp-rs/espup/blob/main/src/main.rs#L96) at all. Related with https://github.com/esp-rs/rust-build/issues/125

SergioGasquez, 2022-11-03:
> We use [update-informer](https://crates.io/crates/update-informer) in `espflash` for checking updates, allows you to specify either crates.io or GitHub as the distribution source, pretty nice.

Thanks for sharing it! It looks really useful, although it won't solve this issue as what we need to check is [rust-build/releases](https://github.com/esp-rs/rust-build/releases/tag/v1.64.0.0). Maybe we could use what [you suggested in rust-build#125](https://github.com/esp-rs/rust-build/issues/125#issuecomment-1195853288) but  `jq` is not always present and, we also have to deal with Windows.

We could also use https://github.com/esp-rs/rust-build/releases/latest but, we still need the version as its in the name of the artifacts. E.g [`rust-1.64.0.0-x86_64-apple-darwin.tar.xz`](https://github.com/esp-rs/rust-build/releases/download/v1.64.0.0/rust-1.64.0.0-x86_64-apple-darwin.tar.xz)

jessebraham, 2022-11-03:
Maybe a bit of an odd approach, but if there are releases on GitHub available those same versions should be available on `crates.io`. `update-informer` will give you the version number of the latest version published on `crates.io`, so you could use that in turn to pull artifacts from GitHub with a known version number.

jessebraham, 2022-11-03:
Actually scratch that, you can specify the repo if using GitHub:

```rust
use update_informer::{registry, Check};

let informer = update_informer::new(registry::GitHub, "owner/repo", "0.1.0");
informer.check_version();
```

closed 2022-11-03 by SergioGasquez