esp-rs/espup #70: espup installed, but no xtensa target available

Apitronix, 2022-11-20:
## Bug description

When I run `rustc --print target-list`, no xtensa target is shown. Furthermore, when I try to build this [demo project](https://github.com/ivmarkov/rust-esp32-std-demo) using the command `cargo build`, then it fails saying that no `xtensa-esp32-espidf` target is found.

## To Reproduce

1. `espup` tool installed using the command `cargo install espup`,
2. `espup` tool is stored in ".cargo/bin" directory but not in "PATH" env var by default,
3. `espup install` command used,
4. `source export-esp.sh` done.

## Expected behavior

Being able to compile this code, and flash it on ESP32-WROOM32 chip.

## Environment

- OS: archinux
- espup version: 0.2.3

SergioGasquez, 2022-11-21:
Hi @Apitronix! It looks like you are not using the proper channel.  Are you overriding the `esp` toolchain? You should probably use `rustc +esp --print target-list` or `cargo +esp build`.  See https://rust-lang.github.io/rustup/overrides.html#the-toolchain-file for more information on toolchain overriding.

SergioGasquez, 2022-11-24:
Hi @Apitronix, do you have any update?

closed 2022-12-23 by SergioGasquez

willlma, 2023-11-10:
Hi @SergioGasquez, I'm running into the same issue. When I run `rustup toolchain list`, the output is
```
stable-aarch64-apple-darwin (default)
nightly-aarch64-apple-darwin
esp (override)
```
which implies that esp is the chosen toolchain. When I try to run `rustup target list --installed`, I get
```
error: toolchain 'esp' does not support components: esp is a custom toolchain
```
When I try `cargo +esp build`, I get
```
error: no such command: `+esp`

	Cargo does not handle `+toolchain` directives.
	Did you mean to invoke `cargo` through `rustup` instead?
```
(I don't know what invoking cargo _through_ rustup means)
and when I try `rustc +esp --print target-list` the error is
```
error: toolchain 'esp' does not support components: esp is a custom toolchain
```

I'm a bit stuck. I've followed the instructions from the README and the Rust on ESP book pretty closely. I'm trying to set up a project using the `esp-rs/esp-idf-template` incuding the standard library, and one thing that confuses me is that the template's cargo.toml refers to the target as `xtensa-esp32-espidf` whereas espup installs `xtensa-esp-elf`.

Thanks for any help

SergioGasquez, 2023-11-10:
How did you install Rust? Did you use any package manager? https://esp-rs.github.io/book/installation/rust.html

> and one thing that confuses me is that the template's cargo.toml refers to the target as xtensa-esp32-espidf whereas espup installs xtensa-esp-elf

There are two different things,`xtensa-esp32-espidf` is the Rust target triple while `xtensa-esp-elf` is the GCC toolchain used for linking the project

willlma, 2023-11-12:
Thanks for the tip. I installed Rust a few months ago so didn't remember, but I see it in `brew list`, so that must be how I installed it. Will reinstall.