esp-rs/espup #80: `espup install` expects `/bin/bash` to exist

frantisekhanzlikbl, 2022-12-05:
## Bug description

`espup install` fetches `rustup-init.sh` and attempts to execute it with `"/bin/bash" "/home/$USER/.espressif/dist/rustup/rustup-init.sh"` However, not all systems have Bash installed, and even those that do don't always have it present in `/bin/bash`

- Would you like to work on a fix? I could, if provided with info about the desired fix, although I suspect that it's going to be such a simple fix that it would be much more time-effective if some maintainer fixes it instead. 

## To Reproduce

Steps to reproduce the behavior:

1. run `espup install`

## Expected behavior

Things work

## Screenshots

```
Error:
  × command '"/bin/bash" "/home/$USER/.espressif/dist/rustup/rustup-init.sh" "--default-toolchain" "nightly" "--default-host" "x86_64-unknown-linux-gnu" "--profile" "minimal" "-y"' failed to start
  ╰─▶ No such file or directory (os error 2)
```

## Environment

- OS: NixOS 23.05 (Stoat)
- espup version: 0.2.3

## Additional context

Since `rustup-init.sh` has a shebang, I don't see any reason not to invoke it directly. If there is one, it would at least be better to use a bash from `$PATH`, for example via `/usr/bin/env bash ...`.

SergioGasquez, 2022-12-16:
Hi! 

Sorry for taking so long to reply. I've just had a look at the issue, but for some reason, the `cmd!` macro does not work without a bash path. I've improved the current state a bit: https://github.com/esp-rs/espup/compare/main...fix/bash-path
But still not ideal, I've also seen that you did some testing, did you have more success than me?

joshuachp, 2023-01-23:
Hi, I manage to run espup on nix with the following nix package:

<details>

```nix
{ pkgs
, lib
, rustPlatform
, fetchCrate
}:

rustPlatform.buildRustPackage rec {
  pname = "espup";
  version = "0.2.6";
  patches = [ ./bash.patch ];

  src = fetchCrate {
    inherit pname version;
    sha256 = "sha256-+G/0Eo1/IULoa3EQmn2CYUzqyQ2iXnFtyUBfzwJNMp4=";
  };

  cargoSha256 = "sha256-lmyeK3Wj/q1uNw4iROuaHkjfBxvXtktK3e3y5GytbXg=";

  nativeBuildInputs = with pkgs;[
    pkg-config
    perl
  ];
  buildInputs = with pkgs; [
    openssl
  ];

  # A test will fail since will try to make a http request
  doCheck = false;

  meta = with lib; {
    description = "Tool for installing and maintaining ESP Rust ecosystem.";
    homepage = "https://github.com/esp-rs/espup";
    license = with licenses; [ asl20 mit ];
  };
}
```

</details>

The patch needed was the following:  [bash.patch](https://github.com/esp-rs/espup/files/10480346/bash.patch)

Hope this helps, it would be great to make it work on NixOS.

SergioGasquez, 2023-01-24:
Just created #161 with @joshuachp patch (thanks for sharing it!), I did some testing and everything was working for me. Please, @frantisekhanzlikbl can you confirm that this fixes your issue?
```
cargo install espup --git https://github.com/SergioGasquez/espup --branch fix/bash
espup install
```

frantisekhanzlikbl, 2023-01-24:
Hey, so sorry for taking forever to reply. I was using the ESP for a short-lived school project, so I haven't had the chance to try it again yet. I unfortunately can't test the fix ATM, but it looks exactly like what I've done locally to make it work.

closed 2023-01-25 by SergioGasquez