esp-rs/espup #85: error: unable to unlink old fallback exe: Access is denied. (os error 5)

dimpolo, 2022-12-09:
Explanation see: https://github.com/rust-lang/rustup/issues/2889

Could esp-up perform the mentioned workaround automatically?
https://github.com/rust-lang/rustup/issues/2889#issuecomment-986216050

SergioGasquez, 2023-01-16:
Hi @dimpolo!

Sorry for taking so long, this flew under my radar. I just created a [small PR with the proposed workaround](https://github.com/esp-rs/espup/pull/145), could you verify that the error disappears whit the fix?

dimpolo, 2023-01-23:
All good!

I can verify that copying `~/.rustup/toolchains/nightly-x86_64-pc-windows-msvc/bin/cargo.exe` into the esp toolchain solves the problem for me.

I'm afraid I don't know enough about rustup or other peoples setups to asses if the proposed PR correctly locates the required `cargo.exe`

SergioGasquez, 2023-01-23:
> All good!
> 
> I can verify that copying `~/.rustup/toolchains/nightly-x86_64-pc-windows-msvc/bin/cargo.exe` into the esp toolchain solves the problem for me.
> 
> I'm afraid I don't know enough about rustup or other peoples setups to asses if the proposed PR correctly locates the required `cargo.exe`

Thanks for the confirmation! Probably, we should do this "fix" in [rust-build](https://github.com/esp-rs/rust-build) and ship the Windows artifacts with the `cargo.exe` in the `esp` toolchain folder.

SergioGasquez, 2023-01-26:
rust-build issue is now closed and merged: https://github.com/esp-rs/rust-build/issues/157 so, from 1.67.0.0 is properly fixed for all installation methods

closed 2023-01-26 by SergioGasquez