esp-rs/espup #98: Add 啊

RedElectricity, 2022-12-18:
## Motivations

- Would you like to implement this feature? [y/n]

## Solution

## Alternatives

## Additional context

SergioGasquez, 2022-12-19:
For reference, we could use:
- https://gitee.com/espressif_systems as mirror of esp-idf and other github dependencies
- And we could upload release binaries to AWS like we do in: https://github.com/espressif/idf-installer/blob/main/.github/workflows/update-index.yml

RedElectricity, 2022-12-20:
🤔maybe `rust-build` need a mirror too

```
espup install --toolchain-version 1.66.0.0 -d x86_64-pc-windows-gnu --esp-idf-version master
[2022-12-20T06:11:23Z INFO ] 💽  Installing esp-rs
[2022-12-20T06:11:23Z INFO ] 🔧  Checking existing Rust installation
[2022-12-20T06:11:24Z INFO ] 🔧  Installing Xtensa Rust 1.66.0.0 toolchain
[2022-12-20T06:11:24Z INFO ] 📥  Downloading file rust.zip from https://github.com/esp-rs/rust-build/releases/download/v1.66.0.0/rust-1.66.0.0-x86_64-pc-windows-gnu.zip
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: InvalidArchive("Invalid zip header")', C:\Users\redel\scoop\persist\rustup\.cargo\registry\src\mirrors.tuna.tsinghua.edu.cn-df7c3c540f42cdbd\espup-0.2.4\src\toolchain\mod.rs:59:65
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

SergioGasquez, 2022-12-20:
We could publish the toolchain release artifacts to AWS.

SergioGasquez, 2023-03-03:
Just a small update on this, on [yesterday's community meeting](https://github.com/esp-rs/rust/discussions/163#discussioncomment-5177365) we discussed this issue. We will be probably creating a mirror of our Xtensa Rust and LLVM artifacts, probably in AWS, but we will wait until `no_std` does not require GCC as a linker so we don't have to create a mirror for it too.

SergioGasquez, 2023-05-18:
Just noticed that @chang196700 is using https://ghproxy.com/. I dont have any experience with it, but if it works, maybe we could just include a `--proxy` argument that when used we download it from ghproxy instead.

wyswill, 2024-08-14:
I modify this source file , path `src/toolchain/mod.rs:73`
```rust
   let client = reqwest::Client::builder()
        .proxy(reqwest::Proxy::https("http://127.0.0.1:20171")?)
        .build()?;

    let resp = client.get(&url).send().await?;
    let bytes = resp.bytes().await?;

```