esp-rs/espup #121: Improve description of arguments and errors when using them incorrectly

LeCyberDucky, 2022-12-29:
## Bug description

I think the options described when doing `espup install -h` could be improved a bit: 

`-e`: 
This description ends abruptly after `ESP-IDF version to install. If empty, no esp-idf is installed. Version format:` without actually specifying the possible formats. They are listed [here](https://github.com/esp-rs/espup#install-subcommand), but they don't show up when running said command in a terminal. Also, the part `If empty, no esp-idf is installed.` made me think that I *need* to choose a version, in order to be able to use `std`. Perhaps the [documentation](https://esp-rs.github.io/book/installation/installation.html) could specify explicitly that this option can be left empty, and that [esp-idf-sys](https://github.com/esp-rs/esp-idf-sys) will automatically take care of things (without having to install it in some other step). After reading this documentations a few times, l understand this now, but perhaps it could be made a bit more clear. 

`-m`:
A few more details describing what the minimal installation entails would be nice. 

`-f`: 
Perhaps this is just me not interpreting things correctly, but the description `Destination of the generated export file` made me think that I could specify the directory, where the file should be placed. Turns out, that you need to specify the full path of the file, however. 

This brings me to my next point, though. When specifying an invalid argument for `-f`, I would like a better error message. Currently, when specifying a directory instead of a file path, the program just fails with an `Access denied. (os error 5)`. It would be nice if we could indicate that the specified path does not make for a valid file. 

Further, it may be good to check the validity of the arguments earlier. Currently, the above mistake results in a partial installation and requires manual deletion of the installed toolchain before the installation can be restarted with a correct argument. This could be circumvented by validating the arguments before starting the installation. 

### Would you like to work on a fix? [y/n]
I don't have a lot of time to do spend on this right now, but I am willing to take a look, if nobody else wants to do it :)

## To Reproduce

Steps to reproduce the behavior:

- run `espup install -h`. 
- run `espup install -f <DIR>` where `<DIR>` is the path to a directory. 

(Tested on the main branch)

## Environment

- OS: Windows 10 21H2
- espup version: 0.2.5-dev

SergioGasquez, 2022-12-29:
Thanks for opening an issue with so many details! I will be working on improving all the mentioned issues

LeCyberDucky, 2022-12-29:
You are very welcome! Thanks for the quick response, and for maintaining this project. It makes for a very pleasant experience when setting up the environment for ESP-development :)

SergioGasquez, 2022-12-29:
A small comment regarding: 

> This description ends abruptly after ESP-IDF version to install. If empty, no esp-idf is installed. Version format: without actually specifying the possible formats. They are listed [...]

`-h` is not the same as `--help`, `-h` only shows a summary, that's why it was only showing the first line. This is still an issue as it ended abruptly and it made no sense.

LeCyberDucky, 2022-12-29:
Oh, I wasn't aware of that. Thanks for pointing it out.

SergioGasquez, 2022-12-29:
Just created #122 fixing all the issues you commented. Can you review them @LeCyberDucky?

LeCyberDucky, 2022-12-29:
Fantastic! I read through the changes, and they look great. Thanks for the super quick update!

closed 2022-12-30 by SergioGasquez