esp-rs/espup #123: [Windows] Allow for using CMD instead of PowerShell

LeCyberDucky, 2022-12-29:
## Motivations

The export file generated on Windows is a PowerShell script. Using this script to set the environment variables requires working in PowerShell. I.e., it is not enough to work in a normal CMD window and run `powershell export-esp.ps1`, since this will not modify the environment variables in the CMD session. See: https://stackoverflow.com/a/68024909

It would be nice to have a something like a batch file, such that the environment variables can also be set in a CMD session, instead of forcing the use of PowerShell. 

- Would you like to implement this feature? [y/n]
I am not sure how to solve this. 

## Solution

I am not quite sure what the best approach would be here. Generating both versions by default, or maybe letting the user choose? 

## Alternatives

## Additional context

SergioGasquez, 2022-12-30:
Hello! Thanks for opening an enhancement request, I am not really sure how to address it, to be honest, I don't really like the idea of adding an argument to choose between the two, and the same goes for generating both cmd and ps1 export files, as most of the user will only use one of them.

Could you try running:
```
PowerShell -File <pathToExport>\export-file.ps1.
```
And see if it works?

LeCyberDucky, 2022-12-30:
Running 

`PowerShell -File <pathToExport>\export-file.ps1.`

gives the same result. It does not change the PATH variable in my CMD window. 

Yes, I'm not sure what an ideal solution would be either. I fully agree that people will likely only use one or the other. Personally, I never use PowerShell. I assume that this is true for many people, but I have no idea what the ratio between CMD and PowerShell users is. Either way, I don't think it's ideal to force a specific shell on the users, though.

SergioGasquez, 2023-02-15:
Hi!

Sorry for taking that long to reply, we were discussing how to approach `espup` future, here is the discussion #154. My idea was to solve this issue with #152 and created #166. But since we are going for a “minimal” approach, I've added Windows environment injection into #182.  Feel free to test it, but this show be a solution as you no longer need to source any file.

LeCyberDucky, 2023-02-15:
Thanks for getting back to me. I currently can't test it, but it's great to see that you have found a solution :)

closed 2023-02-15 by LeCyberDucky