esp-rs/espup #136: Request or response body error: error reading a body from connection

Nukesor, 2023-01-12:
## Bug description

A job in our CI ran into this error.
They didn't occur in other runs, though.

```
thread 'tokio-runtime-worker' panicked at 'called `Result::unwrap()` on an `Err` value: SendError(Ok(["export IDF_PATH=/builds/orga/project/project/firmware/espressif/esp-idf-ae062fbba3ded0aa/v4.4.3", "export PATH=\"/builds/orga/project/project/firmware/espressif/tools/cmake/3.23.1/bin:/builds/orga/project/project/firmware/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/builds/orga/project/project/firmware/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin:/builds/orga/project/project/firmware/espressif/tools/ninja/1.10.2/:/builds/orga/project/project/firmware/espressif/python_env/idf4.4_py3.10_env/bin:/builds/orga/project/project/firmware/espressif/esp-idf-ae062fbba3ded0aa/v4.4.3/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/builds/orga/project/project/firmware/cargo/bin\"", "export IDF_TOOLS_PATH=\"/builds/orga/project/project/firmware/espressif\""]))', /builds/orga/project/project/firmware/cargo/registry/src/github.com-1ecc6299db9ec823/espup-0.2.5/src/main.rs:248:32
Error:
  × request or response body error: error reading a body from connection: end
  │ of file before message length reached
  ├─▶ error reading a body from connection: end of file before message length
  │   reached
  ╰─▶ end of file before message length reached
```

The bug happened at this piece of code:

```rs
    for app in to_install {
        let tx = tx.clone();
        tokio::spawn(async move {
            let res = app.install().await;
            tx.send(res).await.unwrap(); // <---- At the unwrap
        });
    }
```
The error message indicates, that the installation of the app worked as expected. Instead the `tx.send(res)` seems to somehow fail.

- Would you like to work on a fix? [y]
  I just need some context, as I'm not sure how this could happen.

## To Reproduce

I wasn't able to reproduce this behavior, it only happened once until now.
It happened on the very last version, which is the latest commit on `main` at this point.

## Environment

- OS: Arch Linux
- espup version: v0.2.5

SergioGasquez, 2023-01-12:
Hi @Nukesor!

I haven't seen this error before and I am not sure why it panics when it's sending and `Ok` with the correct exports paths, but since it's async stuff it might be worth trying with the fix of #137. Could you update your CI, maybe on a fork, to use `SergioGasquez/xtensa-toolchain@main` and see if the error still appears?

My fork of `xtensa-toolchain` is using and release artefact that includes the changes of #137.

Nukesor, 2023-01-13:
As mentioned above, I wasn't able to reproduce this issue until now :/ 

Though, I assume that your changes in #137 might indeed fix this, as it's probably related to async stuff.
I'll close this issue and re-open it in case I see this issue again. Maybe I manage to gather a bit more context next time as well :)

closed 2023-01-13 by Nukesor