esp-rs/espup #156: Inject the export-esp.sh into the .*rc files for linux.

Frostie314159, 2023-01-22:
## Motivations

Espup prompts you to execute the `export-esp.sh` script on every terminal launch. Since this is obviously tedious, I would suggest injecting the execution of the script into the various `.*rc` files(i.E. `.bashrc`).

- Would you like to implement this feature? y

## Solution

Described in Motivations

## Alternatives

Don't.

## Additional context

igrr, 2023-01-22:
Please note that at least for the ESP-IDF specific exports, we generally recommend against sourcing the export script from the profile script. This is because IDF uses a Python virtual environment, and versions of packages in that environment may be in conflict with versions of packages which other software in your system requires.

SergioGasquez, 2023-01-23:
I don't think we should inject variables by default in Unix systems as it could break the user environment, also there are a lot of different distributions/shells...

In Windows (https://github.com/esp-rs/espup/issues/152), this was suggested by @georgik, and I think it's different as the injection of environment variables is, somewhat, a common thing that installers do.

MathyV, 2023-02-06:
Please don't do this on Linux environments. If you want to do this permanently you can add it to your own environment but don't force it on everyone.

closed 2023-02-06 by SergioGasquez