esp-rs/espup #164: export-esp.sh does not set CLANG_PATH which can confuse bindgen (and any crate using clang-sys)

jasta, 2023-01-24:
## Bug description

export-esp.sh is missing the CLANG_PATH variable which is used by clang-sys to locate the clang runtime.  bindgen uses this to determine -isystem.  Without it, the host's include path will be used while the target's clang toolchain is used.

- Would you like to work on a fix? Yes

## To Reproduce

Steps to reproduce the behavior:

1. . ~/export-esp.sh
2. export TARGET_CC=riscv32-esp-elf-gcc
3. export TARGET_CXX=riscv32-esp-elf-g++
4. export TARGET_AR=riscv32-esp-elf-ar
5. cd <some-esp32-project>
6. cargo add lvgl --git https://github.com/jasta/lvgl-rs --branch esp32-fixup
7. cargo add lvgl-sys --git https://github.com/jasta/lvgl-rs --branch esp32-fixup
8. cargo add lvlg-codegen --git https://github.com/jasta/lvgl-rs --branch esp32-fixup
9. Download https://github.com/rafaelcaricio/lvgl-rs/blob/master/lvgl-sys/vendor/include/lv_conf.h and store in CWD.
10. DEP_LV_CONFIG_PATH=$(pwd) cargo +nightly build --target riscv32imc-esp-espidf

Lots of build errors related to stdint.h types missing (like int16_t).  Similar problems are likely to occur with other crates utilizing bindgen, but not always consistently.  It may also repro differently depending on the host and target clang mismatches for the user testing it.

## Expected behavior

Should compile :)

## Screenshots

N/A

## Environment

- OS: Ubuntu 22.04
- espup version: 0.2.5

## Additional context

Part of a larger body of work to fix cc and bindgen (a couple other issues block general purpose use cases)

jasta, 2023-01-24:
Not sure how exactly to make this work on Windows given: https://github.com/esp-rs/espup/blob/b9aa3e266c132f68f433db401d32bf2a0a358cd8/src/toolchain/llvm.rs#L50 ?  Kinda strange to me that they would be different like this?

SergioGasquez, 2023-01-26:
> Not sure how exactly to make this work on Windows given: https://github.com/esp-rs/espup/blob/b9aa3e266c132f68f433db401d32bf2a0a358cd8/src/toolchain/llvm.rs#L48-L54 ?  Kinda strange to me that they would be different like this?

It's a strange requirement for windows: https://rust-lang.github.io/rust-bindgen/requirements.html#windows

georgik, 2023-01-26:
@SergioGasquez  In case of Windows, it's necessary to set path to DLL which has slightly different location than on *nixes https://github.com/esp-rs/rust-build/blob/main/Install-RustToolchain.ps1#L48

jasta, 2023-01-27:
I think I figured this out, the lib path being the bin path for Windows was a red herring.  That said, I ran into another snag which is that if the minified option is used for espup then only the clang libraries (not binaries) will be installed. 
 This will break cross compilation because bindgen indirectly invokes the clang binary here:

https://github.com/rust-lang/rust-bindgen/blob/a7ff8e136027c9542f65eca97a7ff3c0b870518d/bindgen/lib.rs#L2549

That goes on to apply a set of heuristics looking in CLANG_PATH, then generally anywhere in PATH, for a binary it can run.  Once it finds one, it does things like `clang -E -x c - -v -target whatever ...` to figure out what the include paths and other things should be.

I think I'll just conditionally the output CLANG_PATH.  If you ask for the minified installation, things just won't work out of the box quite like you expect :\

PR #171 (2023-02-01): Export CLANG_PATH so bindgen works with cross compilation
Fixes #164

CLANG_PATH is relevant for bindgen to be able to find required headers and library files.  Without exporting this along with LIBCLANG_PATH, you'll get a mismatch using bindgen and if the host clang doesn't support full cross compilation of C code (which is common for Ubuntu in my case, at least), then ultimately bindgen will not be able to compile the code its trying to generate bindings for.  The user will see weird errors like basic types int16_t and such are missing or common header files like string.h aren't available.
commit 78ee9473 (2023-02-01): Export CLANG_PATH so bindgen works with cross compilation

Fixes #164

CLANG_PATH is relevant for bindgen to be able to find required headers
and library files.  Without exporting this along with LIBCLANG_PATH,
you'll get a mismatch using bindgen and if the host clang doesn't
support full cross compilation of C code (which is common for Ubuntu in
my case, at least), then ultimately bindgen will not be able to compile
the code its trying to generate bindings for.  The user will see weird
errors like basic types int16_t and such are missing or common header
files like string.h aren't available.
closed 2023-02-01 by SergioGasquez