esp-rs/espup #165: How to provide TARGET_CC/CXX/AR when cross compiling?

jasta, 2023-01-24:
## Bug description

Trying to build any crate that utilizes the cc crate is likely to fail because TARGET_CC/CXX/AR (maybe others) are not set.  The toolchain will be defaulted to the host cc which isn't expected to work and gives errors about unknown arguments.

```
  running: "cc" "-Os" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-march=rv32imc" "-mabi=ilp32" "-mcmodel=medany" "-I" "/media/jasta/data/software/lvgl-rs/lvgl-sys/vendor/lvgl/src" "-I" "/media/jasta/data/software/lvgl-rs/lvgl-sys/vendor" "-I" "/home/jasta/software/cc-broken" "-DLV_CONF_INCLUDE_SIMPLE=1" "-o" "/media/jasta/data/software/cc-broken/target/riscv32imc-esp-espidf/debug/build/lvgl-sys-2a3b0bc869059431/out/8142b73eb685051c-lv_indev.o" "-c" "/media/jasta/data/software/lvgl-rs/lvgl-sys/vendor/lvgl/src/lv_core/lv_indev.c"
  cargo:warning=cc: error: unrecognized argument in option ‘-mabi=ilp32’
  cargo:warning=cc: note: valid arguments to ‘-mabi=’ are: ms sysv
  cargo:warning=cc: error: unrecognized argument in option ‘-mcmodel=medany’
  cargo:warning=cc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
  exit status: 1
```

- Would you like to work on a fix? Yes, but I don't know how :)

## To Reproduce

Steps to reproduce the behavior:

1. See: https://github.com/esp-rs/rust-build/issues/61#issue-1213504417

## Expected behavior

Unsure exactly, I guess I expect cc and other crates to be able to automatically figure out which toolchain to use somehow, though I don't really know how that's supposed to work normally.

## Screenshots

N/A

## Environment

- OS: Ubuntu 22.04
- espup version: 0.2.5

## Additional context

I think this may be a broader problem in the Rust cross compilation ecosystem.  Perhaps best we can do is add some notes to the Rust ESP book or troubleshooting section that talk about what to do.

jasta, 2023-01-24:
Digging a little deeper, I think cc is confused because we're using ldproxy as the RUSTC_LINKER:

https://github.com/rust-lang/cc-rs/blob/d69bc7db4493685508a1fecab8b98c5de811bd45/src/lib.rs#L2868

It could also be that we're somehow supposed to make use of the CROSS_COMPILE env variable and somehow modify that big list of targets.  I'm really not sure.

jasta, 2023-01-24:
To my great surprise, this is fixed by modifying `.cargo/config` in my esp32 project:

jasta, 2023-01-24:
Mistakenly closed, sorry :)

jasta, 2023-01-25:
I discovered that CROSS_COMPILE does indeed work.  You need to set CROSS_COMPILE="riscv32-esp-elf" for example to convince cc that it should use that toolchain.  Unfortunately what we really need is for `.cargo/config` to support specifying environment variables on a per target basis.

jasta, 2023-01-25:
Exploring an option to patch `cc-rs` to make this automatic.

jasta, 2023-01-25:
This can be done with https://github.com/rust-lang/cc-rs/pull/776 but I think only for riscv32imc-esp-espidf.  The xtensa targets aren't Tier 3 and might not be accepted yet.

jasta, 2023-01-27:
Closing for now as the cc-rs change has been merged for riscv32imc-esp-espidf.  We can revisit adding the xtensa targets later.  For now, xtensa targets can work around this with:

```
CROSS_COMPILE=xtensa-esp32-elf cargo build
```

closed 2023-01-27 by jasta

pwfff, 2023-06-06:
adding a comment here just to say I ran into this and the above solution worked (also the solution I managed to arrive at, which was `CC=xtensa-esp32s3-elf-cc cargo build`).

adding the specific error I got here for google SEO:
```
547165c3df62f629.rlib: error adding symbols: file format not recognized
          collect2: error: ld returned 1 exit status
```