esp-rs/espup #170: `espup update` removed other installed toolchains

dimpolo, 2023-01-26:
## Environment
- OS: `Windows 10`
- espup version: `espup 0.2.3`
- installed toolchain: `Toolchain '1.67.0.0'`

SergioGasquez, 2023-01-31:
Can you try if the issue still happens with an updated version of `espup` (`0.2.6` for example)?

SergioGasquez, 2023-02-15:
Hi @dimpolo! I was not able to reproduce this issue. Try with a recent release (`0.2.9`) and see if it reproduces.

Also, #182 will change a lot of stuffs, so probably this will be fixed if it wasn't already since `0.2.3` is pretty old (Nov 17 2022).

Since there are not a lot of details and I was not able to reproduce it, I will be closing the issue. Feel free to reopen it if the issue still persists for you (with recent version).

closed 2023-02-15 by SergioGasquez

dimpolo, 2023-02-15:
Ah, sorry I didn't get back to you sooner.
I currently don't have time to investigate this, so closing makes sense!