esp-rs/espup #178: Rust / espup doesn't do shell expansion on Linux

MathyV, 2023-02-06:
## Bug description

Took me quite a bit to figure this one out: if you have `~/.cargo/bin` in your PATH variable instead of the full path, the tilde doesn't get expanded by `std::Process::Command` and thus espup can't execute `rustup` and reinstalls it every time. I now fixed my PATH definition to use the full path instead of the ~ but it might still be interesting to implement this expansion, or at least mention it somewhere in the documentation. I realize this might not be the best place to tackle the issue but I thought I'd hopefully save the next guy some time, I won't be offended if you just close it ;-).

- Would you like to work on a fix? [y/n] maybe

I already did some work trying to debug this, looking at the code I think the `which` crate might be interesting to abstract the Win32 / Linux / MacOs compatibility. I used it to try and get the path for the binary directly but still, that crate also doesn't do variable expansion so this actually might need to move up the chain then.

## To Reproduce

Steps to reproduce the behavior:

1. Use `~/.cargo/bin` in your PATH
2. Cry as `espup install` removes your custom installation of `rustup` each time

## Expected behavior

Not installing `rustup` if it is already there

## Environment

- OS: Gentoo
- espup version: 0.2.8

SergioGasquez, 2023-03-01:
Hello, I will be closing this issue as it should be fixed with `espup v0.3.0`. Feel free to reopen it if the issue still persists.

closed 2023-03-01 by SergioGasquez

MathyV, 2023-03-01:
Sorry, I missed this question but I already reported on Matrix that everything was working. You probably didn't make the link between that internet stranger and this internet stranger ;-)