esp-rs/espup #185: espup installation fails on Ubuntu Server 22.04.1, due to issue with openssl-sys

pilotniq, 2023-02-14:
## Bug description

I have a newly installed Ubuntu Server 22.04.1. When I run `cargo install espup`, I get a bunch of errors:

```
    Updating crates.io index
  Installing espup v0.2.8
  Downloaded fastrand v1.9.0
  Downloaded 1 crate (11.9 KB) in 0.51s
   Compiling libc v0.2.139
   Compiling cfg-if v1.0.0
   Compiling proc-macro2 v1.0.51
   Compiling quote v1.0.23
   Compiling unicode-ident v1.0.6
   Compiling syn v1.0.107
   Compiling jobserver v0.1.25
   Compiling cc v1.0.79
   Compiling autocfg v1.1.0
   Compiling version_check v0.9.4
   Compiling memchr v2.5.0
   Compiling pkg-config v0.3.26
   Compiling log v0.4.17
   Compiling once_cell v1.17.0
   Compiling serde_derive v1.0.152
   Compiling typenum v1.16.0
   Compiling generic-array v0.14.6
   Compiling serde v1.0.152
   Compiling lock_api v0.4.9
   Compiling itoa v1.0.5
   Compiling pin-project-lite v0.2.9
   Compiling parking_lot_core v0.9.7
   Compiling openssl-src v111.25.0+1.1.1t
   Compiling scopeguard v1.1.0
   Compiling smallvec v1.10.0
   Compiling bytes v1.4.0
   Compiling bitflags v1.3.2
   Compiling openssl-sys v0.9.80
   Compiling ring v0.16.20
   Compiling tokio v1.25.0
   Compiling futures-core v0.3.26
   Compiling parking_lot v0.12.1
   Compiling mio v0.8.5
error: failed to run custom build command for `openssl-sys v0.9.80`

Caused by:
  process didn't exit successfully: `/tmp/cargo-installYYc1GF/release/build/openssl-sys-e3c20ba97bf7e10c/build-script-main` (exit status: 101)
  --- stdout
  cargo:rustc-cfg=const_fn
  cargo:rustc-cfg=openssl
  cargo:rerun-if-env-changed=X86_64_UNKNOWN_LINUX_GNU_OPENSSL_NO_VENDOR
  X86_64_UNKNOWN_LINUX_GNU_OPENSSL_NO_VENDOR unset
  cargo:rerun-if-env-changed=OPENSSL_NO_VENDOR
  OPENSSL_NO_VENDOR unset
  cargo:rerun-if-env-changed=CC_x86_64-unknown-linux-gnu
  CC_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CC_x86_64_unknown_linux_gnu
  CC_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CC
  HOST_CC = None
  cargo:rerun-if-env-changed=CC
  CC = None
  cargo:rerun-if-env-changed=CFLAGS_x86_64-unknown-linux-gnu
  CFLAGS_x86_64-unknown-linux-gnu = None
  cargo:rerun-if-env-changed=CFLAGS_x86_64_unknown_linux_gnu
  CFLAGS_x86_64_unknown_linux_gnu = None
  cargo:rerun-if-env-changed=HOST_CFLAGS
  HOST_CFLAGS = None
  cargo:rerun-if-env-changed=CFLAGS
  CFLAGS = None
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("false")
  CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
  running "perl" "./Configure" "--prefix=/tmp/cargo-installYYc1GF/release/build/openssl-sys-f0f596d9745bc047/out/openssl-build/install" "--openssldir=/usr/local/ssl" "no-dso" "no-shared" "no-ssl3" "no-unit-test" "no-comp" "no-zlib" "no-zlib-dynamic" "no-md2" "no-rc5" "no-weak-ssl-ciphers" "no-camellia" "no-idea" "no-seed" "linux-x86_64" "-O2" "-ffunction-sections" "-fdata-sections" "-fPIC" "-m64"
  Configuring OpenSSL version 1.1.1t (0x1010114fL) for linux-x86_64
  Using os-specific seed configuration
  Creating configdata.pm
  Creating Makefile

  **********************************************************************
  ***                                                                ***
  ***   OpenSSL has been successfully configured                     ***
  ***                                                                ***
  ***   If you encounter a problem while building, please open an    ***
  ***   issue on GitHub <https://github.com/openssl/openssl/issues>  ***
  ***   and include the output from the following command:           ***
  ***                                                                ***
  ***       perl configdata.pm --dump                                ***
  ***                                                                ***
  ***   (If you are new to OpenSSL, you might want to consult the    ***
  ***   'Troubleshooting' section in the INSTALL file first)         ***
  ***                                                                ***
  **********************************************************************
  running "make" "depend"

  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', /home/erl/.cargo/registry/src/github.com-1ecc6299db9ec823/openssl-src-111.25.0+1.1.1t/src/lib.rs:497:39
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: failed to compile `espup v0.2.8`, intermediate artifacts can be found at `/tmp/cargo-installYYc1GF`
```

- Would you like to work on a fix? [y/n]

## To Reproduce

Steps to reproduce the behavior:

On a fresh Ubuntu Server 22.0.4.1 machine, , then 
1. install rustup: `curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh (select default installation)
2. source "$HOME/.cargo/env"
3. sudo apt-get install clang
4. cargo install ldproxy 
5. cargo install espup

## Expected behavior

Installation succeeds

## Environment

- OS: Ubuntu Server 22.04.1
- espup version: 0.2.8
- uname -a: Linux noctilux 5.15.0-60-generic #66-Ubuntu SMP Fri Jan 20 14:29:49 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

## Additional context

I tried installing libssl-dev, did not seem to help.

I tried creating a cargo/Rust project and adding openssl-sys 0.9.80 as a dependency, and that build without a problem.

SergioGasquez, 2023-02-14:
Hi!

Thanks for reporting the issue! Do you mind trying a few things:
- Install `pkg-config`: `apt-get install pkg-config`
- See if you can use a [release binary](https://github.com/esp-rs/espup/releases/tag/v0.2.8)
- Check if the same happens on main branch: `cargo install espup --git https://github.com/esp-rs/espup`
- See other possible solutions at https://github.com/sfackler/rust-openssl/issues/1021

SergioGasquez, 2023-02-22:
Hi @pilotniq, do you have any updates on this issue? Also, we've just release `0.3.0` which includes a lot of changes, please try with that version alongside the other suggestions that I left the last week.

pilotniq, 2023-02-22:
Thanks for the replies! 
I solved it, what finally fixed it was `apt-get install build-essentials`.

closed 2023-02-22 by pilotniq

HeroAAX, 2024-04-07:
For any other user, that run into the same problem: the package needed is "build-essential" (without the s):

`sudo apt-get install build-essential`