esp-rs/espup #191: ESP8266 support

Tastaturtaste, 2023-02-23:
## Motivations

I want to write and compile Rust code for my nodemcu esp8266 dev board and install the necessary toolchain using the espup tool.

## Solution

Add a target for the esp8266.

## Additional context
The [esp-rs/rust](https://github.com/esp-rs/rust) seemingly already supports the esp8266 according to its Readme, but the espup tool does not list the esp8266 as a possible target for `espup install -t <target>`. The closest thing would be the esp32 target, but I couldn't find documentation anywhere if this target also installs support for the esp8266. 

Since the tool already installs the patched compiler, and the compiler should support the esp8266 already, I would expect to be able to install the esp8266 toolchain with the espup tool.

SergioGasquez, 2023-02-24:
Hello @Tastaturtaste!

Support for Rust in ESP8266 is limited, and I don't think we should add support for it in `espup`. I would recommend you to switch to any of the ESP32s. The [`esp8266-hal`](https://github.com/esp-rs/esp8266-hal) is in maintenance mode and we don't actively maintain it. Feel free to try with the instructions used in the repo: https://github.com/esp-rs/esp8266-hal#using-the-installation-scripts

Sorry, but I will be closing this issue as I don't want to add and maintain supporting this feature that we not fully support.

closed 2023-02-24 by SergioGasquez

Tastaturtaste, 2023-02-24:
Hmm, I am a little sad to hear that. Thanks though for the honest response. 

Can you tell me if the patched compiler has full support for the ESP8266? I do want to switch to a ESP32-C3 in the future, but wanted to try to get my 8266 running first if possible.

SergioGasquez, 2023-02-24:
> Can you tell me if the patched compiler has full support for the ESP8266?

What do you mean with full support? You should be able to build projects for ESP8266 but I am not sure what features are available in the HAL, so even if you are able to build for your target you may not be able to do much with it. (I don't really know state of the esp8266-hal)