esp-rs/espup #194: `espup update` redownloads toolchain even when there isn't a newer one

VorpalBlade, 2023-02-27:
## Bug description

Unlike `rustup update`, `espup update` apparently will redownload the toolchain even when there are no changes.

- Would you like to work on a fix? [y/n]: n

## To Reproduce

Steps to reproduce the behavior:

1. `espup update`
2. `espup update`

## Expected behavior

Similar to `rustup update` it should not redownload everything if there are no changes.

## Screenshots

```console
$ espup update
[2023-02-27T18:21:23Z INFO ] 💽  Updating Espressif Rust ecosystem
[2023-02-27T18:21:23Z INFO ] 💡  Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases/latest'
[2023-02-27T18:21:23Z INFO ] 💡  Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases'
[2023-02-27T18:21:24Z INFO ] 🔧  Uninstalling Xtensa Rust toolchain
[2023-02-27T18:21:24Z INFO ] 🔧  Uninstalling Xtensa Rust toolchain
[2023-02-27T18:21:24Z INFO ] 🔧  Installing Xtensa Rust 1.67.0.0 toolchain
[2023-02-27T18:21:24Z INFO ] 📥  Downloading file '/tmp/.tmpWocF9U/rust.tar.xz' from 'https://github.com/esp-rs/rust-build/releases/download/v1.67.0.0/rust-1.67.0.0-x86_64-unknown-linux-gnu.tar.xz'
[2023-02-27T18:21:37Z INFO ] 🔧  Uncompressing tar.xz file to '/tmp/.tmpWocF9U'
[2023-02-27T18:21:43Z INFO ] 🔧  Installing 'rust' component for Xtensa Rust toolchain
[2023-02-27T18:21:45Z INFO ] 📥  Downloading file '/tmp/.tmpntRZqP/rust-src.tar.xz' from 'https://github.com/esp-rs/rust-build/releases/download/v1.67.0.0/rust-src-1.67.0.0.tar.xz'
[2023-02-27T18:21:46Z INFO ] 🔧  Uncompressing tar.xz file to '/tmp/.tmpntRZqP'
[2023-02-27T18:21:46Z INFO ] 🔧  Installing 'rust-src' component for Xtensa Rust toolchain
[2023-02-27T18:22:08Z INFO ] ✅  Update successfully completed!
$ espup update
[2023-02-27T18:22:33Z INFO ] 💽  Updating Espressif Rust ecosystem
[2023-02-27T18:22:33Z INFO ] 💡  Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases/latest'
[2023-02-27T18:22:34Z INFO ] 💡  Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases'
[2023-02-27T18:22:35Z INFO ] 🔧  Uninstalling Xtensa Rust toolchain
[2023-02-27T18:22:35Z INFO ] 🔧  Uninstalling Xtensa Rust toolchain
[2023-02-27T18:22:35Z INFO ] 🔧  Installing Xtensa Rust 1.67.0.0 toolchain
[2023-02-27T18:22:35Z INFO ] 📥  Downloading file '/tmp/.tmpTnF0Lb/rust.tar.xz' from 'https://github.com/esp-rs/rust-build/releases/download/v1.67.0.0/rust-1.67.0.0-x86_64-unknown-linux-gnu.tar.xz'
[2023-02-27T18:22:42Z INFO ] 🔧  Uncompressing tar.xz file to '/tmp/.tmpTnF0Lb'
[2023-02-27T18:22:48Z INFO ] 🔧  Installing 'rust' component for Xtensa Rust toolchain
[2023-02-27T18:22:50Z INFO ] 📥  Downloading file '/tmp/.tmp0erG98/rust-src.tar.xz' from 'https://github.com/esp-rs/rust-build/releases/download/v1.67.0.0/rust-src-1.67.0.0.tar.xz'
[2023-02-27T18:22:50Z INFO ] 🔧  Uncompressing tar.xz file to '/tmp/.tmp0erG98'
[2023-02-27T18:22:51Z INFO ] 🔧  Installing 'rust-src' component for Xtensa Rust toolchain
[2023-02-27T18:23:13Z INFO ] ✅  Update successfully completed!
```

## Environment

- OS: Arch Linux (rolling release)
- espup version: 0.3.0 (installed with cargo install).
- Host rust version: 1.67.1 (installed with rustup)

## Additional context

File system is btrfs, with `~/.rustup` on a separate subvolume (to exclude it from snapshots). No idea if that matters.

SergioGasquez, 2023-03-02:
Hi! Thanks for reporting this bug, just created #197 to fix it. Do you mind testing it and verifying that the installation is reused?

VorpalBlade, 2023-03-03:
Tested it. It works. Thanks!

VorpalBlade, 2023-03-03:
Perhaps this bit could use a bit more explaination (or a link to a page with more explainations):

```
[2023-03-03T09:33:00Z WARN ] ⚠️   Previous installation of Xtensa Rust 1.67.0 exists in: '/home/arvid/.rustup/toolchains/esp'. Reusing this installation. Since Xtensa Rust uses an extended semantic versioning, the toolchain might be different from the one you are expecting. If you want to reinstall the toolchain, please run `espup update -v <VERSION>`.
```

closed 2023-03-03 by SergioGasquez

SergioGasquez, 2023-03-03:
> Perhaps this bit could use a bit more explaination (or a link to a page with more explainations):

Thanks for the feedback. There is no deeper explanation of this (IIRC), we use an extended semver as you can see in https://github.com/esp-rs/rust-build/releases, while rust versions are using standard semver. 
We check the installed version with `cargo +<toolchain_name> --version`, which returns a standard semver. So, say that you have installed `1.67.0.0` but you want to update it to `1.67.0.1` (suppose it existed, ofc), in both cases `cargo +<toolchain_name> --version` will return `1.67.0`

VorpalBlade, 2023-03-03:
> Do you have any suggestion on how to improve the message?

Hm, not really. I would suggest that you should not rely on cargo to check semver if it unreliable with extended semver.

* How would I as an user be able to check which extended version I have installed? I for sure won't remember what I downloaded a couple of weeks ago!
* Can't you just use that same way to check programatically?
* If not, consider adding a `.esp-version` file or similar hidden file to the toolchain install that espup can check.

SergioGasquez, 2023-03-06:
> How would I as an user be able to check which extended version I have installed?  

I think there is no way to check that, sadly. The `.esp-version` file would be a good solution, so I will propose it, thanks for the suggestion!

VorpalBlade, 2023-03-06:
> I think there is no way to check that, sadly.

There probably should be, this seems like something the user might want to check! And it might be relevant for bug reports etc.

SergioGasquez, 2023-03-07:
Starting from `1.68.0.0`, which will be probably released later this week, we will have the extended semver as info in `rustc`. See https://github.com/esp-rs/rust-build/issues/194 and https://github.com/esp-rs/rust-build/pull/196/commits/7cae825649800f6b8db6f32e11ad1d7dc7d43bf6 for more information. So, I will update `espup` to check this, so we can assure if we have the same version installed.