esp-rs/espup #215: `espup install` failed.

windoze, 2023-03-26:
## Bug description

- Would you like to work on a fix? [y/n]

## To Reproduce

Steps to reproduce the behavior:

1. Just run `espup install -l debug`
2. It failed with the error
```
❯ espup install -l debug
[2023-03-26T10:28:45Z DEBUG] connecting to crates.io:443 at 13.33.88.97:443
[2023-03-26T10:28:45Z DEBUG] No cached session for DnsName(DnsName(DnsName("crates.io")))
[2023-03-26T10:28:45Z DEBUG] Not resuming any session
[2023-03-26T10:28:45Z DEBUG] Using ciphersuite TLS13_AES_128_GCM_SHA256
[2023-03-26T10:28:45Z DEBUG] Not resuming
[2023-03-26T10:28:45Z DEBUG] TLS1.3 encrypted extensions: [ServerNameAck]
[2023-03-26T10:28:45Z DEBUG] ALPN protocol is None
[2023-03-26T10:28:45Z DEBUG] created stream: Stream(RustlsStream)
[2023-03-26T10:28:45Z DEBUG] sending request GET https://crates.io/api/v1/crates/espup/versions
[2023-03-26T10:28:45Z DEBUG] writing prelude: GET /api/v1/crates/espup/versions HTTP/1.1
    Host: crates.io
    User-Agent: ureq/2.6.2
    Accept: */*
    accept-encoding: gzip
[2023-03-26T10:28:45Z DEBUG] Ticket saved
[2023-03-26T10:28:46Z DEBUG] Chunked body in response
[2023-03-26T10:28:46Z DEBUG] response 200 to GET https://crates.io/api/v1/crates/espup/versions
[2023-03-26T10:28:46Z DEBUG] dropping stream: Stream(RustlsStream)
[2023-03-26T10:28:46Z INFO ] 💽  Installing the Espressif Rust ecosystem
[2023-03-26T10:28:46Z INFO ] 💡  Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases/latest'
[2023-03-26T10:28:46Z DEBUG] 🐞  Auth header added.
[2023-03-26T10:28:46Z DEBUG] starting new connection: https://api.github.com/
[2023-03-26T10:28:47Z DEBUG] 🐞  Parsing Xtensa Rust version: null
[2023-03-26T10:28:47Z INFO ] 💡  Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases'
[2023-03-26T10:28:47Z DEBUG] 🐞  Auth header added.
[2023-03-26T10:28:47Z DEBUG] starting new connection: https://api.github.com/
Error: espup::toolchain::rust::invalid_version

  × ⛔  Invalid toolchain version 'null'. Verify that the format is correct: '<major>.<minor>.<patch>.<subpatch>' or '<major>.<minor>.<patch>', and
  │ that the release exists in https://github.com/esp-rs/rust-build/releases
```

Although I haven't dug into the source code, but I guess it's because `esp-rs/rust-build` still doesn't have 1.68.1 released.

But when I ran `espup install -v 1.68.0`, it still failed with the error:
```
thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', /Users/chenxu/.cargo/registry/src/github.com-1ecc6299db9ec823/espup-0.3.2/src/toolchain/rust.rs:122:44
```

## Expected behavior

Toolchain gets installed correctly.

## Environment

- OS: macOS 13.2.1
- espup version: 0.3.2

SergioGasquez, 2023-03-28:
Hello @windoze!

Thanks for reporting this issue! The issue seems to be related with Github API, we had something similar in the past, see https://github.com/esp-rs/xtensa-toolchain/issues/15. But, the weird thing is that it happens on a host machine, where I don't expect you to be querying the Github API, so you should not reach [the limits](https://docs.github.com/en/rest/overview/resources-in-the-rest-api?apiVersion=2022-11-28).  Can you try to run the `espup install` command again? If it fails, please run `curl -i https://api.github.com/users/octocat` and let's see the `x-ratelimit-remaining` property.

SergioGasquez, 2023-03-29:
In your case you are not in the limit:

> x-ratelimit-limit: 60
> x-ratelimit-remaining: 58

Here is some documentation about the limits: https://docs.github.com/en/rest/overview/resources-in-the-rest-api?apiVersion=2022-11-28#rate-limit-headers 

- The error that you are getting is exactly the same as the one posted in your original post after step `2.`?
- Can you check if you have a `GITHUB_TOKEN` environment variable defined? `echo $GITHUB_TOKEN`

SergioGasquez, 2023-03-29:
> After some digging I realized the root cause is because I accidentally set a GITHUB_TOKEN in the environment, for some reason it's picked up by `espup` and caused the error. I unset the environment variable and now everything works as expected.
> 
> Sorry for the false alarm and thanks for your kind help.
> 
> PS. I'd suggest adding a line in the doc to clarify the env var usage.

Glad to hear that! The [debug message that you were seeing](https://github.com/esp-rs/espup/blob/main/src/toolchain/mod.rs#LL142C32-L142C37) indicates that you have a `GITHUB_TOKEN`, indeed.

This variable is required when using `espup` in CI, otherwise we will reach the limit. I will leave this issue open and add a small section in the README explaining this behavior.

Sorry for your trouble and really happy that is now fixed!

windoze, 2023-03-29:
I dug the exact same line of the source code to be aware of this issue 🤣

SergioGasquez, 2023-03-29:
Just added a note under the installation section, see: https://github.com/esp-rs/espup/commit/ef483edf2799e665cc2af1fc0ac231598c8e1f11

I'll close the issue, if you have any further question or suggestion feel free to post or reopen it again!

Thank you!

closed 2023-03-29 by SergioGasquez

brandonros, 2024-04-27:
Worded another way: make sure if you do have a GITHUB_TOKEN in your profile/environment/shell that it isn't bogus :)

Might be nice to beef up error handling to make it more clear when a GitHub API request fails.