esp-rs/espup #229: espup defaults to installing for latest version of macos

stijlist, 2023-03-30:
## Bug description

It looks like the rustc version espup downloads is for macOS 13.
I'm on macOS 11, so I get a dynamic link error for libc++.

Is there a workaround?
Happy to build from source but I don't think espup or esp-rs/rust-build currently have options to build from source.

```
~/c/esp-test-app::> cargo build --release
error: process didn't exit successfully: `rustc -vV` (signal: 6, SIGABRT: process abort signal)
--- stderr
dyld: Symbol not found: __ZNKSt3__115basic_stringbufIcNS_11char_traitsIcEENS_9allocatorIcEEE3strEv
  Referenced from: /Users/bert/.rustup/toolchains/esp/bin/../lib/librustc_driver-8e0ff663f8797d8e.dylib (which was built for Mac OS X 13.0)
  Expected in: /usr/lib/libc++.1.dylib
```

- Would you like to work on a fix? [y/n]

## To Reproduce

Steps to reproduce the behavior:

1. espup install
2. cargo generate --git https://github.com/esp-rs/esp-idf-template cargo
3. cargo build --release

## Expected behavior

`cargo build` succeeds.

## Environment

- OS: macOS 11.2 Big Sur
- espup version: [e.g. 0.1.0]

SergioGasquez, 2023-03-31:
Hello, 

`espup` and the scripts from `rust-build` both do the same thing, download the generated artifacts that [live in rust-build](https://github.com/esp-rs/rust-build/releases). Those are generated with CI as a result of building the Rust in the esp-rs/rust repo (I suggest moving this issue to `rust-build` repo as It's where the CI generating the artifacts lives).

I am not really sure how to proceed here, to be honest, we had some instructions on how to build esp-rs/rust from source, see: https://github.com/esp-rs/rust/blob/8915ba7b1b6433032400ace217e765bd96686763/README.md#installing-from-source. But this task is slow and tedious. Updating your macOS version, I suppose is not an option (?)

SergioGasquez, 2023-04-01:
Just tried to reproduce your error on a macOS 11.7.1 and I was able to suscesfully build the `esp-idf-template` here is what I had to install:
- `xcode-select`
- `pkg-config` and `openssl` to install `cargo-generate`
- `cargo-generate`
- `espup`
- `ldproxy`

Then I ran the following:
```
espup install
. /Users/<user>/export-esp.sh
cargo generate --git https://github.com/esp-rs/esp-idf-template cargo
# I used an esp32
cargo build
````

stijlist, 2023-04-04:
Strange, this is also what I did.

```
espup install
. /Users/<user>/export-esp.sh
cargo generate --git https://github.com/esp-rs/esp-idf-template cargo
# I used an esp32
cargo build
```

Will update to 11.7.1 and try again.

stijlist, 2023-04-04:
I ended up building my own clang, llvm, and rustc from https://github.com/esp-rs/rust, then tried to cargo build esp-idf-template - that build just completed successfully so I might be good.

[image]

SergioGasquez, 2023-04-05:
> I ended up building my own clang, llvm, and rustc from https://github.com/esp-rs/rust, then tried to cargo build esp-idf-template - that build just completed successfully so I might be good.

Happy to read that! You are now probably an expert building `llvm` and `rustc`!

closed 2023-04-05 by SergioGasquez