esp-rs/espup #257: gdb missing since esp-12.2.0_20230208

svenstaro, 2023-05-19:
Since [esp-12.2.0_20230208](https://github.com/espressif/crosstool-NG/releases/tag/esp-12.2.0_20230208), gdb is no longer part of the official release. You can read about this change from upstream in their [esp-2022r1 release notes](https://github.com/espressif/crosstool-NG/releases/tag/esp-2022r1).

gdb now lives [here](https://github.com/espressif/binutils-gdb/releases) so it needs to be separately downloaded and extracted on top of the current gcc toolchain location.

SergioGasquez, 2023-05-25:
Hi! I brought this topic into today's [community meeting](https://github.com/esp-rs/rust/discussions/175) and we've decided that `espup` should not install GDB, as we discussed in https://github.com/esp-rs/espup/discussions/154, we want to keep `espup` as minimal as possible, that means, installing only what's necessary to generate a binary for ESP32s.

Still, I think GDB is useful/necessary, so I will open an issue on the book to document how to install it.

svenstaro, 2023-05-25:
I'm a bit unhappy about this as `espup` is meant to be a replacement for the otherwise annoying ad-hoc install scripts everybody would otherwise write for themselves. This decision has us make a small script that gets the newest version of esp gdb from just the right place and pops it into place in addition to running `espup`. This functionality used to be there and now it's gone without real replacement which isn't great.

I think it would make sense to at least add an option to optionally install it. Maybe just `espup install --gdb`?

SergioGasquez, 2023-05-26:
Let's proceed the discussion here instead that Matrix (@svenstaro @ivmarkov), as a recap: 
 - It's important to note that [`esp-idf-sys` is able to download/install](https://github.com/esp-rs/esp-idf-sys/blob/63b8713e199ad7d1ed106a73d8187a224afefd34/build/native/cargo_driver.rs#L70) the proper gcc that corresponds with the ESP-IDF version being used, and it could download GDB
   - `no_std` projects dont use GDB
   - We would need to see how it can be integrated with openocd
     - [OpenOCD can also be dowloaded from `esp-idf-sys`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html?highlight=gdb#openocd-esp32)
- `espup` never intentionally installed GDB, it was a sideffect

svenstaro, 2023-05-26:
I think that would work for me.

SergioGasquez, 2023-06-22:
Hello, we wont be including GDB in `espup`, we will be doing some efforts to improve the current support in `probe-rs` (see https://github.com/esp-rs/rust/discussions/181#discussion-5316670) and also, we will better document (probably in the book) how to install GDB/openOCD.