esp-rs/espup #270: Windows: espup should display error when Path is reaching too long

georgik, 2023-05-31:
## Bug description

`espup install` updates Path variable, but does not display any warning about reaching limit 2047 characters which is limit on Windows.
Reaching such value makes it impossible to update the variable by user.

SergioGasquez, 2023-06-02:
I tried to reproduce this error but couldn't, did you encounter this error with version `0.4.1` of espup? Here is my reproduction attempt:
```
PS C:\Users\sergio> $Env:path.length
2026
PS C:\Users\sergio> espup install -v 1.70.0.1
....
<After installation>
PS C:\Users\sergio> $Env:path.length
3165    
```
I think @9names also [tried to use a "very long path"](https://github.com/esp-rs/espup/issues/233#issuecomment-1519227661) with no issues. @9names, do you recall if it was longer than 2047 characters?

9names, 2023-06-03:
I think @georgik isn't saying that `espup` has an issue updating the path.
Rather, they are saying that *after* it's updated, and the path is > 2047 characters they can no longer change the path via other means.
I don't think warning would be sufficient to help here - you'd really need a prompt to ensure the user is okay with their path becoming too large during installation.

@georgik: how are you trying to update your path after `espup` installation?
I recommend using Rapid Environment Editor to remove any old paths, and if necessary use it to convert paths to their shorter `8.3` form.

SergioGasquez, 2023-06-29:
Do you think that the following warning message: `Your PATH environment variable is longer than 2047 characters. Windows will not allow you editting it.` will be sufficient?

georgik, 2023-06-29:
I recommend requesting user consent at this point or hard-stop the installation with a hint for parameters to override the warning.

bugadani, 2023-07-29:
Oh, this has the potential of breaking VSCode terminal. I was wondering why I could no longer use `cargo` there, and the issue was resolved when I removed my duplicate (everything was in the list like 5 or 6 times) PATH variables.

So, user programs may have issues with using very long path variables. VSCode has such issues, while Windows Terminal doesn't.

SergioGasquez, 2023-08-02:
> Oh, this has the potential of breaking VSCode terminal. I was wondering why I could no longer use cargo there, and the issue was resolved when I removed my duplicate (everything was in the list like 5 or 6 times) PATH variables.

Did you install the environment several times?

bugadani, 2023-08-02:
> Did you install the environment several times?

Yes, I did. That it duplicated its PATH entries is a separate issue, but the end result is the same: I ended up with an unworkably long PATH.

Steve-xmh, 2023-08-15:
I also confused about this issue, I'm now using espup 0.5.0 and when I execute install command it will merge PATH environment variable in system environment with those at user environment and then override the user environment with oversized paths. What's worse, it duplicated toolchain paths, which makes it more larger.

The command:
[image]

Before the installation (some unimportant content are masked):
[image]

After the installation (some unimportant content are masked):
[image]