esp-rs/espup #340: Temporary directory doesn't seem to be cleaned after installation / update

AnthonyGrondin, 2023-09-05:
## Bug description

tempfile doesn't seem to automatically clear the directory it creates after updating or installing the toolchain.
I get a lot of `/tmp/.tmpXXXXXX/` directories that are used by espup to store the downloaded assets. Each one is about 1.2Gb, so changing frequently between toolchains can result in a clogged up `/tmp/` quite rapidly.

- Would you like to work on a fix? No

## To Reproduce

Steps to reproduce the behavior:

1. Install any toolchain `espup install -v 1.70.0.0`
2. Note the `/tmp/.tmpXXXXXX/` it creates, in the logs
3. Install any other toolchain `espup update -v 1.72.0.0`
4. Note the `/tmp/.tmpXXXXXX/` it creates, in the logs
5. Check `/tmp/`, both folders are still present after installation.

## Expected behavior

Directories should be cleaned when the program ends or crash.
See [resource leaking](https://docs.rs/tempfile/3.8.0/tempfile/#resource-leaking) in tempfile

## Screenshots

N/A

## Environment

- OS: 5.15.109-1-MANJARO
- espup version: 0.5.0

## Additional context
N/A

SergioGasquez, 2023-09-06:
Thanks for reporting the issue! I think I've found the issue, tomorrow I will do some further testing an open a PR for it.

SergioGasquez, 2023-09-07:
Hi! Just created #344, do you mind trying the fix proposed in the PR?
```sh
cargo uninstall espup
cargo install espup --git https://github.com/SergioGasquez/espup --branch fix/tmp-folders
espup install 
```

AnthonyGrondin, 2023-09-07:
I can confirm that this fixes my issue.

It will still leave the temporary directory there if the process is interrupted with Ctrl^C , but the main case is fixed.

SergioGasquez, 2023-09-07:
Thanks for the testing!

closed 2023-09-07 by SergioGasquez

AnthonyGrondin, 2023-09-07:
Thanks for taking care of this.