esp-rs/espup #341: Compiling it on macOS-arm64

adrien3d, 2023-09-06:
## Bug description

While running `cargo install espup`, I have a bug at compilation time.
I suspect that it is due to the fact I am on a ARM Mac M1.

Here is the error:
```
  = note: ld: warning: ignoring file /opt/homebrew/Cellar/xz/5.4.4/lib/liblzma.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
          Undefined symbols for architecture x86_64:
            "_lzma_code", referenced from:
                _$LT$xz2..bufread..XzDecoder$LT$R$GT$$u20$as$u20$std..io..Read$GT$::read::h0cf12baeac3eea91 in espup-d3739f70c9711ea7.espup-6e07ce6c45161945.espup.1576fe4529cab319-cgu.15.rcgu.o.rcgu.o
            "_lzma_end", referenced from:
                core::ptr::drop_in_place$LT$tar..archive..Archive$LT$xz2..read..XzDecoder$LT$$RF$$u5b$u8$u5d$$GT$$GT$$GT$::hef792c4607495d10 in espup-d3739f70c9711ea7.espup-6e07ce6c45161945.espup.1576fe4529cab319-cgu.00.rcgu.o.rcgu.o
                core::ptr::drop_in_place$LT$tar..archive..Archive$LT$xz2..read..XzDecoder$LT$$RF$$u5b$u8$u5d$$GT$$GT$$GT$::hef792c4607495d10 in espup-d3739f70c9711ea7.espup-6e07ce6c45161945.espup.1576fe4529cab319-cgu.02.rcgu.o.rcgu.o
                core::ptr::drop_in_place$LT$tar..archive..Archive$LT$xz2..read..XzDecoder$LT$$RF$$u5b$u8$u5d$$GT$$GT$$GT$::hef792c4607495d10 in espup-d3739f70c9711ea7.espup-6e07ce6c45161945.espup.1576fe4529cab319-cgu.12.rcgu.o.rcgu.o
                core::ptr::drop_in_place$LT$xz2..read..XzDecoder$LT$$RF$$u5b$u8$u5d$$GT$$GT$::ha2e9c68d18256a65 (.llvm.4747039821596474401) in espup-d3739f70c9711ea7.espup-6e07ce6c45161945.espup.1576fe4529cab319-cgu.13.rcgu.o.rcgu.o
                core::ptr::drop_in_place$LT$xz2..stream..Stream$GT$::hce3b6527b1e25e20 (.llvm.12690339763100604783) in espup-d3739f70c9711ea7.xz2-f0a1d1668fd3957c.xz2.aaba150174409dfe-cgu.1.rcgu.o.rcgu.o
                xz2::stream::Stream::new_stream_decoder::h1d673b021f6a6278 in espup-d3739f70c9711ea7.xz2-f0a1d1668fd3957c.xz2.aaba150174409dfe-cgu.1.rcgu.o.rcgu.o
            "_lzma_stream_decoder", referenced from:
                xz2::stream::Stream::new_stream_decoder::h1d673b021f6a6278 in espup-d3739f70c9711ea7.xz2-f0a1d1668fd3957c.xz2.aaba150174409dfe-cgu.1.rcgu.o.rcgu.o
          ld: symbol(s) not found for architecture x86_64
          clang: error: linker command failed with exit code 1 (use -v to see invocation)
          

error: could not compile `espup` (bin "espup") due to previous error
error: failed to compile `espup v0.5.0`, intermediate artifacts can be found at `/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-installSam5Um`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
```

SergioGasquez, 2023-09-06:
Which channel are you using? Just tried installing it on my M1 Mac with no issue using `stable 1.71.1` and `nightly 2023-08-03`

adrien3d, 2023-09-06:
My bad, working now, but no idea why 😅

SergioGasquez, 2023-09-06:
Maybe the intermediate artifacts were corrupted! Glad it now works!

SergioGasquez, 2025-02-07:
> I have the same issue... any idea how to clean/rebuild?

Try removing the intermediate artifacts. You can also use [`cargo binstall`](https://github.com/cargo-bins/cargo-binstall) or dowload the [release artifacts](https://github.com/esp-rs/espup/releases/tag/v0.14.0)

maelp, 2025-02-07:
I tried cleaning the cache, but it still does this

```
          Undefined symbols for architecture x86_64:
            "_lzma_code", referenced from:
                _$LT$xz2..bufread..XzDecoder$LT$R$GT$$u20$as$u20$std..io..Read$GT$::read::ha0e53d453563bc22 in espup-cebad64c0ee11acd.espup-da3766e98d63a46f.espup.86c0427a865c97b7-cgu.10.rcgu.o.rcgu.o
                _$LT$xz2..bufread..XzDecoder$LT$R$GT$$u20$as$u20$std..io..Read$GT$::read::ha0e53d453563bc22 in espup-cebad64c0ee11acd.espup-da3766e98d63a46f.espup.86c0427a865c97b7-cgu.10.rcgu.o.rcgu.o
            "_lzma_end", referenced from:
                core::ptr::drop_in_place$LT$tar..archive..Archive$LT$xz2..read..XzDecoder$LT$$RF$$u5b$u8$u5d$$GT$$GT$$GT$::h731d59a59386eef3 in espup-cebad64c0ee11acd.espup-da3766e98d63a46f.espup.86c0427a865c97b7-cgu.00.rcgu.o.rcgu.o
                core::ptr::drop_in_place$LT$tar..archive..Archive$LT$xz2..read..XzDecoder$LT$$RF$$u5b$u8$u5d$$GT$$GT$$GT$::h731d59a59386eef3 in espup-cebad64c0ee11acd.espup-da3766e98d63a46f.espup.86c0427a865c97b7-cgu.04.rcgu.o.rcgu.o
                core::ptr::drop_in_place$LT$xz2..read..XzDecoder$LT$$RF$$u5b$u8$u5d$$GT$$GT$::hda5182fa1815e34a (.llvm.3045471006299665561) in espup-cebad64c0ee11acd.espup-da3766e98d63a46f.espup.86c0427a865c97b7-cgu.13.rcgu.o.rcgu.o
                core::ptr::drop_in_place$LT$tar..archive..Archive$LT$xz2..read..XzDecoder$LT$$RF$$u5b$u8$u5d$$GT$$GT$$GT$::h731d59a59386eef3 in espup-cebad64c0ee11acd.espup-da3766e98d63a46f.espup.86c0427a865c97b7-cgu.13.rcgu.o.rcgu.o
                xz2::stream::Stream::new_stream_decoder::h8f34e6abc3c79656 in espup-cebad64c0ee11acd.xz2-1e0afd2aa63b0484.xz2.f0311f4becf3624e-cgu.0.rcgu.o.rcgu.o
                xz2::stream::Stream::new_stream_decoder::h8f34e6abc3c79656 in espup-cebad64c0ee11acd.xz2-1e0afd2aa63b0484.xz2.f0311f4becf3624e-cgu.0.rcgu.o.rcgu.o
            "_lzma_stream_decoder", referenced from:
                xz2::stream::Stream::new_stream_decoder::h8f34e6abc3c79656 in espup-cebad64c0ee11acd.xz2-1e0afd2aa63b0484.xz2.f0311f4becf3624e-cgu.0.rcgu.o.rcgu.o
          ld: symbol(s) not found for architecture x86_64
          clang: error: linker command failed with exit code 1 (use -v to see invocation)

error: could not compile `espup` (bin "espup") due to 1 previous error
error: failed to compile `espup v0.14.0`, intermediate artifacts can be found at `/var/folders/6d/tgjjs7m13yngmk287mk9cv9h0000gp/T/cargo-install7Kv5Z4`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
```

maelp, 2025-02-07:
I'm on `stable`, not sure exactly how to change and which one I could try?

Message ID: ***@***.***>

SergioGasquez, 2025-02-07:
Which stable are you using? You can change stable version or use nigthly by: `rustup toolchain install nightly` `cargo +nightly install espup`

maelp, 2025-02-07:
[image]

I'm using this

SergioGasquez, 2025-02-07:
You can try `cargo +esp install espup`

maelp, 2025-02-07:
So `cargo +esp install espup` works, is that because espup requires the pipeline to be installed? but isn't that circular?

SergioGasquez, 2025-02-07:
> So `cargo +esp install espup` works, is that because espup requires the pipeline to be installed? but isn't that circular?

No, `espup` doesnt require the toolchain to be installed. `stable` and `nightly` should also work, but there seems to be something wrong with your versions of it, maybe they are too old or intermediate artiffacts are corrupt

adrien3d, 2025-06-19:
Edit: fixed reinstalling completely rustup, and removing rust installed from brew

closed 2025-06-19 by adrien3d