esp-rs/espup #342: espup uses large amounts of space in $TMPDIR

jmesmon, 2023-09-06:
## Bug description

`espup` downloads **and extracts** data in `$TMPDIR`, resulting in ~1.2GiB of space usage. `$TMPDIR` is often a ram disk, so we're storing data to ram. We later copy this same data to other locations (directories in `$RUSTUP_HOME`, etc).

## To Reproduce

Steps to reproduce the behavior:

1. Point `TMPDIR` at a tmpfs (probably the default, if you're using linux with systemd)
2. Run `espup` so that it downloads & installs a toolchain
3. Use `df -h` to note 1.2GiB of tmpfs usage

## Expected behavior

1. Use less tmpfs space
2. Minimally, don't extract the tar balls to `$TMPDIR`, instead extracting them adjacent to their final location so they can be moved into place
3. Ideally, extract the file as a streaming operation as it is downloaded to avoid the use of `$TMPDIR` entirely.

## Environment

- OS: nixos
- espup version: 0.5.0

## Additional context

Systemd defaults `RuntimeDirectory` (which `$TMPDIR` points to) to 10% of avaliable memory. This means on a system with 2GiB of ram, we get 200MiB. With the current 1.2GiB of usage, espup users with linux + systemd defaults will need to ensure they have ~12GiB of ram.

jmesmon, 2023-09-07:
A good option here:

Download & extract files to the `rustup` tmp dir (`$HOME/.rustup/tmp`, `$RUSTUP_HOME/tmp`, etc) instead of `$TMPDIR`. This would mirror rustup behavior which folks probably expect.

SergioGasquez, 2023-09-08:
> A good option here:
> 
> Download & extract files to the `rustup` tmp dir (`$HOME/.rustup/tmp`, `$RUSTUP_HOME/tmp`, etc) instead of `$TMPDIR`. This would mirror rustup behavior which folks probably expect. (even if they don't like it much. I have symlinks in place so that `$RUSTUP_HOME/tmp` ends up in `$XDG_CACHE_HOME`).

Good idea! Just implemented this, do you mind testing it?

```
cargo uninstall espup
cargo install espup --git https://github.com/SergioGasquez/espup --branch feature/update-tmp-loc
espup install 
```

SergioGasquez, 2023-09-12:
Updated the branch to use `$RUSTUP_HOME/tmp` and simplified the code a bit!

jmesmon, 2023-09-13:
Works for me 👍

closed 2023-09-14 by SergioGasquez