esp-rs/espup #343: If rustup toolchain dir includes non-directory entries, `espup install` panics

jmesmon, 2023-09-06:
## Bug description

When a rustup toolchain dir managed by espup (ie: `~/.rustup/toolchains/esp`) includes a non-directory entry (ie: a normal file) at the top level, `espup` panics when it attepts to remove that file using `remove_dir_all()`

## To Reproduce

Steps to reproduce the behavior:

1. `espup install`
2. `touch ~/.rustup/toolchains/esp/hello.sh`
3. `espup install`
4. Note the panic

## Expected behavior

`espup` should either return an error or (as it's already wiping out the directory anyhow) remove the file.

## Environment

- OS: nixos
- espup version: 0.5.0

SergioGasquez, 2023-09-07:
Thanks for reporting the issue! Just opened #346 fixing this, now when reproducing the issue you get:
```
Error: espup::remove_directory

  × ⛔  Failed to remove '/home/esp/.rustup/toolchains/esp/hello.sh'.
```

Do you mind trying the fix to validate it works?
```
cargo uninstall espup
cargo install espup --git https://github.com/SergioGasquez/espup --branch fix/rustup-panic
mkdir ~/.rustup/toolchains/esp/ && touch ~/.rustup/toolchains/esp/hello.sh
espup install 
```

jmesmon, 2023-09-07:
It seems to kind-of resolve the issue, though there's still a panic that might be related before that (after some repeated uninstall attempts).

```
[2023-09-07T14:50:18Z INFO ] 🔧  Uninstalling Xtensa Rust toolchain
[2023-09-07T14:50:18Z WARN ] ⚠️   Installation for 'Xtensa Rust' failed, retrying
[2023-09-07T14:50:18Z INFO ] 🔧  Uninstalling Xtensa Rust toolchain
[2023-09-07T14:50:18Z WARN ] ⚠️   Installation for 'Xtensa Rust' failed, retrying
[2023-09-07T14:50:18Z INFO ] 🔧  Uninstalling Xtensa Rust toolchain
[2023-09-07T14:50:18Z WARN ] ⚠️   Installation for 'Xtensa Rust' failed, retrying
[2023-09-07T14:50:18Z INFO ] 🔧  Uninstalling Xtensa Rust toolchain
[2023-09-07T14:50:18Z WARN ] ⚠️   Installation for 'Xtensa Rust' failed, retrying
thread 'tokio-runtime-worker' panicked at 'called `Result::unwrap()` on an `Err` value: SendError { .. }', /home/cody/.cargo/git/checkouts/espup-c286dba105940cd2/fe2e25d/src/toolchain/mod.rs:248:32
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Error: espup::remove_directory

  × ⛔  Failed to remove '/home/cody/.rustup/toolchains/esp/hello.sh'.

```

Given that we're already removing all the other files under this directory, it probably would make sense to resolve by removing the top-level files as well.

SergioGasquez, 2023-09-08:
> Given that we're already removing all the other files under this directory, it probably would make sense to resolve by removing the top-level files as well.

Just updated the branch to delete everything! Mind validating it?
```
cargo uninstall espup
cargo install espup --git https://github.com/SergioGasquez/espup --branch fix/rustup-panic
mkdir ~/.rustup/toolchains/esp/ && touch ~/.rustup/toolchains/esp/hello.sh
espup install 
```

jmesmon, 2023-09-08:
Looks good to me 👍

closed 2023-09-12 by SergioGasquez