esp-rs/espup #387: ESP32 and lv-binding-rs does not build

enelson1001, 2023-11-05:
This may not be the correct forum for this problem but it is similar to Issue #163, where jasta couldn't find string.h.

I am trying to build esp32s3 app with lv-binding-rs but I am getting errors.

The first error I received was the following
```
--- stderr
  /usr/include/stdint.h:26:10: fatal error: 'bits/libc-header-start.h' file not found
  thread 'main' panicked at /home/ed/.cargo/registry/src/index.crates.io-6f17d22bba15001f/lvgl-sys-0.6.2/build.rs:236:10:
  Unable to generate bindings: ClangDiagnostic("/usr/include/stdint.h:26:10: fatal error: 'bits/libc-header-start.h' file not found\n")
```
I was able to fix this error by updating my config.toml and adding BINDGEN_EXTRA_CLANG_ARGS.

My config.toml
```
[build]
target = "xtensa-esp32s3-espidf"

[target.xtensa-esp32s3-espidf]
linker = "ldproxy"
# runner = "espflash --monitor" # Select this runner for espflash v1.x.x
runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x
rustflags = [ "--cfg",  "espidf_time64"] # Extending time_t for ESP IDF 5: https://github.com/esp-rs/rust/issues/110

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32s3"
#TARGET="esp32s3"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.1.1"

# directory that has lv_conf.h, lv_drv_conf.h
DEP_LV_CONFIG_PATH = { relative = true, value = "include" } # directory that has lv_conf.h, lv_drv_conf.h

# Required or get error ("/usr/include/stdint.h:26:10: fatal error: 'bits/libc-header-start.h' file not found\n")
# Seems to be using the wrong SYS ROOT
BINDGEN_EXTRA_CLANG_ARGS = '--sysroot=/home/ed/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include'
```
But then the next error I receive is the following:
```
--- stderr
  /home/ed/.cargo/registry/src/index.crates.io-6f17d22bba15001f/lvgl-sys-0.6.2/vendor/lvgl/src/misc/lv_mem.h:20:10: fatal error: 'string.h' file not found
  thread 'main' panicked at /home/ed/.cargo/registry/src/index.crates.io-6f17d22bba15001f/lvgl-sys-0.6.2/build.rs:236:10:
  Unable to generate bindings: ClangDiagnostic("/home/ed/.cargo/registry/src/index.crates.io-6f17d22bba15001f/lvgl-sys-0.6.2/vendor/lvgl/src/misc/lv_mem.h:20:10: fatal error: 'string.h' file not found\n")
```

But /home/ed/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include does include string.h

```
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/bits
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/c++
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/machine
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/rpc
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/ssp
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/sys
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/xtensa
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/alloca.h
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/_ansi.h
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/ar.h
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/argz.h
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/assert.h
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/complex.h
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/cpio.h
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/ctype.h
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/devctl.h
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/dirent.h
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/elf.h
/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/envlock.h
[... 90 lines cut ...]

#lvgl = { version = "0.6.2"}
#lvgl = { version = "0.6.2", default-features = false, features = ["rust_timer", ]}
#lvgl = { version = "0.6.2", default-features = false, features = ["drivers", "rust_timer", "unsafe_no_autoinit", ]}
#lvgl = { version = "0.6.2", default-features = false, features = ["rust_timer", "unsafe_no_autoinit", ]}

lvgl = { version = "0.6.2", default-features = false, features = ["drivers", "unsafe_no_autoinit"]}

#lvgl = { version = "0.6.2", default-features = false, features = ["unsafe_no_autoinit"]}

anyhow = "1"

sdl2 = "0.35.2"

[build-dependencies]
embuild = "0.31.3"
cc = "1.0"
```

My export-esp.sh
```
export LIBCLANG_PATH="/home/ed/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.0-20230516/esp-clang/lib"
export PATH="/home/ed/.rustup/toolchains/esp/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:$PATH"
export PATH="/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:$PATH"
```

SergioGasquez, 2023-11-06:
Have you tried installing full LLVM? `espup install --extended-llvm` and setting the env vars mentioned in this reply https://github.com/esp-rs/espup/issues/163#issuecomment-1404076113 ?

enelson1001, 2023-11-07:
I installed ```esup install --extended-llvm``` and added the following env vars to export-esp.sh but still the same error.
```
export-esp.sh 

#export CROSS_COMPILE="xtensa-esp32-elf"
#export CC="xtensa-esp32-elf-cc"
#export AR="xtensa-esp32-elf-ar"
#export C_INCLUDE_PATH="/home/ed/.rustup/toolchains/esp/xtensa-esp-elf/esp-12.2.0_20230208/xtensa-esp-elf/lib/gcc/extensa-esp-elf/13.2.0/include"
```

I did make a project like jasta (esp-ec-broken) and modified to double.h that included <string.h> and used printf inside double.c and everything built and ran OK. So string.h can be found on simple build.

I have cloned lv-binding-rs and playing around with lvgl/build.rs but so far every thing I have tried hasn't eliminated the error.  

I run the cargo build with the command: ```cargo build -vvv -p lvgl &>log3.txt``` to try and debug.

This is the lvgl-sys compile output.  
```
Compiling lvgl-sys v0.6.2 (/home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys)
     Running `
BINDGEN_EXTRA_CLANG_ARGS=--sysroot=.embuild/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/xtensa-esp32s3-elf 
CARGO=/home/ed/.rustup/toolchains/esp/bin/cargo 
CARGO_CRATE_NAME=build_script_build 
CARGO_MANIFEST_DIR=/home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys 
CARGO_PKG_AUTHORS='Rafael Caricio <crates.lvgl-sys@caric.io>:Nia Espera <a5b6@riseup.net>' 
CARGO_PKG_DESCRIPTION='Raw bindings to the LVGL C library.' 
CARGO_PKG_HOMEPAGE='https://github.com/rafaelcaricio/lvgl-rs' 
CARGO_PKG_LICENSE=MIT 
CARGO_PKG_LICENSE_FILE='' 
CARGO_PKG_NAME=lvgl-sys 
CARGO_PKG_README=README.md 
CARGO_PKG_REPOSITORY='https://github.com/rafaelcaricio/lvgl-rs' 
CARGO_PKG_RUST_VERSION='' 
CARGO_PKG_VERSION=0.6.2 
CARGO_PKG_VERSION_MAJOR=0 
CARGO_PKG_VERSION_MINOR=6 
CARGO_PKG_VERSION_PATCH=2 
CARGO_PKG_VERSION_PRE='' 
DEP_LV_CONFIG_PATH=/home/ed/esp-rust-projects/esp32-s3/hello-tft/include 
ESP_IDF_VERSION=v5.1.1 
LD_LIBRARY_PATH='/home/ed/esp-rust-projects/esp32-s3/hello-tft/target/debug/deps:
                 /home/ed/.rustup/toolchains/esp/lib:
                 /home/ed/.rustup/toolchains/esp/lib' 
MCU=esp32s3 
TARGET=xtensa-esp32s3-espidf 
/home/ed/.rustup/toolchains/esp/bin/rustc 
--crate-name build_script_build 
--edition=2021 /home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/build.rs 
--error-format=json 
--json=diagnostic-rendered-ansi,artifacts,future-incompat 
--crate-type bin 
--emit=dep-info,link 
-C embed-bitcode=no 
-C debuginfo=2 
--cfg 'feature="drivers"' 
-C metadata=96134a94547ec701 
-C extra-filename=-96134a94547ec701 
--out-dir /home/ed/esp-rust-projects/esp32-s3/hello-tft/target/debug/build/lvgl-sys-96134a94547ec701 
-C incremental=/home/ed/esp-rust-projects/esp32-s3/hello-tft/target/debug/incremental 
-L dependency=/home/ed/esp-rust-projects/esp32-s3/hello-tft/target/debug/deps 
--extern bindgen=/home/ed/esp-rust-projects/esp32-s3/hello-tft/target/debug/deps/libbindgen-86b341473cead6d4.rlib 
--extern cc=/home/ed/esp-rust-projects/esp32-s3/hello-tft/target/debug/deps/libcc-3af3a451b3e6eaab.rlib`
   ```
Do you see anything that looks wrong?

Here is a typical compile for an lvgl file
```
[lvgl-sys 0.6.2] running: 
"cc" 
"-Os" 
[... 36 lines cut ...]
    10:     0x55a83de657b3 - core::panicking::panic_fmt::he0893255bd367bf2
    11:     0x55a83de66003 - core::result::unwrap_failed::h458e1c6bbe8e82c7
    12:     0x55a83de6cdf2 - core::result::Result<T,E>::expect::h06a112e23e3bd6ed
                                 at /home/rust/rust/library/core/src/result.rs:1034:23
    13:     0x55a83de7035d - build_script_build::main::h42d6494bec9bf86b
                                 at /home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/build.rs:308:20
    14:     0x55a83de6628b - core::ops::function::FnOnce::call_once::hd5673faacdaca33e
                                 at /home/rust/rust/library/core/src/ops/function.rs:250:5
    15:     0x55a83de6a37e - std::sys_common::backtrace::__rust_begin_short_backtrace::h053ad9f9ebcca64b
                                 at /home/rust/rust/library/std/src/sys_common/backtrace.rs:154:18
    16:     0x55a83de691e1 - std::rt::lang_start::{{closure}}::h65bd0efbfeaa3530
                                 at /home/rust/rust/library/std/src/rt.rs:166:18
    17:     0x55a83e657c07 - std::rt::lang_start_internal::h493f682db1da9f4c
    18:     0x55a83de691ba - std::rt::lang_start::h93c151c94d8d3724
                                 at /home/rust/rust/library/std/src/rt.rs:165:17
    19:     0x55a83de70c1e - main
    20:     0x7f4b54fe0083 - __libc_start_main
                                 at /build/glibc-BHL3KM/glibc-2.31/csu/../csu/libc-start.c:308:16
    21:     0x55a83de6603e - _start
    22:                0x0 - <unknown>
```

Do you have any suggestions on how to debug?

The other thing I should mention is that the lvgl build.rs does two builds - one that builds for the target = x86_64-unknown-linux-gnu and another for target = xtensa-esp32s3-espidf.  which is kinda weird.  Both error with string.h not found.

enelson1001, 2023-11-07:
This is partial listing of log file run with the command :```cargo build -j1 -vvv -p lvgl &>log3.txt```.  It may give you some clues?
I did search on lv_mem.o to find all occurrences. 

Looks like build for x86_64-unknown-linux-gnu ???????

If I leave out -j1 then I see both x86_64-unknown-linux-gnu and xtensa-esps32s3-espidf
```
==========================================
Line starts at 441
==========================================
[lvgl-sys 0.6.2] ************** The lvgl-sys starting build ********************
[lvgl-sys 0.6.2] cargo:rerun-if-changed=/home/ed/esp-rust-projects/esp32-s3/hello-tft/include/lv_conf.h
[lvgl-sys 0.6.2] cargo:rerun-if-changed=/home/ed/esp-rust-projects/esp32-s3/hello-tft/include/lv_drv_conf.h
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=LVGL_INCLUDE
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=LVGL_LINK
[lvgl-sys 0.6.2] TARGET = Some("x86_64-unknown-linux-gnu")
[lvgl-sys 0.6.2] OPT_LEVEL = Some("0")
[lvgl-sys 0.6.2] HOST = Some("x86_64-unknown-linux-gnu")
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=CC_x86_64-unknown-linux-gnu
[lvgl-sys 0.6.2] CC_x86_64-unknown-linux-gnu = None
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=CC_x86_64_unknown_linux_gnu
[lvgl-sys 0.6.2] CC_x86_64_unknown_linux_gnu = None
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=HOST_CC
[lvgl-sys 0.6.2] HOST_CC = None
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=CC
[lvgl-sys 0.6.2] CC = None
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
[lvgl-sys 0.6.2] CRATE_CC_NO_DEFAULTS = None
[lvgl-sys 0.6.2] DEBUG = Some("true")
[lvgl-sys 0.6.2] CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=CFLAGS_x86_64-unknown-linux-gnu
[lvgl-sys 0.6.2] CFLAGS_x86_64-unknown-linux-gnu = None
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=CFLAGS_x86_64_unknown_linux_gnu
[lvgl-sys 0.6.2] CFLAGS_x86_64_unknown_linux_gnu = None
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=HOST_CFLAGS
[lvgl-sys 0.6.2] HOST_CFLAGS = None
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=CFLAGS
[lvgl-sys 0.6.2] CFLAGS = None
[lvgl-sys 0.6.2] running: "cc" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-gdwarf-4" "-fno-omit-frame-pointer" "-m64" "-I" "/home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/vendor/lvgl/src" "-I" "/home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/vendor" "-I" "/home/ed/esp-rust-projects/esp32-s3/hello-tft/include" "-I" "/home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/vendor/lv_drivers" "-I" "/usr/include" "-I" "/usr/local/include" "-DLV_CONF_INCLUDE_SIMPLE=1" "-o" "/home/ed/esp-rust-projects/esp32-s3/hello-tft/target/debug/build/lvgl-sys-d72660327e743ef4/out/51cccdca01579f47-lv_lru.o" "-c" "/home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/vendor/lvgl/src/misc/lv_lru.c"
[lvgl-sys 0.6.2] exit status: 0
========================================
Line ends at 470
========================================

==========================================
Line starts at 503
==========================================
[lvgl-sys 0.6.2] running: "cc" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-gdwarf-4" "-fno-omit-frame-pointer" "-m64" "-I" "/home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/vendor/lvgl/src" "-I" "/home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/vendor" "-I" "/home/ed/esp-rust-projects/esp32-s3/hello-tft/include" "-I" "/home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/vendor/lv_drivers" "-I" "/usr/include" "-I" "/usr/local/include" "-DLV_CONF_INCLUDE_SIMPLE=1" "-o" "/home/ed/esp-rust-projects/esp32-s3/hello-tft/target/debug/build/lvgl-sys-d72660327e743ef4/out/51cccdca01579f47-lv_mem.o" "-c" "/home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/vendor/lvgl/src/misc/lv_mem.c"
[lvgl-sys 0.6.2] exit status: 0
========================================
Line ends at 504
========================================

==========================================
Line starts at 890
==========================================
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=AR_x86_64-unknown-linux-gnu
[lvgl-sys 0.6.2] AR_x86_64-unknown-linux-gnu = None
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=AR_x86_64_unknown_linux_gnu
[lvgl-sys 0.6.2] AR_x86_64_unknown_linux_gnu = None
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=HOST_AR
[lvgl-sys 0.6.2] HOST_AR = None
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=AR
[lvgl-sys 0.6.2] AR = None
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=ARFLAGS_x86_64-unknown-linux-gnu
[lvgl-sys 0.6.2] ARFLAGS_x86_64-unknown-linux-gnu = None
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=ARFLAGS_x86_64_unknown_linux_gnu
[lvgl-sys 0.6.2] ARFLAGS_x86_64_unknown_linux_gnu = None
[lvgl-sys 0.6.2] cargo:rerun-if-env-changed=HOST_ARFLAGS
[lvgl-sys 0.6.2] HOST_ARFLAGS = None
[... 137 lines cut ...]
     9:     0x557dc8082b32 - rust_begin_unwind
    10:     0x557dc787c7b3 - core::panicking::panic_fmt::he0893255bd367bf2
    11:     0x557dc787d003 - core::result::unwrap_failed::h458e1c6bbe8e82c7
    12:     0x557dc7883d52 - core::result::Result<T,E>::expect::h06a112e23e3bd6ed
                                 at /home/rust/rust/library/core/src/result.rs:1034:23
    13:     0x557dc7886dfe - build_script_build::main::h42d6494bec9bf86b
                                 at /home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/build.rs:304:20
    14:     0x557dc787d28b - core::ops::function::FnOnce::call_once::hd5673faacdaca33e
                                 at /home/rust/rust/library/core/src/ops/function.rs:250:5
[... cut ...]
    18:     0x557dc788019a - std::rt::lang_start::h93c151c94d8d3724
                                 at /home/rust/rust/library/std/src/rt.rs:165:17
    19:     0x557dc78876be - main
    20:     0x7f91a1247083 - __libc_start_main
                                 at /build/glibc-BHL3KM/glibc-2.31/csu/../csu/libc-start.c:308:16
    21:     0x557dc787d03e - _start
    22:                0x0 - <unknown>
==========================================
Line ends at 1467 end of document
==========================================
```

enelson1001, 2023-11-08:
I was able to get lvgl to build on my ESP32S3 device without errors.

I used the following command to build lvgl so I could look for errors
```
cargo build -j1 -vvv -p lvgl &>log3.txt
```

My first error was not finding 'bits/libc-header-start.h' file 
```
--- stderr
  /usr/include/stdint.h:26:10: fatal error: 'bits/libc-header-start.h' file not found
  thread 'main' panicked at /home/ed/.cargo/registry/src/index.crates.io-6f17d22bba15001f/lvgl-sys-0.6.2/build.rs:236:10:
  Unable to generate bindings: ClangDiagnostic("/usr/include/stdint.h:26:10: fatal error: 'bits/libc-header-start.h' file not found\n")
```
This error was solved by doing the following:
```
$ sudo apt update
$ sudo apt upgrade
$ sudo apt-get install gcc-multilib
```

My second error was not finding string.h file
```
[lvgl-sys 0.6.2] exit status: 0
[lvgl-sys 0.6.2] running: "ar" "s" "/home/ed/esp-rust-projects/esp32-s3/hello-tft/target/debug/build/lvgl-sys-d72660327e743ef4/out/liblvgl.a"
[lvgl-sys 0.6.2] exit status: 0
[lvgl-sys 0.6.2] cargo:rustc-link-lib=static=lvgl
[lvgl-sys 0.6.2] cargo:rustc-link-search=native=/home/ed/esp-rust-projects/esp32-s3/hello-tft/target/debug/build/lvgl-sys-d72660327e743ef4/out
[lvgl-sys 0.6.2] /home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/vendor/lvgl/src/misc/lv_mem.h:20:10: fatal error: 'string.h' file not found
[lvgl-sys 0.6.2] thread 'main' panicked at /home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/build.rs:313:10:
[lvgl-sys 0.6.2] Unable to generate bindings: ClangDiagnostic("/home/ed/lv-binding-rust/lv_binding_rust/lvgl-sys/vendor/lvgl/src/misc/lv_mem.h:20:10: fatal error: 'string.h' file not found\n")
```
This was solved by editing the config.toml file to add additional items to rustflags.  Since I was editing the file I added DEP_LV_CONFIG_PATH environmental variable setting
```
[build]
target = "xtensa-esp32s3-espidf"

[target.xtensa-esp32s3-espidf]
linker = "ldproxy"
# runner = "espflash --monitor" # Select this runner for espflash v1.x.x
runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x

rustflags = [
    # Extending time_t for ESP IDF 5: https://github.com/esp-rs/rust/issues/110
    "--cfg",
    "espidf_time64",
    # Added the following 2 entries so lvgl will build without errors
    "--sysroot",
    "/home/ed/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include",
] 

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU = "esp32s3"

# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.1.1"

# The directory that has the lvgl config files - lv_conf.h, lv_drv_conf.h
DEP_LV_CONFIG_PATH = { relative = true, value = "lvgl-configs" }
```

The last error I got was a ctor error. I updated lvgl in Cargo.toml to look like the following.
```
lvgl = { version = "0.6.2", features = ["rust_timer", "unsafe_no_autoinit", ]}
```

closed 2023-11-08 by enelson1001

SergioGasquez, 2023-11-08:
Thanks for sharing your debug process!