esp-rs/espup #388: espup installs potential riscv targets that are never used

Vollbrecht, 2023-11-06:
## Bug description
If i understand it correctly espup just reuse the build-artifacts from [crosstool](https://github.com/espressif/crosstool-NG). Currently it installs gcc libs for the following 4 targets in the case of the riscv eco-system.
- rv32i_zicsr_zifencei
- rv32imac_zicsr_zifencei
- v32imafc_zicsr_zifencei
- rv32imc_zicsr_zifencei
This libs are relative huge, and if never used should not be installed. 

## To Reproduce

Steps to reproduce the behavior:

install esp toolchain with espup

## Expected behavior

Only install tools that eventually will be used

## Screenshots
[image]

## Proposal

If its true that we only use the imc and imac targets i would want that the installer directly delete the other  for me ( simple option) or that croostool only produces that one that espressif uses.

SergioGasquez, 2023-11-06:
This would slim-down what `espup` install, but it might not be worth to invest time into something that should be temporary, as soon as we can use LLD as linker for no_std, `espup` won't install GCC at all.

We can leave the issue open for the moment, in case LLD supports takes too long to land, but hopefully it won't take too much!

Vollbrecht, 2023-11-08:
also now less relevant duo to: #391