esp-rs/espup #404: Install python3 and python3-venv.

r12f, 2023-12-10:
## Motivations

When following the [esp-idf-template](https://github.com/esp-rs/esp-idf-template) to setup the example repo, the new repo might fail to be build, because ESP-IDF requires python3-venv to be installed. 

However, installing the step of installing python-venv is not listed in either esp-idf-template or [Rust on ESP book](https://esp-rs.github.io/book/installation/index.html). And it can be easily missed.

- Would you like to implement this feature? [n]

## Solution

I feel making espup installing python and python-venv as part of frequently used dependency might make sense and will be helpful. 

## Alternatives

Updating both Rust on ESP book and esp-idf-template README should also work.

## Additional context

SergioGasquez, 2023-12-11:
I feel like the alternative solution is more clean, since:
1. It's way simpler to just suggest the user to install them, we do not need to worry of which OS is he running and if he has any conflicts with those packages
2. `espup` does not install ESP-IDF and has nothing to do with it, `no_std` user do not require python and I would like to keep it this way.

I agree that book and `esp-idf-template` can be updated to include them, though.

r12f, 2023-12-11:
Hi Sergio, makes sense! And thanks a lot for checking! I believe the issue can be closed now.

SergioGasquez, 2023-12-11:
Just noticed that the book already includes those, in the last line of https://esp-rs.github.io/book/installation/std-requirements.html

closed 2023-12-11 by SergioGasquez