esp-rs/espup #410: Installation duplicates contents of PATH environment variable on Windows

marsfan, 2024-01-10:
## Bug description

When using `espup` on Windows, the tool messes up user PATH environment variable in two ways.
1. It duplicates all existing entries in the user PATH environment variable
2. It adds all entries in the system PATH environment variable to the user PATH environment variables 

To help clear things up, the following is a bare-bones example of what I am seeing. 

Before running `espup` I have the following configuration

#### User PATH varriable
```
C:\path1
C:\path2
```

####  System PATH variable
```
C:\WINDOWS
```

After running, the system PATH environment variable remains unchanged, but the user PATH environment variable is as follows:
```
C:\WINDOWS
C:\path1
C:\path2
C:\users\ME\.rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-clang\bin
C:\users\ME\.rustup\toolchains\esp\xtensa-esp-elf\bin
C:\path1
C:\path2
```

As you can see, the system PATH members are prepended to the PATH members, and then the user PATH members are duplicated.

- Would you like to work on a fix? [y/n]

## To Reproduce

Steps to reproduce the behavior:

1. Use espup to install the toolchain
2. View the list of defined user environment variables in Windows. 

## Expected behavior

Only the additional paths are added to the user PATH environment variable necessary for compiling Rust programs for the ESP32 have been added.

## Screenshots

## Environment

- OS: Windows 10 22H2
- espup version: 0.10.0

## Additional context

SergioGasquez, 2024-01-10:
I see, it looks like the `std::env::var`  mixes both user and system `PATH` environment variable. I did some testing and the following seems to fix the issue, could you test and verify that the issue is solved:

```
cargo uninstall espup
cargo install espup --git https://github.com/SergioGasquez/espup --branch fix/windows-path
```

marsfan, 2024-01-11:
@SergioGasquez That seems to have fixed it.

SergioGasquez, 2024-01-11:
Thanks for testing the issue! I will probably cut a release before the next Xtensa Rust release

closed 2024-01-11 by SergioGasquez

marsfan, 2024-01-12:
Awesome. I'll have to give it a go then. I've been having a few different issues with getting the ESP32 working with Windows that lead to a lot of uninstalls and re-installs. Having the tool fix the path automatically will help me with that.

marsfan, 2024-01-12:
@SergioGasquez This is a bit off-topic, but you mentioned "Before the next Xtensa Rust release" Do you have a recommendation on the best way to keep track of when those happen, so I know when I need to run `espup` to get the new version of the toolchain?

SergioGasquez, 2024-01-12:
> @SergioGasquez This is a bit off-topic, but you mentioned "Before the next Xtensa Rust release" Do you have a recommendation on the best way to keep track of when those happen, so I know when I need to run `espup` to get the new version of the toolchain? 

Usually, the schedule is very similar to the stable channel (see https://forge.rust-lang.org/#current-release-versions for dates), but we may have some delay since we sometimes give a few days before marking the release as `latest` on gh. Releases are published in https://github.com/esp-rs/rust-build/releases so you can enable notifications for it or just join the matrix channel (https://matrix.to/#/#esp-rs:matrix.org) where we notify when the new version is published.

marsfan, 2024-01-13:
Awesome. I enabled release notifications. Thanks