esp-rs/espup #412: Toolchain installed by espup does not work under Ubuntu 20.04, GLIBC version error

jobafr, 2024-01-22:
## Bug description

After using `espup install` under Ubuntu 20.04 (aarch64, NVidia Jetson AGX Xavier), running the cargo binary produces a linker error.

```
$ ~/.rustup/toolchains/esp/bin/cargo
/home/me/.rustup/toolchains/esp/bin/cargo: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/me/.rustup/toolchains/esp/bin/cargo)
/home/me/.rustup/toolchains/esp/bin/cargo: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/me/.rustup/toolchains/esp/bin/cargo)
/home/me/.rustup/toolchains/esp/bin/cargo: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /home/me/.rustup/toolchains/esp/bin/cargo)
```

- Would you like to work on a fix? [y/n]

n

## To Reproduce

Steps to reproduce the behavior:

1. run `espup install` on Ubuntu 20.04 aarch64
2. try to run cargo

## Expected behavior

Cargo works and compiles my project

## Environment

```
$ cat /etc/issue
Ubuntu 20.04.6 LTS \n \l

$ uname -a
Linux aql57 5.10.104-tegra #1 SMP PREEMPT Wed Aug 10 20:17:07 PDT 2022 aarch64 aarch64 aarch64 GNU/Linux

$ espup --version
espup 0.10.0
```

## Additional context

This looks similar to [this issue in espflash](https://github.com/esp-rs/espflash/issues/454) from last year.

SergioGasquez, 2024-01-22:
Hi! This does not really belong to `espup` repo, as the artifacts which contain the Xtensa Rust are produced in `rust-build`, so I'll move the issue there

SergioGasquez, 2024-01-22:
We [moved our builders to ubuntu 22.04](https://github.com/esp-rs/rust-build/pull/251/commits/3bc91a5c3fb6c2c76f9ffd9cc88e15f1cff5ddc6) in latest release (1.75), hence this error when using ubuntu 20.04