esp-rs/espup #440: Build failed due to failure of loading libclang.dll when generating bindings

Sydonian, 2024-07-25:
building logs:
```
   Compiling byteorder v1.5.0
   Compiling nb v1.1.0
   Compiling embedded-io v0.6.1
   Compiling stable_deref_trait v1.2.0
   Compiling embedded-hal v1.0.0
   Compiling futures-core v0.3.30
   Compiling futures-task v0.3.30
   Compiling pin-utils v0.1.0
   Compiling esp-idf-sys v0.35.0
   Compiling pin-project-lite v0.2.14
   Compiling enumset v1.1.3
   Compiling libc v0.2.155
   Compiling futures-util v0.3.30
   Compiling nb v0.1.3
   Compiling const_format v0.2.32
   Compiling embedded-io-async v0.6.1
   Compiling hash32 v0.3.1
   Compiling void v1.0.2
   Compiling critical-section v1.1.2
   Compiling cfg-if v1.0.0
   Compiling num_enum v0.7.2
   Compiling heapless v0.8.0
   Compiling embedded-hal v0.2.7
   Compiling embedded-hal-nb v1.0.0
   Compiling embedded-hal-async v1.0.0
   Compiling embedded-can v0.4.1
   Compiling serde v1.0.204
   Compiling atomic-waker v1.1.2
   Compiling log v0.4.22
   Compiling uncased v0.9.10
   Compiling embassy-time-driver v0.1.0
   Compiling embassy-futures v0.1.1
   Compiling embassy-sync v0.6.0
   Compiling embedded-svc v0.28.0
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.35.0: (esp-idf-sys) Too long output directory: `\\?\E:\ESP\projects\testlcd\target\xtensa-esp32-espidf\debug\build\esp-idf-sys-3ae04a86ffd25549\out`. Shorten your project path down to no more than 10 characters (or use WSL2 and its native Linux filesystem). Note that tricks like Windows `subst` do NOT work!
warning: esp-idf-sys@0.35.0: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `cc` installed? (see https://docs.rs/cc/latest/cc/#compile-time-requirements for help)
warning: esp-idf-sys@0.35.0: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `c++` installed? (see https://docs.rs/cc/latest/cc/#compile-time-requirements for help)
warning: esp-idf-sys@0.35.0: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `cc` installed? (see https://docs.rs/cc/latest/cc/#compile-time-requirements for help)

error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `E:\ESP\projects\testlcd\target\debug\build\esp-idf-sys-d7ae078c1bb86ffe\build-script-build` (exit code: 101)
  --- stdout
  cargo:warning=(esp-idf-sys) Too long output directory: `\\?\E:\ESP\projects\testlcd\target\xtensa-esp32-espidf\debug\build\esp-idf-sys-3ae04a86ffd25549\out`. Shorten your project path down to no more than 10 characters (or use WSL2 and its native Linux filesystem). Note that tricks like Windows `subst` do NOT work!
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  Python 3.12.4
  pip 24.1.2 from E:\ESP\projects\testlcd\.embuild\espressif\python_env\idf5.2_py3.12_env\Lib\site-packages\pip (python 3.12)
  Requirement already satisfied: pip in e:\esp\projects\testlcd\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (24.1.2)
  Requirement already satisfied: setuptools in e:\esp\projects\testlcd\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (71.0.0)
  Collecting setuptools
    Using cached setuptools-71.0.4-py3-none-any.whl.metadata (6.5 kB)
  Using cached setuptools-71.0.4-py3-none-any.whl (2.3 MB)
  Installing collected packages: setuptools
    Attempting uninstall: setuptools
[... 152 lines cut ...]
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.2.2",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v5.2.2") }
  Using esp-idf v5.2.2 at 'E:\ESP\projects\testlcd\.embuild\espressif\esp-idf\v5.2.2'
  Built components: xtensa, esp_ringbuf, efuse, esp_mm, driver, esp_pm, mbedtls, esp_bootloader_format, esp_app_format, bootloader_support, bootloader, esptool_py, partition_table, esp_partition, app_update, spi_flash, pthread, esp_system, esp_rom, hal, log, heap, soc, esp_hw_support, freertos, newlib, cxx, esp_common, esp_timer, app_trace, esp_event, nvs_flash, esp_phy, vfs, lwip, esp_netif_stack, esp_netif, wpa_supplicant, esp_coex, esp_wifi, bt, unity, cmock, console, http_parser, esp-tls, esp_adc, esp_eth, esp_gdbstub, esp_hid, tcp_transport, esp_http_client, esp_http_server, esp_https_ota, esp_https_server, esp_psram, esp_lcd, protobuf-c, protocomm, esp_local_ctrl, espcoredump, wear_levelling, sdmmc, fatfs, idf_test, ieee802154, json, mqtt, nvs_sec_provider, openthread, perfmon, spiffs, ulp, usb, wifi_provisioning, main
  thread 'main' panicked at C:\Users\79nos\.cargo\registry\src\mirrors.tuna.tsinghua.edu.cn-df7c3c540f42cdbd\bindgen-0.69.4\lib.rs:622:31:
  Unable to find libclang: "the `libclang` shared library at C:\\Users\\79nos\\.rustup\\toolchains\\esp\\xtensa-esp32-elf-clang\\esp-clang\\bin\\libclang.dll could not be opened: LoadLibraryExW failed"
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

and I have set the LIBCLANG_PATH env:
[image]

taks, 2024-07-26:
The same error occurred in my environment.
I was able to compile successfully by replacing LLVM17 files with LLVM16 files in `.rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-clang` directory.

SergioGasquez, 2024-07-29:
> The same error occurred in my environment. I was able to compile successfully by replacing LLVM17 files with LLVM16 files in `.rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-clang` directory.

Out of curiosity, what happens if you only replace the `libclang.dll` from LLVM 17 for the on in LLVM 16?

taks, 2024-07-30:
> Out of curiosity, what happens if you only replace the libclang.dll from LLVM 17 for the on in LLVM 16?

If only `libclang.dll` is replaced, the same error occurs.
(This is because there is no dependency dll.)

I tried to get the cause of the load error with the `GetLastError` function.
The error code when loading the LLVM17 dll was 193 (ERROR_BAD_EXE_FORMAT: %1 is not a valid Win32 application.)

Incidentally, the error code when replacing only `libclang.dll` was 126 (ERROR_MOD_NOT_FOUND: The specified module could not be found.)

SergioGasquez, 2024-07-30:
Thanks for the detailed information, at the moment I don't have a Windows environment to reproduce the error and Im not sure what we can do from the `espup` side (but removing LLVM 17 and just using LLVM 16 for Windows). I would like to ask a few extra tests if you dont mind:
- Have you tried using the  `-e, --extended-llvm` flag?
- Could you try using the [following LLVM 17 release](https://github.com/espressif/llvm-project/releases/tag/esp-17.0.1_20240408), we are using [the latest one](https://github.com/espressif/llvm-project/releases/tag/esp-17.0.1_20240419) in `espup`
  - To use this release, just download the Windows asset, unzip it and replace the files in the folder like you did for LLVM 16.

taks, 2024-07-30:
I found the cause of the error.

When loading `libclang.dll` (LLVM17), the dependency `libstdc++-6.dll` is also loaded.
However, instead of `libstdc++-6.dll`, which is bundled with `libclang.dll`, the one in `embuild\espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin` is loaded.
These `libstdc++-6.dll` are different and this error is caused.

(the `libstdc++-6.dll` is not bundled in LLVM16)

> - Have you tried using the -e, --extended-llvm flag?
> - Could you try using the following LLVM 17 release, we are using the latest one in espup

Both do not work well.

KiritoMC03, 2024-09-01:
Hi !
I had the same problem.
Solved by replacing the contents of
> ".rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-clang\bin"
and
> ".rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-clang\lib" 
with the contents of the same folders from https://github.com/espressif/llvm-project/releases/tag/esp-16.0.4-20231113
> and renaming of the file **".rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-17.0.1_20240419"** to **"esp-16.0.4-20231113"**

Build logs (cargo clean -> cargo build)
[no_backtrace_logs.txt](https://github.com/user-attachments/files/16828232/no_backtrace_logs.txt)
[backtrace_full_logs.txt](https://github.com/user-attachments/files/16828243/backtrace_full_logs.txt)

Project config files:
[config.toml.txt](https://github.com/user-attachments/files/16828274/config.toml.txt)
[Cargo.toml.txt](https://github.com/user-attachments/files/16828278/Cargo.toml.txt)

Environment:
- Windows 11 23H2 x64
- rustc 1.80.0-nightly (1b259e0b8 2024-07-22) (1.80.0.0)
- espup 0.12.2
- python 3.12.5
- std project for esp32 (xtensa-esp32-espidf), generated from https://github.com/esp-rs/esp-idf-template

licsber, 2024-09-25:
Same above.
Replace `~/.rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-clang` with
 `libs_llvm-esp-16.0.4-20231113-win64.tar.xz` works for me.

pai2shub, 2025-08-08:
> espup

是的，我使用的就是 0.15.1 版本的。不幸的是这个错误依然存在

```bash
PS D:\ett1> espup  --version
espup 0.15.1
```

为了解决这个问题，需要手动下载  https://github.com/espressif/llvm-project/releases/tag/esp-16.0.4-20231113 中的 libs-llvm-esp 并解压到 .rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-clang 进行替换

SergioGasquez, 2025-08-08:
Why you want to install that LLVM version? `espup` now installs LLVM 19. LLVM 16 was used before Rust 1.81.0

pai2shub, 2025-08-11:
> Why you want to install that LLVM version? `espup` now installs LLVM 19. LLVM 16 was used before Rust 1.81.0为什么要安装该 LLVM 版本？`espup` 现在安装 LLVM 19。LLVM 16 在 Rust 1.81.0 之前使用

Perhaps it is due to some special issues with LLVM in windows

mcitem, 2025-09-19:
> > Perhaps it is due to some special issues with LLVM in windows
> 
> Im not sure Im following your reply, Windows shouldn't be any different regarding LLVM versions

Are there any solutions to this issue? (LoadLibraryExW)

mcitem, 2025-09-24:
> Can you try using the latest version (of the toolchain and of espup):
> 
> ```
> rustup update
> cargo install espup --locked
> espup install
> ```

I’ve just ran these commands and regenerated the project. Now it no longer reports errors and can run normally:

```sh
rustup self update
rustup update
cargo install espup --locked
cargo install cargo-generate --locked
espup install
cargo generate esp-rs/esp-idf-template cargo
```

Thanks! I can finally put WSL2 aside for now.

closed 2025-09-25 by MabezDev

terk3n, 2025-12-19:
Sorry for necroing this issue, just to make some annotations because this issue shows up in Google search regarding libclang.dll broken search order. Set LIBCLANG_PATH environment variable where libclang and libLLVM-*.dll is, make sure they're the x64 version of the binaries, some people reported having mixed arch libraries in their installations, also add them to your PATH env var, try to put it in first place so Windows loader picks up the one you want to use.