esp-rs/espup #442: M1 toolchain setup

adrien3d, 2024-08-09:
On my M1 Mac, since I was unable to compile my project, having the following error:

> error: failed to run custom build command for `esp-idf-sys v0.35.0`

> 
>   Using activated esp-idf v5.2.2 environment at '/Users/adrien/Dev/esp/esp-idf'
>   CMake Error at /Users/adrien/Dev/esp/esp-idf/tools/cmake/tool_version_check.cmake:36 (message):
>     
> 
>     Tool doesn't match supported version from list ['esp-13.2.0_20230928']:
>     /Users/adrien/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc

I then launched an `espup install` and then an `espup update`

Now I have this error:
> error: "/Users/adrien/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/src/rust/Cargo.lock" does not exist, unable to build with the standard library, try:
        rustup component add rust-src --toolchain nightly-x86_64-apple-darwin

I launched the recommended command, but I still have the same error (with and without launching the `export-esp.sh` script that have the following content:

> 
> export PATH="/Users/adrien/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:$PATH"
> export LIBCLANG_PATH="/Users/adrien/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-17.0.1_20240419/esp-clang/lib"

SergioGasquez, 2024-08-09:
Hi! What chip are you targeting? `espup` can install the environment for RISC-V targets, but is not needed. I would recommend doing the following:
- Install the RISC-V toolchain: either via `espup` or `rustup`. See https://docs.esp-rs.org/book/installation/riscv.html
- Do not use an activated environment at start, let `esp-idf-sys` download the correct esp-idf that your project is configured to use.
  - Not sure what project are you trying to build, but you might have the wrong esp-idf version activated.
  
Not sure if you already did, but I recommend reading https://docs.esp-rs.org/book/installation/index.html

adrien3d, 2024-08-09:
I am targeting a C6.
I've tried what you recommended, but I still have the same error.
My project is open-source: https://github.com/adrien3d/iss/blob/feat/ntp%26config/Cargo.toml
But basically, all latest commits on each branch (master, feat/ntp & feat/ntp&config) are not compiling on both my Mac and an Ubuntu computer, giving always the same error.

May be the issue is in the rust-toolchain.toml that is a nightly one, and not a nightly fixed at a specific day?

SergioGasquez, 2024-08-09:
Just tried with my M1, and I was able to successfully build your project `master` branch. I assume you have already `nightly` channel installed, either by `espup` or by `rustup toolchain install nightly --component rust-src` 

> May be the issue is in the rust-toolchain.toml that is a nightly one, and not a nightly fixed at a specific day? 

That should not be the issue. By the way, you could also use `stable` now.

Can you share the error? Please, make sure that you don't have any ESP-IDF activated.

adrien3d, 2024-08-09:
Yep, same as above.
When I build using `nightly`, I have:                                          
> error: "/home/achapelet/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/Cargo.lock" does not exist, unable to build with the standard library, try:
>         rustup component add rust-src --toolchain nightly-x86_64-unknown-linux-gnu

When I build using `stable`, I have:

> Compiling autocfg v1.3.0
> error[E0463]: can't find crate for `core`
>   |
>   = note: the `riscv32imac-esp-espidf` target may not be installed
>   = help: consider downloading the target with `rustup target add riscv32imac-esp-espidf`
> 
> For more information about this error, try `rustc --explain E0463`.
> error: could not compile `nb` (lib) due to 1 previous error
> warning: build failed, waiting for other jobs to finish...

and then:
> rustup target add riscv32imac-esp-espidf
> error: toolchain 'stable-x86_64-unknown-linux-gnu' does not support target 'riscv32imac-esp-espidf'
> note: you can see a list of supported targets with `rustc --print=target-list`
> note: if you are adding support for a new target to rustc itself, see https://rustc-dev-guide.rust-lang.org/building/new-target.html

SergioGasquez, 2024-08-09:
But if you run `rustup component add rust-src --toolchain nightly-x86_64-unknown-linux-gnu` you get any errors?

adrien3d, 2024-08-10:
No, just `info: component 'rust-src' is up to date`, same on my mac :(

SergioGasquez, 2024-08-10:
Did you installed Rust using any package manager? I would recommend uninstalling your Rust and installing it again with: 
```
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
rustup toolchain install nightly --component rust-src
```

adrien3d, 2024-08-10:
It was already installed with rustup but I reinstalled it as you asked. Now when doing a build I have:
```
   Compiling esp-idf-sys v0.35.0
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.35.0: Ignoring configuration setting `ESP_IDF_VERSION` (Tag v5.2.2): custom esp-idf repository detected via $IDF_PATH

error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `/Users/adrien/Dev/esp/projects/rust/iss/target/debug/build/esp-idf-sys-a1d7fc177259ff4e/build-script-build` (exit status: 101)
  ....
....
....
 --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: Some(
          "esp32c6",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.2.2",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "/Users/adrien/Dev/esp/esp-idf",
          ),
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using custom user-supplied esp-idf repository at '/Users/adrien/Dev/esp/esp-idf' (detected from env variable `IDF_PATH`)
  WARNING: Ignoring invalid distribution -etuptools (/Users/adrien/Dev/esp/projects/rust/iss/.embuild/espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages)
  WARNING: Ignoring invalid distribution -etuptools (/Users/adrien/Dev/esp/projects/rust/iss/.embuild/espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages)
  WARNING: Ignoring invalid distribution -etuptools (/Users/adrien/Dev/esp/projects/rust/iss/.embuild/espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages)
  WARNING: Ignoring invalid distribution -etuptools (/Users/adrien/Dev/esp/projects/rust/iss/.embuild/espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages)
  WARNING: Ignoring invalid distribution -etuptools (/Users/adrien/Dev/esp/projects/rust/iss/.embuild/espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages)
  WARNING: Ignoring invalid distribution -etuptools (/Users/adrien/Dev/esp/projects/rust/iss/.embuild/espressif/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages)
  Using esp-idf v5.2.2 at '/Users/adrien/Dev/esp/esp-idf'
  Built components: riscv, esp_ringbuf, efuse, esp_mm, driver, esp_pm, mbedtls, esp_bootloader_format, esp_app_format, bootloader_support, bootloader, esptool_py, partition_table, esp_partition, app_update, spi_flash, pthread, esp_system, esp_rom, hal, log, heap, soc, esp_hw_support, freertos, newlib, cxx, esp_common, esp_timer, app_trace, esp_event, nvs_flash, esp_phy, vfs, lwip, esp_netif_stack, esp_netif, wpa_supplicant, esp_coex, esp_wifi, bt, unity, cmock, console, http_parser, esp-tls, esp_adc, esp_eth, esp_gdbstub, esp_hid, tcp_transport, esp_http_client, esp_http_server, esp_https_ota, esp_https_server, esp_psram, esp_lcd, protobuf-c, protocomm, esp_local_ctrl, espcoredump, wear_levelling, sdmmc, fatfs, idf_test, ieee802154, json, mqtt, nvs_sec_provider, openthread, spiffs, ulp, usb, wifi_provisioning, main
  thread 'main' panicked at /Users/adrien/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.69.4/lib.rs:622:31:
  Unable to find libclang: "couldn't find any valid shared libraries matching: ['libclang.dylib'], set the `LIBCLANG_PATH` environment variable to a path where one of these files can be found (invalid: [])"
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

SergioGasquez, 2024-08-10:
Seems like you are using an activated ESP-IDF environment `custom esp-idf repository detected via $IDF_PATH`, please unset `IDF_PATH` and let `esp-idf-sys` download it

adrien3d, 2024-08-10:
Now fixed with what you asked, thank you very much for your reactive support.

closed 2024-08-10 by adrien3d