esp-rs/espup #444: Missing core from xtensa-esp32-none-elf

IsCoffeeTho, 2024-08-28:
## Bug description

Upon a fresh install from cargo (`cargo install espup`): `xtensa-esp32-none-elf` has no `core` crate
As well as attempting to remedy the 'may not be installed' message: rust errors saying `toolchain 'esp' does not support components`

## To Reproduce

Steps to reproduce the behavior:

1. Remove previous installation of espup (if applicable)
2. Remove `~/.rustup/toolchains/esp` (if applicable)
3. `$ cargo install espup`
4. Navigate to a clean embedded project
5. `$ cargo +esp build`

## Expected behavior

Clean build of project

## Screenshots

![error[E0463]: can't find crate for `core`](https://github.com/user-attachments/assets/ed110a5c-84de-4e86-974b-d109dd6c0982)

[image]

## Environment

- OS: Arch Linux x86_64 (Linux 6.10.5-arch1-1)
- espup version: 0.12.2

IsCoffeeTho, 2024-08-30:
The previous comment has told me nothing about continuing,

- Did this reproduction consider 'Operating System' and 'Kernel'?
- What are some more steps to continue with resolving this issue?
- Is there more information needed, or am i just stuck in the mud?

Any information and/or questions more than "it works on my machine" would be appreciated

SergioGasquez, 2024-09-02:
> Did this reproduction consider 'Operating System' and 'Kernel'?

In the dev container uses [`bookworm-slim`](https://hub.docker.com/layers/library/debian/bookworm-slim/images/sha256-70d4c04302bdcd71c4fa21b6c12e99888380a07f04e3d44452b961bca046489d?context=explore)

> What are some more steps to continue with resolving this issue?

Have you tried doing `espup install` and sourcing the `export-esp.sh` file? Your reproduction steps do not include this steps.

> Is there more information needed, or am i just stuck in the mud?

I think you provided all the required information, but there is something wrong in the steps. FYI, I use `6.10.6-arch1-1` and `espup 0.12.2` on my local machine and no issues encountered.

SergioGasquez, 2025-01-09:
> I can compile project on windows but i can't list targets.

Try using `rustc +esp --print target-list `

SergioGasquez, 2025-04-30:
Closing due to inactivity, if the issue persists feel free to reopen the issue!

closed 2025-04-30 by SergioGasquez

SergioGasquez, 2026-02-16:
> i'm also getting similar issues

Have you installed your toolchain? See https://docs.espressif.com/projects/rust/book/getting-started/toolchain.html#xtensa-devices