esp-rs/espup #447: `xtensa-esp32-none-elf` not installed with espup

nbaraz, 2024-10-05:
After runnint `espup install` on Windows 10, when I run `rustc +esp --print target-list` I only get the following xtensa targets:
```
xtensa-esp32-espidf
xtensa-esp32s2-espidf
xtensa-esp32s3-espidf
xtensa-esp32s3-none-elf
```

(`xtensa-esp32-none-elf` is missing)

SergioGasquez, 2024-10-07:
That's very strange, as `no_std` version of s3 is installed. Have you tried building a `no_std` project for ESP32 and see if you get any error? Also, might be worth to try `rustc +esp --print target-list | grep  "xtensa"`

nbaraz, 2024-10-07:
I don't have grep installed, and `find` was giving me issues, but I'll install it and try once I get home. I'm fairly certain it's not in the list though.

Is there an extra verbose espup flag/patch I can apply to gather more info, and any cache I need to clean after `espup uninstall`?

My first installation was with `-t esp32`, but I uninstalled and reinstalled with `-t all` after I didn't see the 8266 target installed.

nbaraz, 2024-10-07:
I'm really sorry, I meant `xtensa-esp8266-none-elf`. I must have been really tired when I opened the issue.

```
>rustc +esp --print target-list | find "xtensa"
xtensa-esp32-espidf
xtensa-esp32-none-elf
xtensa-esp32s2-espidf
xtensa-esp32s2-none-elf
xtensa-esp32s3-espidf
xtensa-esp32s3-none-elf
```

nbaraz, 2024-10-07:
The issue is that cargo-pio creates a project which relies on a non-existent target, I'll open an issue there.

closed 2024-10-07 by nbaraz

nbaraz, 2024-10-08:
Ok, `cargo-pio` was ok. It seems that support for `xtensa-esp8266-none-elf` was dropped in 1.81.
I opened an issue: https://github.com/esp-rs/rust/issues/237