esp-rs/espup #455: Instllation of 1.82.0.3 on M1 mac failed

georgik, 2024-11-01:
Command: `espup install`

Log:
```
[info]: Installing the Espressif Rust ecosystem
[info]: Checking Rust installation
[info]: Installing Xtensa Rust 1.82.0.3 toolchain
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[info]: Installing GCC (xtensa-esp-elf)
[info]: Installing Xtensa LLVM
[info]: Downloading 'rust.tar.xz'
[info]: Downloading 'xtensa-esp-elf.tar.xz'
[info]: Downloading 'idf_tool_xtensa_elf_clang.libs.tar.xz'
[info]: Creating symlink between '/Users/georgik/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/Users/georgik/.espup/esp-clang'
[warn]: Installation for 'LLVM' failed, retrying
[warn]: Previous installation of LLVM exists in: '/Users/georgik/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'. Reusing this installation
[info]: Creating symlink between '/Users/georgik/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/Users/georgik/.espup/esp-clang'
[warn]: Installation for 'LLVM' failed, retrying
[warn]: Previous installation of LLVM exists in: '/Users/georgik/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'. Reusing this installation
[info]: Creating symlink between '/Users/georgik/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/Users/georgik/.espup/esp-clang'
[warn]: Installation for 'LLVM' failed, retrying
[warn]: Previous installation of LLVM exists in: '/Users/georgik/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'. Reusing this installation
[info]: Creating symlink between '/Users/georgik/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/Users/georgik/.espup/esp-clang'
[warn]: Installation for 'LLVM' failed, retrying
thread 'tokio-runtime-worker' panicked at /Users/georgik/.cargo/registry/src/index.crates.io-6f17d22bba15001f/espup-0.13.0/src/toolchain/mod.rs:244:32:
called `Result::unwrap()` on an `Err` value: SendError { .. }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Error:   × File exists (os error 17)
```

Expected behavior: Display error message with suggestion on remedy, not a RUST_BACKTRACE.

SergioGasquez, 2024-11-04:
Just tried reproducing the issue but had no troubles: 

```
> rustc +esp --version
rustc 1.80.0-nightly (1b259e0b8 2024-07-22) (1.80.0.0)
> espup update
[info]: Updating the Espressif Rust ecosystem
[info]: Checking Rust installation
[warn]: Previous installation of LLVM exists in: '/Users/sergio/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'. Reusing this installation
[info]: Installing GCC (xtensa-esp-elf)
[warn]: Previous installation of GCC exists in: '/Users/sergio/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906'. Reusing this installation
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[info]: Creating symlink between '/Users/sergio/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/Users/sergio/.espup/esp-clang'
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.82.0.3 toolchain
[info]: Downloading 'rust.tar.xz'
[info]: Installing 'rust' component for Xtensa Rust toolchain
[info]: Downloading 'rust-src.tar.xz'
[info]: Installing 'rust-src' component for Xtensa Rust toolchain
[info]: Update successfully completed!

	To get started, you need to set up some environment variables by running: '. /Users/sergio/export-esp.sh'
	This step must be done every time you open a new terminal.
	    See other methods for setting the environment in https://esp-rs.github.io/book/installation/riscv-and-xtensa.html#3-set-up-the-environment-variables
```

After sourcing the new export file I was able to build Xtensa Rust projects

SergioGasquez, 2024-11-04:
Were you using latest espup (0.13.0)?

anjumkaiser, 2024-11-20:
having the same issue on macOS 15, M2

➜  ~ espup --version
espup 0.13.0
➜  ~ espup install
[info]: Installing the Espressif Rust ecosystem
[info]: Checking Rust installation
[info]: Installing Xtensa Rust 1.82.0.3 toolchain
[info]: Installing Xtensa LLVM
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[info]: Installing GCC (xtensa-esp-elf)
[info]: Downloading 'rust.tar.xz'
[info]: Downloading 'idf_tool_xtensa_elf_clang.libs.tar.xz'
[info]: Downloading 'xtensa-esp-elf.tar.xz'
[info]: Creating symlink between '/Users/anjumkaiser/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/Users/anjumkaiser/.espup/esp-clang'
[info]: Installing 'rust' component for Xtensa Rust toolchain
[info]: Downloading 'rust-src.tar.xz'
[warn]: Installation for 'RISC-V Rust target' failed, retrying
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying
Error: espup::toolchain::rust::install_riscv_target

  × Failed to Install RISC-V targets for 'nightly' toolchain

SergioGasquez, 2024-11-20:
> Installation for 'RISC-V Rust target' failed, retrying

That doesn't look like the same issue. [How did you install rust?](https://docs.esp-rs.org/book/installation/rust.html). If you use `espup install --targets=esp32,esp32s2,esp32s3` you should avoid this error. Could you also share the output of running the failing command with verbose log level?

tronical, 2024-11-22:
I'm seeing exactly the same issue (on apple silicon only). Thanks @georgik for the workaround! I think this is some kind of race condition.

GaryCraft, 2025-01-31:
I'm getting the same issue in Linux WSL2 (Ubuntu 2204)
Python 3.10.12

The workaround did not work for me

```
espup install --targets=esp32,esp32s2,esp32s3
[info]: Installing the Espressif Rust ecosystem
[info]: Checking Rust installation
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: Installing Xtensa LLVM
[info]: Installing GCC (xtensa-esp-elf)
[info]: Creating symlink between '/root/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/root/.espup/esp-clang'
[warn]: Installation for 'LLVM' failed, retrying. Error: File exists (os error 17)
[warn]: Previous installation of LLVM exists in: '/root/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'. Reusing this installation
[info]: Creating symlink between '/root/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/root/.espup/esp-clang'
[warn]: Installation for 'LLVM' failed, retrying. Error: File exists (os error 17)
[warn]: Previous installation of LLVM exists in: '/root/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'. Reusing this installation
[info]: Creating symlink between '/root/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/root/.espup/esp-clang'
[warn]: Installation for 'LLVM' failed, retrying. Error: File exists (os error 17)
[warn]: Previous installation of LLVM exists in: '/root/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'. Reusing this installation
[info]: Creating symlink between '/root/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/root/.espup/esp-clang'
[warn]: Installation for 'LLVM' failed, retrying. Error: File exists (os error 17)
Error: 
  × File exists (os error 17)
```

artisdom, 2025-04-09:
> having similar issue on macOS 15, M2
> 
> ➜ ~ espup --version espup 0.13.0 ➜ ~ espup install [info]: Installing the Espressif Rust ecosystem [info]: Checking Rust installation [info]: Installing Xtensa Rust 1.82.0.3 toolchain [info]: Installing Xtensa LLVM [info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain [info]: Installing GCC (xtensa-esp-elf) [info]: Downloading 'rust.tar.xz' [info]: Downloading 'idf_tool_xtensa_elf_clang.libs.tar.xz' [info]: Downloading 'xtensa-esp-elf.tar.xz' [info]: Creating symlink between '/Users/anjumkaiser/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/Users/anjumkaiser/.espup/esp-clang' [info]: Installing 'rust' component for Xtensa Rust toolchain [info]: Downloading 'rust-src.tar.xz' [warn]: Installation for 'RISC-V Rust target' failed, retrying [info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain [warn]: Installation for 'RISC-V Rust target' failed, retrying [info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain [warn]: Installation for 'RISC-V Rust target' failed, retrying [info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain [warn]: Installation for 'RISC-V Rust target' failed, retrying Error: espup::toolchain::rust::install_riscv_target
> 
> × Failed to Install RISC-V targets for 'nightly' toolchain

got similar error log on Ubuntu 24.04,  noticed there's a new version of "espup", 
update it using "cargo install espup", then "espup install" succeed without any issue.

```
$ espup uninstall
[warn]: A new version of espup ('v0.15.0') is available
[info]: Uninstalling the Espressif Rust ecosystem
[info]: Uninstalling Xtensa LLVM
[info]: Uninstalling GCC
[info]: Uninstalling Xtensa Rust toolchain
[info]: Uninstallation successfully completed!

$ cargo install espup
   Compiling espup v0.15.0
    Finished `release` profile [optimized] target(s) in 4m 53s
   Replacing /home/nws/.cargo/bin/espup
    Replaced package `espup v0.11.0` with `espup v0.15.0` (executable `espup`)

$ espup install
espup install
[info]: Installing the Espressif Rust ecosystem
[info]: Checking Rust installation
[info]: Installing Xtensa Rust 1.85.0.0 toolchain
[info]: Installing Xtensa LLVM
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'stable' toolchain
[info]: Creating symlink between '/home/nws/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225/esp-clang/lib' and '/home/nws/.espup/esp-clang'
[info]: All downloads complete
[info]: Installing 'rust' component for Xtensa Rust toolchain
[info]: Installing 'rust-src' component for Xtensa Rust toolchain
 [info]: Installation successfully completed!

	To get started, you need to set up some environment variables by running: '. /home/nws/export-esp.sh'
	This step must be done every time you open a new terminal.
	    See other methods for setting the environment in https://esp-rs.github.io/book/installation/riscv-and-xtensa.html#3-set-up-the-environment-variables
```

MabezDev, 2025-04-14:
I just ran into this:

```bash
mabez espup install --toolchain-version 1.86.0.0
[info]: Installing the Espressif Rust ecosystem
[info]: Checking Rust installation
[info]: Installing Xtensa Rust 1.86.0.0 toolchain
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'stable' toolchain
[info]: Installing GCC (xtensa-esp-elf)
[info]: Installing Xtensa LLVM
[info]: Creating symlink between '/home/mabez/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225/esp-clang/lib' and '/home/mabez/.espup/esp-clang'
[warn]: Installation for 'LLVM' failed, retrying. Error: File exists (os error 17)
[warn]: Previous installation of LLVM exists in: '/home/mabez/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225'. Reusing this installation
[info]: Creating symlink between '/home/mabez/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225/esp-clang/lib' and '/home/mabez/.espup/esp-clang'
[warn]: Installation for 'LLVM' failed, retrying. Error: File exists (os error 17)
[warn]: Previous installation of LLVM exists in: '/home/mabez/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225'. Reusing this installation
[info]: Creating symlink between '/home/mabez/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225/esp-clang/lib' and '/home/mabez/.espup/esp-clang'
[warn]: Installation for 'LLVM' failed, retrying. Error: File exists (os error 17)
[warn]: Previous installation of LLVM exists in: '/home/mabez/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225'. Reusing this installation
[info]: Creating symlink between '/home/mabez/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225/esp-clang/lib' and '/home/mabez/.espup/esp-clang'
[warn]: Installation for 'LLVM' failed, retrying. Error: File exists (os error 17)
[00:00:08] ######################################## 76034696/76034696 idf_tool_xtensa_elf_clang.libs.tar.xz download complete
[00:00:01] ######################################## 3294360/3294360 rust-src.tar.xz download complete                        
thread 'tokio-runtime-worker' panicked at /home/mabez/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/espup-0.15.0/src/toolchain/mod.rs:343:32:
called `Result::unwrap()` on an `Err` value: SendError { .. }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Error:   × File exists (os error 17)
```

I think it's because I still have clang 18 installed.

brainwhat, 2025-04-28:
I get the same problem on M1 Air. The installation stalls at either
```
[info]: Installing Xtensa Rust 1.85.0.0 toolchain
```
or
```
[info]: Installing GCC (xtensa-esp-elf)
```
with zero cpu usage. None of the workarounds helped. 
Has anyone found the solution?

SergioGasquez, 2025-04-29:
Hi @brainwhat can you run the installation command with `-l debug` and print the output here?

brainwhat, 2025-04-29:
Hi, now it doesn't stall for some reason
```
 espup install -l debug

[debug]: connecting to crates.io:443 at 108.157.229.68:443
[debug]: No cached session for DnsName("crates.io")
[debug]: Not resuming any session
[debug]: Using ciphersuite TLS13_AES_128_GCM_SHA256
[debug]: Not resuming
[debug]: TLS1.3 encrypted extensions: [ServerNameAck]
[debug]: ALPN protocol is None
[debug]: created stream: Stream(RustlsStream)
[debug]: sending request GET https://crates.io/api/v1/crates/espup/versions
[debug]: writing prelude: GET /api/v1/crates/espup/versions HTTP/1.1
Host: crates.io
User-Agent: ureq/2.12.1
Accept: */*
accept-encoding: gzip
[debug]: Chunked body in response
[debug]: response 200 to GET https://crates.io/api/v1/crates/espup/versions
[debug]: dropping stream: Stream(RustlsStream)
[info]: Installing the Espressif Rust ecosystem
[debug]: Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases/latest'
[debug]: starting new connection: https://api.github.com/
[debug]: Parsing Xtensa Rust version: 1.85.0.0
[debug]: Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases?page=1&per_page=100'
[debug]: starting new connection: https://api.github.com/
[debug]: Latest Xtensa Rust version: 1.85.0.0
[debug]: Arguments:
            - Export file: "/Users/vasilijfrolov/export-esp.sh"
            - Host triple: aarch64-apple-darwin
            - LLVM Toolchain: Llvm { extended: false, file_name_libs: Some("libs-clang-esp-19.1.2_20250225-aarch64-apple-darwin.tar.xz"), file_name_full: None, host_triple: Aarch64AppleDarwin, path: "/Users/vasilijfrolov/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225", repository_url: "https://github.com/espressif/llvm-project/releases/download/esp-19.1.2_20250225", version: "esp-19.1.2_20250225" }
            - Stable version: "stable"
            - Rust Toolchain: Some(XtensaRust { cargo_home: "/Users/vasilijfrolov/.cargo", dist_file: "rust-1.85.0.0-aarch64-apple-darwin.tar.xz", dist_url: "https://github.com/esp-rs/rust-build/releases/download/v1.85.0.0/rust-1.85.0.0-aarch64-apple-darwin.tar.xz", host_triple: "aarch64-apple-darwin", path: "/Users/vasilijfrolov/.rustup/toolchains/esp", rustup_home: "/Users/vasilijfrolov/.rustup", src_dist_file: "rust-src-1.85.0.0.tar.xz", src_dist_url: "https://github.com/esp-rs/rust-build/releases/download/v1.85.0.0/rust-src-1.85.0.0.tar.xz", toolchain_destination: "/Users/vasilijfrolov/.rustup/toolchains/esp", version: "1.85.0.0" })
            - Skip version parsing: false
            - Targets: {ESP32C2, ESP32S2, ESP32C6, ESP32C3, ESP32, ESP32S3, ESP32P4, ESP32H2}
            - Toolchain path: "/Users/vasilijfrolov/.rustup/toolchains/esp"
            - Toolchain version: None
[info]: Checking Rust installation
[info]: Installing Xtensa Rust 1.85.0.0 toolchain
[info]: Installing GCC (xtensa-esp-elf)
[debug]: GCC path: /Users/vasilijfrolov/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906
[info]: Installing Xtensa LLVM
[debug]: Creating directory: '/Users/vasilijfrolov/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906'
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'stable' toolchain
[debug]: Creating directory: '/Users/vasilijfrolov/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225'
[debug]: starting new connection: https://github.com/
[debug]: starting new connection: https://github.com/
[debug]: starting new connection: https://github.com/
[debug]: redirecting 'https://github.com/espressif/llvm-project/releases/download/esp-19.1.2_20250225/libs-clang-esp-19.1.2_20250225-aarch64-apple-darwin.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/234050162/f125c1f2-1126-4386-89ec-5e905e392fd4?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250429%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250429T100559Z&X-Amz-Expires=300&X-Amz-Signature=7a4c66969ed44604b4cc3983b06a93c2bcf597937384aef7a3b34ef39b5545af&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Dlibs-clang-esp-19.1.2_20250225-aarch64-apple-darwin.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[debug]: redirecting 'https://github.com/espressif/crosstool-NG/releases/download/esp-14.2.0_20240906/xtensa-esp-elf-14.2.0_20240906-aarch64-apple-darwin.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/66057540/1549c302-f520-4e75-b7ce-1fcee5ba5cd4?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250429%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250429T100559Z&X-Amz-Expires=300&X-Amz-Signature=7fb7e48444a8ab6ad8ffaf297f36f0e3fbcf8c1dde29c641290c6c2e7a9f81d9&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Dxtensa-esp-elf-14.2.0_20240906-aarch64-apple-darwin.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[debug]: redirecting 'https://github.com/esp-rs/rust-build/releases/download/v1.85.0.0/rust-src-1.85.0.0.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/404564617/26627945-1be6-4f1d-8c5e-10d641d1c1f0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250429%2Fus-east-1%2Fs3%2Faws4_request&X-A [... cut ...]
Error:   × error sending request for url (https://objects.githubusercontent.com/github-production-release-asset-2e65be/404564617/26627945-1be6-4f1d-8c5e-10d641d1c1f0?X-Amz-
  │ Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250429%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250429T100559Z&X-Amz-Expires=300&X-Amz-
  │ Signature=cdda8f8b03714a6763addd08f807aa3a6eeffaa858a1d018a96143f1a3785576&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Drust-src-
  │ 1.85.0.0.tar.xz&response-content-type=application%2Foctet-stream)
  ├─▶ client error (Connect)
  ╰─▶ connection closed via 
```

brainwhat, 2025-04-29:
Thank you so much. I completely forgot I had a proxy set up. Without it everything works. 
Have a nice day