esp-rs/espup #458: Is `espup` command supported on Raspberry Pi 4B?

hzbd, 2024-11-26:
## Bug description

```
eddie@raspberrypi:~ $ rustup toolchain list
stable-aarch64-unknown-linux-gnu
nightly-aarch64-unknown-linux-gnu (default)
esp

eddie@raspberrypi:~ $ espup uninstall
[info]: Uninstalling the Espressif Rust ecosystem
[info]: Uninstalling Xtensa LLVM
[info]: Uninstalling GCC
[info]: Uninstalling Xtensa Rust toolchain
[info]: Uninstallation successfully completed!

eddie@raspberrypi:~ $ espup install
[info]: Installing the Espressif Rust ecosystem
[info]: Checking Rust installation
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[info]: Installing Xtensa LLVM
[info]: Installing GCC (xtensa-esp-elf)
[info]: Installing Xtensa Rust 1.82.0.3 toolchain
[info]: Downloading 'rust.tar.xz'
[info]: Downloading 'idf_tool_xtensa_elf_clang.libs.tar.xz'
[info]: Downloading 'xtensa-esp-elf.tar.xz'
[warn]: Installation for 'RISC-V Rust target' failed, retrying
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying
Error: espup::toolchain::rust::install_riscv_target

  × Failed to Install RISC-V targets for 'nightly' toolchain

```

Does `espup` support use on Raspberry Pi 4B? `espup install` keeps reporting errors and the `RISC-V target` can't be installed. Where can I view more detailed log information? or how can I fix it?

## Environment

```
eddie@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

eddie@raspberrypi:~ $ rustc  --version
rustc 1.85.0-nightly (a47555110 2024-11-22)

eddie@raspberrypi:~ $ espup --version
espup 0.13.0
```

hzbd, 2024-11-26:
It may be the same problem.

https://github.com/esp-rs/espup/issues/455#issuecomment-2487134598

hzbd, 2024-11-27:
The following is the list of commands that I have executed during the deployment of the environment（Raspberry PI 4B）.

```
$ sudo apt update && \
        sudo apt install -y openssl git libssl-dev ca-certificates \
        build-essential pkg-config pkgconf flex bison gperf \
        python3-venv python-is-python3 

$ sudo apt install -y libudev-dev librust-libudev-sys-dev \
        gcc-riscv64-unknown-elf \
        llvm-dev libclang-dev clang \
        libuv1-dev:armhf libuv1-dev \
        zlib1g-dev \
        cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0
        
$ mkdir -p ~/.espressif && \
    cd ~/.espressif && \
    git clone -b v5.3.1 --recursive git@github.com:espressif/esp-idf.git && \
    cd ~/.espressif/esp-idf && \
    ./install.sh esp32,esp32s2,esp32c3 && \
    . $HOME/.espressif/esp-idf/export.sh
     
$ curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
$ rustup toolchain install nightly --component rust-src
$ rustup default nightly

$ cargo install cargo-espflash ldproxy rustfilt espup cargo-generate
$ espup install --targets=esp32,esp32s2,esp32s3,esp32c3
```

```
eddie@raspberrypi:~/Projects $ rustup show
Default host: aarch64-unknown-linux-gnu
rustup home:  /home/eddie/.rustup

installed toolchains
--------------------

stable-aarch64-unknown-linux-gnu
nightly-aarch64-unknown-linux-gnu (default)
esp

installed targets for active toolchain
--------------------------------------

aarch64-unknown-linux-gnu
riscv32imc-unknown-none-elf

active toolchain
----------------

nightly-aarch64-unknown-linux-gnu (default)
rustc 1.85.0-nightly (a47555110 2024-11-22)
```

hzbd, 2024-11-27:
Sorry, I may not have made it clear above: The problem still hasn't been solved. @SergioGasquez

hzbd, 2024-11-27:
Sorry, I might not have made it clear above. The problem still hasn't been solved. I just pasted out all my relevant operation commands. @SergioGasquez

SergioGasquez, 2024-11-27:
- Which is esp32 variant are you targeting?
- Does `espup install --targets=esp32,esp32s2,esp32s3,esp32c3` also fails with `Failed to Install RISC-V targets for 'nightly' toolchain`?
- `esp` toolchain is listed when you do `rustup show`, have you tried building a project with it?

hzbd, 2024-11-28:
I'm extremely sorry for not replying in time.

> Does `espup install --targets=esp32,esp32s2,esp32s3,esp32c3` also fails with `Failed to Install RISC-V targets for 'nightly' toolchain`?

```
$ espup install
OR
eddie@raspberrypi:~ $ espup install --targets=esp32s2,esp32s3,esp32,esp32c3
[info]: Installing the Espressif Rust ecosystem
[info]: Checking Rust installation
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[info]: Installing Xtensa LLVM
[info]: Installing Xtensa Rust 1.82.0.3 toolchain
[info]: Downloading 'rust.tar.xz'
[info]: Installing GCC (xtensa-esp-elf)
[info]: Downloading 'idf_tool_xtensa_elf_clang.libs.tar.xz'
[info]: Downloading 'xtensa-esp-elf.tar.xz'
[warn]: Installation for 'RISC-V Rust target' failed, retrying
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'Xtensa Rust' failed, retrying
[warn]: Installation for 'LLVM' failed, retrying
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.82.0.3 toolchain
[info]: Downloading 'rust.tar.xz'
[warn]: Previous installation of LLVM exists in: '/home/eddie/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'. Reusing this installation
[info]: Creating symlink between '/home/eddie/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/home/eddie/.espup/esp-clang'
[warn]: Installation for 'RISC-V Rust target' failed, retrying
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying
Error: espup::toolchain::rust::install_riscv_target

  × Failed to Install RISC-V targets for 'nightly' toolchain
```

Neither of the above two execution methods has been successful. I don't know whether this situation is caused by problems with my environment or if the program doesn't support such execution by default. 

However, if it is modified as follows to specify only the target of the `Xtensa` architecture, the installation can be successful.
```
# Install target for Xtensa
eddie@raspberrypi:~ $ espup install --targets=esp32s2,esp32s3,esp32
[info]: Installing the Espressif Rust ecosystem
[info]: Checking Rust installation
[info]: Installing Xtensa Rust 1.82.0.3 toolchain
[info]: Installing Xtensa LLVM
[info]: Installing GCC (xtensa-esp-elf)
[info]: Downloading 'idf_tool_xtensa_elf_clang.libs.tar.xz'
[info]: Downloading 'xtensa-esp-elf.tar.xz'
[info]: Downloading 'rust.tar.xz'
[info]: Creating symlink between '/home/eddie/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/home/eddie/.espup/esp-clang'
[info]: Installing 'rust' component for Xtensa Rust toolchain
[info]: Downloading 'rust-src.tar.xz'
[info]: Installing 'rust-src' component for Xtensa Rust toolchain
[info]: Installation successfully completed!

	To get started, you need to set up some environment variables by running: '. /home/eddie/export-esp.sh'
	This step must be done every time you open a new terminal.
	    See other methods for setting the environment in https://esp-rs.github.io/book/installation/riscv-and-xtensa.html#3-set-up-the-environment-variables

# Install target for RISC-V
eddie@raspberrypi:~ $ rustup target add riscv32imc-unknown-none-elf
info: downloading component 'rust-std' for 'riscv32imc-unknown-none-elf'
info: installing component 'rust-std' for 'riscv32imc-unknown-none-elf'
  9.8 MiB /   9.8 MiB (100 %)   2.5 MiB/s in  4s ETA:  0s
eddie@raspberrypi:~ $ rustup target list|grep install
aarch64-unknown-linux-gnu (installed)
riscv32imc-unknown-none-elf (installed)
```

> `esp` toolchain is listed when you do `rustup show`, have you tried building a project with it?

After the espup install steps mentioned above are executed successfully, the example project can be compiled successfully.

SergioGasquez, 2024-11-28:
Sorry, the `esp32c3` was not supposed to be in the command args. Glad that you can build the projects for all the targets. Id say that the problem is related with the Rust installation, maybe running the espup failing command with some extra level (`-l debug`) would give us some more information. 

I'll be closing the info as I cant really reproduce it and we already have a workarround

closed 2024-11-28 by SergioGasquez