esp-rs/espup #460: Github API request returns only a few releases

SergioGasquez, 2024-12-09:
## Bug description

While working on https://github.com/esp-rs/xtensa-toolchain/pull/30, noticed that the `xtensa-toolchain` started failing (its trying to install `1.64.0.0`), looking at the workflow and the reply of the github api request (https://api.github.com/repos/esp-rs/rust-build/releases) we see that the requested version is not there as gh only gives 9 latest releases. 

## Expected behavior

We can install any version, no matter how old it is.

jessebraham, 2024-12-09:
`per_page` should be set to 30 by default, does explicitly specifying a value for this in the request (e.g. 100) resolve the issue?

SergioGasquez, 2024-12-09:
> `per_page` should be set to 30 by default, does explicitly specifying a value for this in the request (e.g. 100) resolve the issue?

Yes, using:
```diff
-const XTENSA_RUST_API_URL: &str = "https://api.github.com/repos/esp-rs/rust-build/releases";
+const XTENSA_RUST_API_URL: &str =
+    "https://api.github.com/repos/esp-rs/rust-build/releases?page=1&per_page=1000";
```
seems to solve the issue

closed 2024-12-10 by SergioGasquez