esp-rs/espup #466: Incompatible Version of GLIBC

dragazo, 2025-01-04:
## Bug description

Hi, I'm not sure exactly which repo to open this issue, but I'll start with espup since it's the updater for the esp toolchain.

About a year ago, I was using the esp toolchain (installed via espup) and everything was totally fine. However, I recently decided to get back into that project, so I did an `espup update` and opened up my project. When I try to build, or even use `cargo` for anything whatsoever, I get the following error:

```
/home/devin/.rustup/toolchains/esp/bin/cargo: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/devin/.rustup/toolchains/esp/bin/cargo)
/home/devin/.rustup/toolchains/esp/bin/cargo: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /home/devin/.rustup/toolchains/esp/bin/cargo)
/home/devin/.rustup/toolchains/esp/bin/cargo: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/devin/.rustup/toolchains/esp/bin/cargo)
```

I can't really update glibc without obliterating my computer, but I did a side installation (from source) of glibc 2.34 and tried to use that:

```
$ LD_LIBRARY_PATH=/home/devin/glibc/glibc-2.34-install/lib /home/devin/.rustup/toolchains/esp/bin/cargo
Segmentation fault (core dumped)
```

I guess this means I can no longer use `espup` and have to instead build the toolchain from source? Is there any info on how to do this?

## To Reproduce

Steps to reproduce the behavior:

1. Use a system with glibc 2.31 or below (mine is precisely 2.31)
2. Run `espup update`
3. Attempt to use `cargo +esp`

## Expected behavior

I expected espup to install a build for my current glibc (or else error out, rather than installing an incompatible one)

## Screenshots

N/A (terminal snippets above)

## Environment

- OS: Linux Mint 20
- espup version: 0.14.0

## Additional context

N/A

SergioGasquez, 2025-01-07:
Is there any reason not to update your OS? See https://github.com/esp-rs/rust-build/issues/252

dragazo, 2025-01-07:
If I do two system updates/migrations I could get to 2.35 (from 2.31), but based on the error message, that would actually be too _high_ of a version to work.

SergioGasquez, 2025-01-08:
Glibc 2.35 should work fine afaik

dragazo, 2025-02-12:
I think it'll only be resolved when I do a fresh linux install with a newer version (which I'm planning in a few months). Last time I did a distro upgrade it borked my machine, so I'm not risking that until I'm ready to do a clean install anyway. But we can close this issue since that'll technically fix it anyway, one way or another.

closed 2025-02-12 by dragazo