esp-rs/espup #467: Bug when installing/updating/re-installing Xtensa Rust toolchain with `espup`

avshenuk, 2025-01-16:
## Bug description

Whenever installing/updating/re-installing, getting this error every time:
```
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
Error: 
  × failed to iterate over archive
  ╰─▶ premature eof
```

## To Reproduce

Steps to reproduce the behavior:

1. `espup uninstall && espup install`, or
2. `espup install`, or
3. `espup update` 

## Expected behavior

Xtensa Rust 1.84.0.0 toolchain properly insalled.

## Environment

- OS: MacOS 14.6.1 (23G93)
- espup version: espup 0.14.0

SergioGasquez, 2025-01-16:
Can you run the command enabling debug logs (`-l debug`)?

avshenuk, 2025-01-16:
Certainly:
```
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[debug]: starting new connection: https://github.com/
[debug]: redirecting 'https://github.com/esp-rs/rust-build/releases/download/v1.84.0.0/rust-src-1.84.0.0.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/404564617/15b0cd57-a21b-4d85-a9f1-4ebca64861be?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250116%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250116T114106Z&X-Amz-Expires=300&X-Amz-Signature=8e5609a0ff6b30a864f0cf096410a02bc96740aaf0fdb962008f06d617ab52a2&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Drust-src-1.84.0.0.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[info]: All downloads complete
[debug]: Extracting tar.xz file to '/Users/avsheniuk/.rustup/tmp/.tmpN4g8BQ'
[debug]: starting new connection: https://github.com/
[info]: All downloads complete
[debug]: Extracting tar.xz file to '/Users/avsheniuk/.rustup/tmp/.tmpN4g8BQ'
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[debug]: starting new connection: https://github.com/
[debug]: redirecting 'https://github.com/esp-rs/rust-build/releases/download/v1.84.0.0/rust-src-1.84.0.0.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/404564617/15b0cd57-a21b-4d85-a9f1-4ebca64861be?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250116%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250116T114106Z&X-Amz-Expires=300&X-Amz-Signature=8e5609a0ff6b30a864f0cf096410a02bc96740aaf0fdb962008f06d617ab52a2&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Drust-src-1.84.0.0.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[info]: All downloads complete
[debug]: Extracting tar.xz file to '/Users/avsheniuk/.rustup/tmp/.tmpefbo1K'
[debug]: starting new connection: https://github.com/
[info]: All downloads complete
[debug]: Extracting tar.xz file to '/Users/avsheniuk/.rustup/tmp/.tmpefbo1K'
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[debug]: starting new connection: https://github.com/
[debug]: redirecting 'https://github.com/esp-rs/rust-build/releases/download/v1.84.0.0/rust-src-1.84.0.0.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/404564617/15b0cd57-a21b-4d85-a9f1-4ebca64861be?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250116%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250116T114106Z&X-Amz-Expires=300&X-Amz-Signature=8e5609a0ff6b30a864f0cf096410a02bc96740aaf0fdb962008f06d617ab52a2&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Drust-src-1.84.0.0.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[info]: All downloads complete
[debug]: Extracting tar.xz file to '/Users/avsheniuk/.rustup/tmp/.tmpn3nx4z'
[debug]: starting new connection: https://github.com/
[info]: All downloads complete
[debug]: Extracting tar.xz file to '/Users/avsheniuk/.rustup/tmp/.tmpn3nx4z'
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[debug]: starting new connection: https://github.com/
[debug]: redirecting 'https://github.com/esp-rs/rust-build/releases/download/v1.84.0.0/rust-src-1.84.0.0.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/404564617/15b0cd57-a21b-4d85-a9f1-4ebca64861be?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250116%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250116T114106Z&X-Amz-Expires=300&X-Amz-Signature=8e5609a0ff6b30a864f0cf096410a02bc96740aaf0fdb962008f06d617ab52a2&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Drust-src-1.84.0.0.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[info]: All downloads complete
[debug]: Extracting tar.xz file to '/Users/avsheniuk/.rustup/tmp/.tmp9dOYip'
[debug]: starting new connection: https://github.com/
[info]: All downloads complete
[debug]: Extracting tar.xz file to '/Users/avsheniuk/.rustup/tmp/.tmp9dOYip'
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
Error: 
  × failed to iterate over archive
  ╰─▶ premature eof
```

SergioGasquez, 2025-01-16:
Tried reproducing the issue on my mac 15.2 with no luck:
```
> espup --version
espup 0.14.0

> espup install
[info]: Installing the Espressif Rust ecosystem
[info]: Checking Rust installation
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: Installing Xtensa LLVM
[info]: Installing GCC (xtensa-esp-elf)
[info]: Installing 'rust' component for Xtensa Rust toolchain
[info]: Installing 'rust-src' component for Xtensa Rust toolchain
[info]: Creating symlink between '/Users/sergio/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/Users/sergio/.espup/esp-clang'
[info]: All downloads complete
[info]: Installation successfully completed!
[00:00:01] ######################################## 120558072/120558072 rust.tar.xz download complete                      
	To get started, you need to set up some environment variables by running: '. /Users/sergio/export-esp.sh'
	This step must be done every time you open a new terminal.
	    See other methods for setting the environment in https://esp-rs.github.io/book/installation/riscv-and-xtensa.html#3-set-up-the-environment-variables
```

Is your mac and intel mac?

SergioGasquez, 2025-01-16:
Not sure why the redirection to that user, but its also happening for me:
```
espup install -l debug
[debug]: connecting to crates.io:443 at [2600:9000:237e:6e00:c:7ed3:240:93a1]:443
[debug]: No cached session for DnsName("crates.io")
[debug]: Not resuming any session
[debug]: Using ciphersuite TLS13_AES_128_GCM_SHA256
[debug]: Not resuming
[debug]: TLS1.3 encrypted extensions: [ServerNameAck]
[debug]: ALPN protocol is None
[debug]: created stream: Stream(RustlsStream)
[debug]: sending request GET https://crates.io/api/v1/crates/espup/versions
[debug]: writing prelude: GET /api/v1/crates/espup/versions HTTP/1.1
Host: crates.io
User-Agent: ureq/2.12.1
Accept: */*
accept-encoding: gzip
[debug]: Chunked body in response
[debug]: response 200 to GET https://crates.io/api/v1/crates/espup/versions
[debug]: dropping stream: Stream(RustlsStream)
[info]: Installing the Espressif Rust ecosystem
[debug]: Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases/latest'
[debug]: starting new connection: https://api.github.com/
[debug]: Parsing Xtensa Rust version: 1.84.0.0
[debug]: Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases?page=1&per_page=100'
[debug]: starting new connection: https://api.github.com/
[debug]: Latest Xtensa Rust version: 1.84.0.0
[debug]: Arguments:
            - Export file: "/Users/sergio/export-esp.sh"
            - Host triple: aarch64-apple-darwin
            - LLVM Toolchain: Llvm { extended: false, file_name_libs: Some("libs-clang-esp-18.1.2_20240912-aarch64-apple-darwin.tar.xz"), file_name_full: None, host_triple: Aarch64AppleDarwin, path: "/Users/sergio/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912", repository_url: "https://github.com/espressif/llvm-project/releases/download/esp-18.1.2_20240912", version: "esp-18.1.2_20240912" }
            - Nightly version: "nightly"
            - Rust Toolchain: Some(XtensaRust { cargo_home: "/Users/sergio/.cargo", dist_file: "rust-1.84.0.0-aarch64-apple-darwin.tar.xz", dist_url: "https://github.com/esp-rs/rust-build/releases/download/v1.84.0.0/rust-1.84.0.0-aarch64-apple-darwin.tar.xz", host_triple: "aarch64-apple-darwin", path: "/Users/sergio/.rustup/toolchains/esp", rustup_home: "/Users/sergio/.rustup", src_dist_file: "rust-src-1.84.0.0.tar.xz", src_dist_url: "https://github.com/esp-rs/rust-build/releases/download/v1.84.0.0/rust-src-1.84.0.0.tar.xz", toolchain_destination: "/Users/sergio/.rustup/toolchains/esp", version: "1.84.0.0" })
            - Skip version parsing: false
            - Targets: {ESP32S3, ESP32C2, ESP32P4, ESP32, ESP32C3, ESP32H2, ESP32S2, ESP32C6}
            - Toolchain path: "/Users/sergio/.rustup/toolchains/esp"
            - Toolchain version: None
[info]: Checking Rust installation
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: Installing Xtensa LLVM
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[info]: Installing GCC (xtensa-esp-elf)
[debug]: GCC path: /Users/sergio/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906
[debug]: Creating directory: '/Users/sergio/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906'
[debug]: Creating directory: '/Users/sergio/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'
[debug]: starting new connection: https://github.com/
[debug]: starting new connection: https://github.com/
[debug]: starting new connection: https://github.com/
[debug]: redirecting 'https://github.com/espressif/llvm-project/releases/download/esp-18.1.2_20240912/libs-clang-esp-18.1.2_20240912-aarch64-apple-darwin.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/234050162/5e916b0d-5974-4261-aeb1-4148ad9eabcb?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250116%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250116T133931Z&X-Amz-Expires=300&X-Amz-Signature=6b4d41f63ddf747b7c5f192aa8dc1f4eba19a04d1b6425701a529e226ded433c&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Dlibs-clang-esp-18.1.2_20240912-aarch64-apple-darwin.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[debug]: redirecting 'https://github.com/espressif/crosstool-NG/releases/download/esp-14.2.0_20240906/xtensa-esp-elf-14.2.0_20240906-aarch64-apple-darwin.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/66057540/1549c302-f520-4e75-b7ce-1fcee5ba5cd4?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250116%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250116T133931Z&X-Amz-Expires=300&X-Amz-Signature=f4a4e45398e16cb5d1514c8c9241edf77732bdf0e39970ae6500477576edf4f7&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Dxtensa-esp-elf-14.2.0_20240906-aarch64-apple-darwin.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[debug]: redirecting 'https://github.com/esp-rs/rust-build/releases/download/v1.84.0.0/rust-src-1.84.0.0.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/404564617/15b0cd57-a21b-4d85-a9f1-4ebca64861be?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250116%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250 [... cut ...]
[info]: Installation successfully completed!

	To get started, you need to set up some environment variables by running: '. /Users/sergio/export-esp.sh'
	This step must be done every time you open a new terminal.
	    See other methods for setting the environment in https://esp-rs.github.io/book/installation/riscv-and-xtensa.html#3-set-up-the-environment-variables
```

The premature eof error migth indicate that the file was not properly donwloaded, but you also got the message of All downloads complete....

avshenuk, 2025-01-16:
Alright. I, instead, manually went through the steps performed in the source code of `espup` here: https://github.com/esp-rs/espup/blob/v0.14.0/src/toolchain/rust.rs#L237 . And that seemed to do the trick for me to install "Xtensa Rust 1.84.0.0". Running `espup update` afterwards just wrapped it up :)
```
[warn]: Previous installation of Xtensa Rust 1.84.0.0 exists in: '/Users/avsheniuk/.rustup/toolchains/esp'. Reusing this installation
[debug]: Creating export file
[info]: Update successfully completed!

	To get started, you need to set up some environment variables by running: '. /Users/avsheniuk/export-esp.sh'
	This step must be done every time you open a new terminal.
	    See other methods for setting the environment in https://esp-rs.github.io/book/installation/riscv-and-xtensa.html#3-set-up-the-environment-variables
```
So hopefully, it should work going forward. 

Thank you for your help, and I guess, you can close the issue now...

closed 2025-01-16 by SergioGasquez

madhugoundla, 2025-01-16:
rust-src-nightly specialuser$ espup install
[info]: Installing the Espressif Rust ecosystem
[info]: Checking Rust installation
[warn]: Previous installation of LLVM exists in: '/Users/specialuser/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'. Reusing this installation
[info]: Installing GCC (xtensa-esp-elf)
[warn]: Previous installation of GCC exists in: '/Users/specialuser/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906'. Reusing this installation
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[info]: Creating symlink between '/Users/specialuser/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/Users/specialuser/.espup/esp-clang'
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
Error: 
  × failed to iterate over archive
  ╰─▶ premature eof

SergioGasquez, 2025-01-22:
Duplication of #468, see coments there