esp-rs/espup #468: when installing/updating/re-installing Xtensa Rust toolchain with espup

madhugoundla, 2025-01-16:
Hello what is the manual steps for macos? can you please provide the exact steps?, i am having the same issue

Getting same error on Mac Pro and Mac mini, 14.6.1 and 15.*

Whenever installing/updating/re-installing, getting this error every time:

`rust-src-nightly specialuser$ espup install
[info]: Installing the Espressif Rust ecosystem
[info]: Checking Rust installation
[warn]: Previous installation of LLVM exists in: '/Users/specialuser/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'. Reusing this installation
[info]: Installing GCC (xtensa-esp-elf)
[warn]: Previous installation of GCC exists in: '/Users/specialuser/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906'. Reusing this installation
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[info]: Creating symlink between '/Users/specialuser/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/Users/specialuser/.espup/esp-clang'
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
Error:
× failed to iterate over archive
╰─▶ premature eof`

zpg6, 2025-01-18:
+1 

```
espup update --nightly-version 1.84.0
```

```
[info]: Updating the Espressif Rust ecosystem
[info]: Checking Rust installation
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for '1.83' toolchain
[warn]: Previous installation of LLVM exists in: '/Users/me/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'. Reusing this installation
[info]: Installing GCC (xtensa-esp-elf)
[warn]: Previous installation of GCC exists in: '/Users/me/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906'. Reusing this installation
[info]: Creating symlink between '/Users/me/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912/esp-clang/lib' and '/Users/me/.espup/esp-clang'
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: All downloads complete
[info]: All downloads complete
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: failed to iterate over archive
Error: 
  × failed to iterate over archive
  ╰─▶ premature eof
```

SergioGasquez, 2025-01-21:
Can you try installing a different Xtensa Rust version? `espup update -v 1.83.0.1` and see if the issue still appear

> `espup update --nightly-version 1.84.0`,
The nightly version arg is for installing a 1.84.0 nightly channel (used for Riscv devices, not for Xtensa devices)

0x53A, 2025-01-21:
I'm also on macOS, had the same issue where ``espup install`` would fail with "Error: failed to iterate over archive", and  with v1.83.0.1 it did install the toolchain.

liebman, 2025-01-21:
I think the issue is that for some reason there is not apple `x86_64` build of the rust compiler....

MabezDev, 2025-01-21:
For some reason the build didn't trigger - I might have skipped over it by accident.

I've started the build now: https://github.com/esp-rs/rust-build/actions/runs/12896349203 hopfully that will go green in the next few hours and things will be resolved. Sorry about that!

SergioGasquez, 2025-01-22:
Can we confirm that the issue is now resolved? @0x53A @zpg6 

I think we should improve the error message as its quite misleading that the download does not fail and the error is `failed to iterate over archive`

madhugoundla, 2025-01-22:
The issue seems to be resolved. Can close this Bug.

0x53A, 2025-01-22:
It sounds a bit like the http error code is not checked. That is, github returns a 404, but espup ignores that and just writes a zero-length file, which then can’t be extracted.

zpg6, 2025-01-27:
> I still am at 1.83.0
> 
> (Open new terminal)
> 
> ```
> . $HOME/export-esp.sh
> rustc --version
> ```
> 
> Returns:
> 
> ```
> rustc 1.83.0 (90b35a623 2024-11-26)
> ```
> 
> 

Uninstall and reinstall fixed my issue

closed 2025-01-28 by SergioGasquez