esp-rs/espup #469: Windows aarch64 support

ShyneTurtle, 2025-01-21:
## Motivations

I recently bought a windows on ARM laptop (Snapdragon X), however `espup` does not feature a build for it and since my rustup installation is `aarch64`, i can't use the x86 one with emulation.

- Would you like to implement this feature? [y/n]

No, i don't have time to jump into a project i don't know the inner workings of.

## Solution

Since `linux` and `darwin` both feature `aarch64` builds i assume it would be as simple as adding the builds for `windows aarch64` in some kind of config

## Alternatives

N/S

## Additional context

N/S

SergioGasquez, 2025-01-21:
In the meantime, can you build it from source?? `cargo install espup`

MabezDev, 2025-01-21:
Thanks for the report @ShyneTurtle. As github don't currently offer a aarch64 windows runner (except for paid users), and we don't have any hardware in our server room with this combo, we are unfortunately unable to provide prebuilt compilers for this host.

You may be able to get around this limitation by using `x86-64-pc-windows-msvc` compiler build by installing it manually, and letting (if it exists, like it does on macOS) the x86 emulation take care of running the compiler. This might make compile times slightly longer though.

The final option is building the compiler itself, which honestly isn't that difficult provided you can install the required dependencies for your platform. See https://github.com/esp-rs/rust#building-from-source

ShyneTurtle, 2025-01-21:
I'm considering building from source but i'm worried about maintaining the installation: how would the fork compare to rustup in terms of ease of use and updates ?

MabezDev, 2025-01-21:
> I'm considering building from source but i'm worried about maintaining the installation: how would the fork compare to rustup in terms of ease of use and updates ?

All the dependencies are the same as upstream Rust, so it works the same way. We rebase on the stable release branch only, which is currently https://github.com/esp-rs/rust 1.84, when we rebase for 1.85 you can choose to upgrade (meaning compile it again) or stay on your current version if it's working.

ShyneTurtle, 2025-01-21:
Thank you for your answer, i'll try that next.

I'm not sure if the issue is the fact that i installed aarch64 rust and clang+llvm, while using esptool on x86, but while compiling my project i get an error while compiling esp-idf-sys : `Neither any or platform specifc match found. Please create an issue on https://github.com/esp-rs/embuild and report your operating system`
(currently rerunning the build with rust backtrace to get more info)

MabezDev, 2025-01-21:
embuild, and likely the entire esp-idf build system (this might not be true, I just don't know) doesn't know about your host machine. You might have better luck with the officially supported bare metal programming environment: https://github.com/esp-rs/esp-hal.

ShyneTurtle, 2025-01-22:
The downer is really that the laptop is capable of emulating x86 for compatibility, but the software won't run because of runtime compatibility checks.

For this to work i'd have to self-compile and edit code in most of `esp-rs` 's build stack i imagine

How would  building `no_std` apps help the compilation ? isn't the build stack shared ?

ShyneTurtle, 2025-02-25:
Any updates on this ? (don't wanna rush anyone, just a head's up)

SergioGasquez, 2025-03-11:
> Any updates on this ? (don't wanna rush anyone, just a head's up)

Hi! We wont be implementing this until the Windows aarch64 gh-hosted runner is available for us, not sure when gh will make it publicly available

ShyneTurtle, 2025-05-06:
The problem is still NOT platform-native compilation, but rather not preventing the program from starting at runtime with a triple check as the laptop features emulation, and is capable of running the x64 version.

MabezDev, 2025-09-22:
I don't think we want to increase the maintenance burden we have further. Our efforts are better spent on the upstreaming of LLVM/Xtensa into rustc proper.