esp-rs/espup #470: Library not loaded on macOS

HaoboGu, 2025-01-21:
## Bug description

When using `cargo-binstall` to install the espup, the installed binary seems broken?

The following is the log:

```shell
# haobogu @ HaobodeMac-mini in ~ [23:08:25]
$ cargo binstall espup
 INFO resolve: Resolving package: 'espup'
 WARN The package espup v0.14.0 (aarch64-apple-darwin) has been downloaded from github.com
 INFO This will install the following binaries:
 INFO   - espup => /Users/haobogu/.cargo/bin/espup
Do you wish to continue? [yes]/no
? yes
 INFO Installing binaries...
 INFO Done in 8.375863s

# haobogu @ HaobodeMac-mini in ~ [23:09:18]
$ espup -h
dyld[58722]: Library not loaded: /opt/homebrew/opt/xz/lib/liblzma.5.dylib
  Referenced from: <4148C9A1-6C54-3394-B660-D2E1EE168DD0> /Users/haobogu/.cargo/bin/espup
  Reason: tried: '/opt/homebrew/opt/xz/lib/liblzma.5.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/xz/lib/liblzma.5.dylib' (no such file), '/opt/homebrew/opt/xz/lib/liblzma.5.dylib' (no such file)
[1]    58722 abort      espup -h

# haobogu @ HaobodeMac-mini in ~ [23:09:23] C:134
$ espup install
dyld[58789]: Library not loaded: /opt/homebrew/opt/xz/lib/liblzma.5.dylib
  Referenced from: <4148C9A1-6C54-3394-B660-D2E1EE168DD0> /Users/haobogu/.cargo/bin/espup
  Reason: tried: '/opt/homebrew/opt/xz/lib/liblzma.5.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/xz/lib/liblzma.5.dylib' (no such file), '/opt/homebrew/opt/xz/lib/liblzma.5.dylib' (no such file)
[1]    58789 abort      espup install
```

## Expected behavior

I expect the installed binary to be correct

## Screenshots

[image]

## Environment

- OS: macOS 15.2
- espup version: [e.g. 0.1.0]

## Additional context

SergioGasquez, 2025-01-22:
Just to confirm, this issue doesnt happen if you use `cargo install`, right?

HaoboGu, 2025-01-22:
> Just to confirm, this issue doesnt happen if you use `cargo install`, right?

Yes, `cargo install` works perfectly

SergioGasquez, 2025-01-22:
Can you try installing `xz`: `brew install xz` and then rerunning the `binstall` command?

HaoboGu, 2025-01-22:
Yes it worked, I don't need to uninstall and reinstall it, just running `brew install xz` is enough:

```shell
# haobogu @ HaobodeMac-mini in ~ [20:54:55] 
$ espup -h                                                                                                   
dyld[76371]: Library not loaded: /opt/homebrew/opt/xz/lib/liblzma.5.dylib
  Referenced from: <4148C9A1-6C54-3394-B660-D2E1EE168DD0> /Users/haobogu/.cargo/bin/espup
  Reason: tried: '/opt/homebrew/opt/xz/lib/liblzma.5.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/xz/lib/liblzma.5.dylib' (no such file), '/opt/homebrew/opt/xz/lib/liblzma.5.dylib' (no such file)
[1]    76371 abort      espup -h

# haobogu @ HaobodeMac-mini in ~ [20:54:59] 
$ brew install xz                                  
==> Downloading https://mirrors.tuna.tsinghua.edu.cn/homebrew-bottles/api/formula.jws.json
==> Downloading https://mirrors.tuna.tsinghua.edu.cn/homebrew-bottles/api/cask.jws.json
==> Fetching xz
==> Downloading https://mirrors.tuna.tsinghua.edu.cn/homebrew-bottles/xz-5.6.3.arm64_sequoia.bottle.tar.gz
########################################################################################################################################################################################################################### 100.0%
==> Pouring xz-5.6.3.arm64_sequoia.bottle.tar.gz
🍺  /opt/homebrew/Cellar/xz/5.6.3: 96 files, 2.0MB
==> Running `brew cleanup xz`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).

# haobogu @ HaobodeMac-mini in ~ [20:55:10] 
$ espup -h       
Tool for installing and maintaining Espressif Rust ecosystem.

Usage: espup <COMMAND>

Commands:
  completions  Generate completions for the given shell
  install      Installs Espressif Rust ecosystem
  uninstall    Uninstalls Espressif Rust ecosystem
  update       Updates Xtensa Rust toolchain
  help         Print this message or the help of the given subcommand(s)

Options:
  -h, --help     Print help
  -V, --version  Print version
```

closed 2025-01-23 by SergioGasquez