esp-rs/espup #474: M1 Mac run "espup uninstall && espup install -l debug " Error

limooooooon, 2025-02-09:
## Bug description

M1 Mac run "espup uninstall && espup install -l debug " Error

- Would you like to work on a fix? [y/n]

## To Reproduce

[info]: Uninstalling the Espressif Rust ecosystem
[info]: Uninstalling Xtensa LLVM
[info]: Uninstalling GCC
[info]: Uninstalling Xtensa Rust toolchain
[info]: Uninstallation successfully completed!
[debug]: connecting to crates.io:443 at [2600:9000:2753:3e00:c:7ed3:240:93a1]:443
[debug]: No cached session for DnsName("crates.io")
[debug]: Not resuming any session
[debug]: Using ciphersuite TLS13_AES_128_GCM_SHA256
[debug]: Not resuming
[debug]: TLS1.3 encrypted extensions: [ServerNameAck]
[debug]: ALPN protocol is None
[debug]: created stream: Stream(RustlsStream)
[debug]: sending request GET https://crates.io/api/v1/crates/espup/versions
[debug]: writing prelude: GET /api/v1/crates/espup/versions HTTP/1.1
Host: crates.io
User-Agent: ureq/2.12.1
Accept: */*
accept-encoding: gzip
[debug]: Chunked body in response
[debug]: response 200 to GET https://crates.io/api/v1/crates/espup/versions
[debug]: dropping stream: Stream(RustlsStream)
[info]: Installing the Espressif Rust ecosystem
[debug]: Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases/latest'
[debug]: starting new connection: https://api.github.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://api.github.com/'
[debug]: Parsing Xtensa Rust version: 1.84.0.0
[debug]: Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases?page=1&per_page=100'
[debug]: starting new connection: https://api.github.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://api.github.com/'
[debug]: Latest Xtensa Rust version: 1.84.0.0
[debug]: Arguments:
            - Export file: "/Users/lemon/export-esp.sh"
            - Host triple: aarch64-apple-darwin
            - LLVM Toolchain: Llvm { extended: false, file_name_libs: Some("libs-clang-esp-18.1.2_20240912-aarch64-apple-darwin.tar.xz"), file_name_full: None, host_triple: Aarch64AppleDarwin, path: "/Users/lemon/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912", repository_url: "https://github.com/espressif/llvm-project/releases/download/esp-18.1.2_20240912", version: "esp-18.1.2_20240912" }
            - Nightly version: "nightly"
            - Rust Toolchain: Some(XtensaRust { cargo_home: "/Users/lemon/.cargo", dist_file: "rust-1.84.0.0-aarch64-apple-darwin.tar.xz", dist_url: "https://github.com/esp-rs/rust-build/releases/download/v1.84.0.0/rust-1.84.0.0-aarch64-apple-darwin.tar.xz", host_triple: "aarch64-apple-darwin", path: "/Users/lemon/.rustup/toolchains/esp", rustup_home: "/Users/lemon/.rustup", src_dist_file: "rust-src-1.84.0.0.tar.xz", src_dist_url: "https://github.com/esp-rs/rust-build/releases/download/v1.84.0.0/rust-src-1.84.0.0.tar.xz", toolchain_destination: "/Users/lemon/.rustup/toolchains/esp", version: "1.84.0.0" })
            - Skip version parsing: false
            - Targets: {ESP32C3, ESP32H2, ESP32C6, ESP32S2, ESP32P4, ESP32, ESP32C2, ESP32S3}
            - Toolchain path: "/Users/lemon/.rustup/toolchains/esp"
            - Toolchain version: None
[info]: Checking Rust installation
[info]: Installing Xtensa Rust 1.84.0.0 toolchain
[info]: Installing GCC (xtensa-esp-elf)
[debug]: GCC path: /Users/lemon/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906
[debug]: Creating directory: '/Users/lemon/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906'
[info]: Installing Xtensa LLVM
[debug]: Creating directory: '/Users/lemon/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[debug]: starting new connection: https://github.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://github.com/'
[debug]: starting new connection: https://github.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://github.com/'
[debug]: starting new connection: https://github.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://github.com/'
[debug]: redirecting 'https://github.com/espressif/llvm-project/releases/download/esp-18.1.2_20240912/libs-clang-esp-18.1.2_20240912-aarch64-apple-darwin.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/234050162/5e916b0d-5974-4261-aeb1-4148ad9eabcb?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250209%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250209T040142Z&X-Amz-Expires=300&X-Amz-Signature=08f12f7f4e3e24994ef137faf28ca0cc56fe35bc7fd8d0191c425f2ec56d5945&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Dlibs-clang-esp-18.1.2_20240912-aarch64-apple-darwin.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://objects.githubusercontent.com/'
[debug]: redirecting 'https://github.com/espressif/crosstool-NG/releases/download/esp-14.2.0_20240906/xtensa-esp-elf-14.2.0_20240906-aarch64-apple-darwin.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/66057540/1549c302-f520-4e75-b7ce-1fcee5ba5cd4?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250209%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250209T040142Z&X-Amz-Expires=300&X-Amz-Signature=3d0221ded125377d6c35b39ae604e71aa301c9bebc8efc16b87dc121f1e536df&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Dxtensa-esp-elf-14.2.0_20240906-aarch64-apple-darwin.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://objects.githubusercontent.com/'
[debug]: redirecting 'https://github.com/esp-rs/rust-build/releases/download/v1.84.0.0/rust-src-1.84.0.0.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/404564617/15b0cd57-a21b-4d85-a9f1-4ebca64861be?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250209%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250209T040142Z&X-Amz-Expires=300&X-Amz-Signature=42f60abe4ba437c98fa15e0c041b24ef4e2bfa0c4016bf05ec457b5030aa5a73&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Drust-src-1.84.0.0.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://objects.githubusercontent.com/'
[warn]: Installation for 'RISC-V Rust target' failed, retrying. Error: Failed to Install RISC-V targets for 'nightly' toolchain
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying. Error: Failed to Install RISC-V targets for 'nightly' toolchain
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying. Error: Failed to Install RISC-V targets for 'nightly' toolchain
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying. Error: Failed to Install RISC-V targets for 'nightly' toolchain
Error: espup::toolchain::rust::install_riscv_target

  × Failed to Install RISC-V targets for 'nightly' toolchain

- OS: [macOS Sequoia 15.3]
- espup version: [0.14.0]

SergioGasquez, 2025-02-10:
Do you have some proxy?
Here is what I see when installing:
```
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[debug]: starting new connection: https://github.com/
[debug]: starting new connection: https://github.com/
[debug]: starting new connection: https://github.com/
```
While you:
```
[debug]: starting new connection: https://github.com/
[debug]: proxy(http://127.0.0.1:7890/) intercepts 'https://github.com/'
....
```

Also, can you try running: `espup install -t esp32,esp32s2, esp32s3`?

limooooooon, 2025-02-11:
"espup install -t esp32,esp32s2, esp32s3" worked!!! # thank you very much!!! @SergioGasquez

closed 2025-02-11 by SergioGasquez