esp-rs/espup #475: Invalid GitHub token but I don't have one

liolok, 2025-02-09:
[image]

`cargo install espup` results the same.

SergioGasquez, 2025-02-10:
> `cargo install espup` results the same.

Mind elaborating this sentence? Have you uninstalled the `espup` installed by pacman? Id reccomend the following:
1. Uninstall the espup installed by pacman
2. Install it via cargo: `cargo install espup`
3. Just to make sure, unset GITHUB_TOKEN: `unset GITHUB_TOKEN`
4. `espup install`

liolok, 2025-02-10:
Two installation methods result the same.

SergioGasquez, 2025-02-10:
Can you try the following and paste the result:
```
$ pacman -R espup
$ cargo install espup
$ unset GITHUB_TOKEN
$ espup install -l debug
```

liolok, 2025-02-10:
I couldn't reproduce now, it just worked after a few hours. (espup installed by  pacman)

liolok, 2025-02-10:
Maybe a network (like my ISP) caused issue?

SergioGasquez, 2025-02-10:
Mmmm not sure, but I guess it was something in the isp or environment. Glad is now working!

closed 2025-02-10 by SergioGasquez

wrsturgeon, 2025-04-15:
Running into this same issue, for what it's worth. I also don't have `GITHUB_TOKEN` set.

From what I can tell, the error can be produced in two locations: [this line](https://github.com/esp-rs/espup/blob/edc9d641341f6980e85e456f8ea014fee95b8c07/src/toolchain/mod.rs#L397C35-L397C53) and [this line](https://github.com/esp-rs/espup/blob/edc9d641341f6980e85e456f8ea014fee95b8c07/src/toolchain/mod.rs#L243). The former makes perfect sense, but the latter has nothing to do with tokens: instead, anytime `XtensaRust::get_latest_version()` fails, it throws away the real error and returns `GithubTokenInvalid` instead. Let me see if I can track down what might be happening...

wrsturgeon, 2025-04-15:
here's the error with `RUST_LOG=debug`:
```text
  [debug]: connecting to crates.io:443 at 3.163.125.74:443
  [debug]: No cached session for DnsName("crates.io")
  [debug]: Not resuming any session
  [debug]: Using ciphersuite TLS13_AES_128_GCM_SHA256
  [debug]: Not resuming
  [debug]: TLS1.3 encrypted extensions: [ServerNameAck]
  [debug]: ALPN protocol is None
  [debug]: created stream: Stream(RustlsStream)
  [debug]: sending request GET https://crates.io/api/v1/crates/espup/versions
  [debug]: writing prelude: GET /api/v1/crates/espup/versions HTTP/1.1
  Host: crates.io
  User-Agent: ureq/2.12.1
  Accept: */*
  accept-encoding: gzip
  [debug]: Chunked body in response
  [debug]: response 200 to GET https://crates.io/api/v1/crates/espup/versions
  [debug]: dropping stream: Stream(RustlsStream)
  [info]: Installing the Espressif Rust ecosystem
  [debug]: Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases/latest'
  [debug]: starting new connection: https://api.github.com/
  [debug]: starting new connection: https://api.github.com/
  [debug]: starting new connection: https://api.github.com/
  [debug]: starting new connection: https://api.github.com/
  [debug]: starting new connection: https://api.github.com/
  [debug]: starting new connection: https://api.github.com/
  Error: espup::toolchain::rust::query_github

    × Failed to query GitHub API: Invalid Github token

```

There's a `debug!` statement [here](https://github.com/esp-rs/espup/blob/edc9d641341f6980e85e456f8ea014fee95b8c07/src/toolchain/mod.rs#L266) that isn't reached, so I'm pretty sure the error originates right after `XtensaRust::get_latest_version()` [here](https://github.com/esp-rs/espup/blob/edc9d641341f6980e85e456f8ea014fee95b8c07/src/toolchain/mod.rs#L243).

There's also a `debug!` conditional on `GITHUB_TOKEN` being set ([here](https://github.com/esp-rs/espup/blob/edc9d641341f6980e85e456f8ea014fee95b8c07/src/toolchain/mod.rs#L377)) which also isn't reached. So something's going wrong in `XtensaRust::get_latest_version()`, then the error is being thrown away.

Fwiw, directly `curl`ing `XTENSA_RUST_LATEST_API_URL` (=https://api.github.com/repos/esp-rs/rust-build/releases/latest) works.

wrsturgeon, 2025-04-15:
Adding a valid `GITHUB_TOKEN` also does not work. :(

Hopefully this will just work in a bit, like it did for OP . . .