esp-rs/espup #476: Install To Platform-Standard Locations | Reduce Home Directory Clutter

taep96, 2025-02-19:
## Motivations
Files are currently placed directly in `$HOME`, which creates clutter. Using standard platform directories would improve system integration.

- Would you like to implement this feature? — Not sure.

## Solution
1. Use [`directories`](https://lib.rs/crates/directories) crate with the following fallback order:
   1. `$XDG_DATA_HOME/espup`
   2. `$HOME/.local/share/espup`
   3. `$HOME/.espup`

2. Add flag to optionally symlink export-esp.sh from `$HOME` to `$XDG_DATA_HOME/espup/export-esp.sh`

3. Create an extension-less version of the script with a shebang and export to `$PATH` - the `$HOME` symlink could point to this instead

## Alternatives
- Add an environment variable to customize the location — `$ESP_HOME`

## Additional context
The [`directories`](https://lib.rs/crates/directories) crate provides cross-platform support for standard directory locations.

SergioGasquez, 2025-02-21:
> Files are currently placed directly in $HOME??

What files do you mean, we only generate the export file in $HOME, and this can be modified with the `-f, --export-file` argument. 

The Xtensa Rust toolchain, uses the [`RUSTUP_HOME` environment variable, as rusutp does, ](https://rust-lang.github.io/rustup/environment-variables.html#environment-variables)to set where the toolchain is installed

taep96, 2025-02-21:
The `.espup` directory

[image]

SergioGasquez, 2025-02-24:
The `.espup` is just a symlink, see https://github.com/esp-rs/espup/pull/380. If we have different options of where it could leave, we will have to modify how `esp-idf-sys` behaves.

taep96, 2025-02-24:
Can the same cascade be implemented there?

SergioGasquez, 2025-02-28:
Yes, we could apply the same priority. That being said, implementing this won't be a priority for us, we would accept any contribution implementing this.

taep96, 2025-03-07:
Made a PR: https://github.com/esp-rs/espup/pull/489

taep96, 2025-03-19:
https://github.com/esp-rs/esp-idf-sys/pull/373