esp-rs/espup #482: espup fails toolchain installation

tommasoclini, 2025-03-04:
## Bug description

Today I tried installing the rust toolchain for esp, this is what I got instead:
```
espup install --toolchain-version 1.85.0.0 --log-level debug
[debug]: connecting to crates.io:443 at 3.165.255.21:443
[debug]: No cached session for DnsName("crates.io")
[debug]: Not resuming any session
[debug]: Using ciphersuite TLS13_AES_128_GCM_SHA256
[debug]: Not resuming
[debug]: TLS1.3 encrypted extensions: [ServerNameAck]
[debug]: ALPN protocol is None
[debug]: created stream: Stream(RustlsStream)
[debug]: sending request GET https://crates.io/api/v1/crates/espup/versions
[debug]: writing prelude: GET /api/v1/crates/espup/versions HTTP/1.1
Host: crates.io
User-Agent: ureq/2.12.1
Accept: */*
accept-encoding: gzip
[debug]: Chunked body in response
[debug]: response 200 to GET https://crates.io/api/v1/crates/espup/versions
[debug]: dropping stream: Stream(RustlsStream)
[info]: Installing the Espressif Rust ecosystem
[debug]: Parsing Xtensa Rust version: 1.85.0.0
[debug]: Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases?page=1&per_page=100'
[debug]: starting new connection: https://api.github.com/
[debug]: Arguments:
            - Export file: "/home/tommaso/export-esp.sh"
            - Host triple: x86_64-unknown-linux-gnu
            - LLVM Toolchain: Llvm { extended: false, file_name_libs: Some("libs-clang-esp-18.1.2_20240912-x86_64-linux-gnu.tar.xz"), file_name_full: None, host_triple: X86_64UnknownLinuxGnu, path: "/home/tommaso/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912", repository_url: "https://github.com/espressif/llvm-project/releases/download/esp-18.1.2_20240912", version: "esp-18.1.2_20240912" }
            - Nightly version: "nightly"
            - Rust Toolchain: Some(XtensaRust { cargo_home: "/home/tommaso/.cargo", dist_file: "rust-1.85.0.0-x86_64-unknown-linux-gnu.tar.xz", dist_url: "https://github.com/esp-rs/rust-build/releases/download/v1.85.0.0/rust-1.85.0.0-x86_64-unknown-linux-gnu.tar.xz", host_triple: "x86_64-unknown-linux-gnu", path: "/home/tommaso/.rustup/toolchains/esp", rustup_home: "/home/tommaso/.rustup", src_dist_file: "rust-src-1.85.0.0.tar.xz", src_dist_url: "https://github.com/esp-rs/rust-build/releases/download/v1.85.0.0/rust-src-1.85.0.0.tar.xz", toolchain_destination: "/home/tommaso/.rustup/toolchains/esp", version: "1.85.0.0" })
            - Skip version parsing: false
            - Targets: {ESP32C6, ESP32H2, ESP32P4, ESP32, ESP32S3, ESP32C3, ESP32C2, ESP32S2}
            - Toolchain path: "/home/tommaso/.rustup/toolchains/esp"
            - Toolchain version: Some("1.85.0.0")
[info]: Checking Rust installation
[info]: Installing Xtensa Rust 1.85.0.0 toolchain
[info]: Installing Xtensa LLVM
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[info]: Installing GCC (xtensa-esp-elf)
[debug]: GCC path: /home/tommaso/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906
[debug]: Creating directory: '/home/tommaso/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-18.1.2_20240912'
[debug]: Creating directory: '/home/tommaso/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906'
[warn]: Installation for 'RISC-V Rust target' failed, retrying. Error: Failed to Install RISC-V targets for 'nightly' toolchain
[debug]: starting new connection: https://github.com/
[debug]: starting new connection: https://github.com/
[debug]: starting new connection: https://github.com/
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying. Error: Failed to Install RISC-V targets for 'nightly' toolchain
[debug]: redirecting 'https://github.com/esp-rs/rust-build/releases/download/v1.85.0.0/rust-src-1.85.0.0.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/404564617/26627945-1be6-4f1d-8c5e-10d641d1c1f0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250304%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250304T120700Z&X-Amz-Expires=300&X-Amz-Signature=a3789659887a2823b65d33e8dd581131dd37192c14b16142065749209f28afa1&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Drust-src-1.85.0.0.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying. Error: Failed to Install RISC-V targets for 'nightly' toolchain
[debug]: redirecting 'https://github.com/espressif/llvm-project/releases/download/esp-18.1.2_20240912/libs-clang-esp-18.1.2_20240912-x86_64-linux-gnu.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/234050162/40be3d55-3ad8-42b8-8c17-5e92527ef3bf?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250304%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250304T120700Z&X-Amz-Expires=300&X-Amz-Signature=0a791ff037b865da43e8badbbf51065e8c3cffdf90413b34187ad8ea471668bc&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Dlibs-clang-esp-18.1.2_20240912-x86_64-linux-gnu.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[debug]: redirecting 'https://github.com/espressif/crosstool-NG/releases/download/esp-14.2.0_20240906/xtensa-esp-elf-14.2.0_20240906-x86_64-linux-gnu.tar.xz' to 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/66057540/6fd45494-6a43-4bdb-a982-640ed774f02a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250304%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250304T120700Z&X-Amz-Expires=300&X-Amz-Signature=9274f417912a615ecebffc5b6272ba2581b5af9596f0b3be235da4fae7d440ce&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Dxtensa-esp-elf-14.2.0_20240906-x86_64-linux-gnu.tar.xz&response-content-type=application%2Foctet-stream'
[debug]: starting new connection: https://objects.githubusercontent.com/
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'nightly' toolchain
[warn]: Installation for 'RISC-V Rust target' failed, retrying. Error: Failed to Install RISC-V targets for 'nightly' toolchain
Error: espup::toolchain::rust::install_riscv_target

  × Failed to Install RISC-V targets for 'nightly' toolchain
```

- Would you like to work on a fix? n

## To Reproduce

Steps to reproduce the behavior:

`espup install`

The install fails both with `--toolchain-version 1.85.0.0` and without any arguments.

## Expected behavior

Installation of esp rust toolchain.

## Environment

- OS: Debian GNU/Linux bookworm 12.9 x86_64
- espup version: 0.14.0

## Additional context

I've just updated rustup to 1.28 from 1.27.1.

tommasoclini, 2025-03-04:
I just saw the fix, should I keep the issue open?

tommasoclini, 2025-03-04:
I think I've encountered another problem, I don't know though, this is the output when building one of my projects (after deleting `Cargo.lock` and `cargo clean`):
`Locking 187 packages to latest Rust 1.85.0-nightly compatible versions`
Is this supposed to happen?

SergioGasquez, 2025-03-04:
> I just saw the fix, should I keep the issue open?

No, Ill close it as the fix is already merged (and Ill create a release for it tomorrow)

> I think I've encountered another problem

Mind giving more info about the issue? And how to reproduce. It doesnt look to be related to `espup` though

closed 2025-03-04 by SergioGasquez

SergioGasquez, 2025-03-04:
`espup@0.14.1` was just released! Use `cargo install espup` to update