esp-rs/espup #495: Misleading error: "Invalid GitHub token" whenever query fails

wrsturgeon, 2025-04-15:
Initially, this started out as the same issue as `https://github.com/esp-rs/espup/issues/475`. I've left a few comments there, but I didn't realize it was already closed, so let me summarize here:

[This line](https://github.com/esp-rs/espup/blob/edc9d641341f6980e85e456f8ea014fee95b8c07/src/toolchain/mod.rs#L243) seems to delete errors received from `XtensaRust::get_latest_version()` and return `Error::GithubTokenInvalid` instead (a bit misleading, since `GithubTokenInvalid` is also returned later in the file for an explicit authorization error). Because the original errors are suppressed, it's difficult for me to give you exactly what's going wrong, but I believe I've tracked it down to the function `github_query` by observing which `log::debug!` statements are printed before the program exits.

So, I guess this error is two-fold:
1. Is there a way to uncover the real error and work around it?
2. Is it possible to provide a more informative error message on the above line? I'd be willing to help out as well.

Thanks for anything that might help!

Best,
Will

SergioGasquez, 2025-04-17:
Hi! Thanks for opening the issue and the details, mind trying https://github.com/SergioGasquez/espup/tree/fix/latest-version?

wrsturgeon, 2025-04-17:
Thanks for writing that up so quickly!

OK, interesting developments. I've been using Nix to manage Rust & Cargo, so I have `rustc` and `cargo` in `PATH`, but not `rustup`. Because of that, the first error I got was "Rust is not installed" ([`espup::toolchain::rust::missing_rust`](https://github.com/esp-rs/espup/blob/edc9d641341f6980e85e456f8ea014fee95b8c07/src/error.rs#L40)), which is raised exclusively by [`check_rust_installation`](https://github.com/esp-rs/espup/blob/edc9d641341f6980e85e456f8ea014fee95b8c07/src/toolchain/rust.rs#L430), which in turn is called exclusively by `install` [here](https://github.com/esp-rs/espup/blob/edc9d641341f6980e85e456f8ea014fee95b8c07/src/toolchain/mod.rs#L288)--but it's called _after_ a debug message that didn't print before. Before fully installing, let me go back to the original `espup` and see if I get the same error, or if some networking error cleared itself up.

Thanks again for your help, and be back in a sec!

wrsturgeon, 2025-04-17:
OK, so downloading upstream `espup` (`main`) and running with `RUST_LOG=debug cargo run -- install` works. I'm guessing that the `cargo-binstall` version is behind `main`, and something else fixed it. I'd be willing to close--if I install `rustup`, I bet the installation will go through.

wrsturgeon, 2025-04-17:
Yes, installing with `main` works. I'm guessing this is a `cargo-binstall` problem.

closed 2025-04-22 by SergioGasquez