esp-rs/espup #502: Can't install espup by cargo

chenxuuu, 2025-05-16:
## Bug description

```
> cargo install espup
    Updating crates.io index
  Installing espup v0.15.0
    Updating crates.io index
error: failed to compile `espup v0.15.0`, intermediate artifacts can be found at `C:\Users\chenxu\AppData\Local\Temp\cargo-installc4tx7s`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.

Caused by:
  failed to select a version for the requirement `zip = "^2.6.1"`
    version 2.6.1 is yanked
  location searched: crates.io index
  required by package `espup v0.15.0`
```

- Would you like to work on a fix? [y/n]

seems there is a pull request wait for marge: https://github.com/esp-rs/espup/pull/501

## To Reproduce

Steps to reproduce the behavior:

just run `cargo install espup`

## Expected behavior

espup successfuly installed

## Environment

- OS: windows 11 23H2
- espup version: none

## Additional context

is there anyway to install espup before new version release?

chenxuuu, 2025-05-16:
solved by adding `--locked` in cargo

closed 2025-05-16 by chenxuuu

SergioGasquez, 2025-05-16:
This was related to #503 #504