esp-rs/espup #503: esp-up 0.15.0 fails to install on Fedora 42 due to yanked dependency - zip version v2.6.1

snarkyboojum, 2025-05-16:
espup fails to install on Fedora 42 due to yanked zip crate version v2.6.1 (which is current failing CI on crates.io).

`$ cargo install espup
    Updating crates.io index
  Downloaded espup v0.15.0
  Downloaded 1 crate (52.9 KB) in 0.72s
  Installing espup v0.15.0
    Updating crates.io index
error: failed to compile espup v0.15.0, intermediate artifacts can be found at /tmp/cargo-installGqBoO7.
To reuse those artifacts with a future compilation, set the environment variable CARGO_TARGET_DIR to that path.

Caused by:
  failed to select a version for the requirement zip = "^2.6.1"
    version 2.6.1 is yanked
  location searched: crates.io index
  required by package espup v0.15.0`

Patching Cargo.toml in release tag v0.15.0 to use zip v3.0.0 and installing locally from there appears to work just fine.

`diff --git a/Cargo.toml b/Cargo.toml
index a2d3395..2fd8672 100644
--- a/Cargo.toml
+++ b/Cargo.toml
@@ -40,7 +40,7 @@ tokio-retry          = "0.3.0"
 tokio-stream         = "0.1.17"
 update-informer      = "1.2.0"
 xz2                  = "0.1.7"
-zip                  = "2.6.1"
+zip                  = "3.0.0"
 
 [target.'cfg(unix)'.dependencies]
 openssl = { version = "0.10.72", features = ["vendored"] }
`

I'm not sure if this is a transient error with that version of zip failing CI on crates.io, but opening this issue to flag that espup currently fails with this zip version.

brendanzab, 2025-05-16:
Seems like `zip` 2.6.0 and 2.6.1 were yanked as a result of a breaking semver change: https://github.com/zip-rs/zip2/issues/337

I’d imagine downgrading to 2.5 or using 3.0.0 could fix this issue.

brendanzab, 2025-05-16:
Seems like there's a PR to fix this here: https://github.com/esp-rs/espup/pull/501

closed 2025-05-16 by SergioGasquez