esp-rs/espup #505: Failed to query GitHub API: Invalid Github token

liucan233, 2025-05-16:
## Bug description

same [this issue](https://github.com/esp-rs/espup/issues/495)

debug output:

> [info]: Installing the Espressif Rust ecosystem
[debug]: Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases/latest'
[debug]: starting new connection: https://api.github.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://api.github.com/'
[debug]: starting new connection: https://api.github.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://api.github.com/'
[debug]: starting new connection: https://api.github.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://api.github.com/'
[debug]: starting new connection: https://api.github.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://api.github.com/'
[debug]: starting new connection: https://api.github.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://api.github.com/'
[debug]: starting new connection: https://api.github.com/
[debug]: proxy(http://127.0.0.1:7890) intercepts 'https://api.github.com/'
Error: espup::toolchain::rust::query_github

- Would you like to work on a fix?
 yes

## To Reproduce

Steps to reproduce the behavior:

1. cargo install espup@0.14.1
2. espup install

## Expected behavior

espup install success

## Screenshots

[image]

## Environment

- OS: Mac OS 13.7.4 (22H420)
- espup version: 0.14.1

SergioGasquez, 2025-05-21:
Have you tried using the latest version of `espup`?? `cargo install espup`

SergioGasquez, 2025-05-26:
> I was getting rate-limited from GitHub due unauthenticated requests.

Are you using the GH api for any other thing? A regular `espup install/update` should do about ~10 gh API calls.  If you dont use a GH Token, you should get 60 requests per hour, which should be enough for several install/udpate commands.

liucan233, 2025-06-06:
I think espup to install maybe out of date,I used below method:

cargo install esp-generate
esp-generate --chip=esp32c6 your-project

more see docs: https://docs.espressif.com/projects/rust/esp-hal/1.0.0-beta.1/esp32s3/esp_hal/index.html

SergioGasquez, 2025-06-06:
> I think espup to install maybe out of date,I used below method

`espup` is not related to `esp-generate`:
- `espup` is still required for installing the Rust toolchain for Xtensa targets (ESP32, ESP32-S2 and ESP32-S3)
- `esp-generate` is a tool for generating a starting project

In your case, you are using `esp-generate` to create a project for a C6 which can be used with stable Rust, hence `espup` is not required. See more at https://docs.esp-rs.org/book/installation/index.html

liucan233, 2025-06-06:
@SergioGasquez You are right, my device is ESP32S3, and I actually used espup v0.14 and successfully installed it. Then I encountered an error during cargo build, and then, I used ESP generate to recreate the project, which finally ran successfully