esp-rs/espup #509: `cargo install espup` compilation failure

Erhannis, 2025-07-12:
## Bug description

`cargo install espup` ends in compilation failure, rather than successfully install espup.

- Would you like to work on a fix? y, maybe?  if you can tell me what needs fixed?

## To Reproduce

Steps to reproduce the behavior:

1. rustup upgrade
2. cargo install espup
3. get fail message (see log below)

## Expected behavior

`cargo install espup` should successfully install espup.

## Screenshots

Console log:
```
erhannis@mnode7:~/projects/ESP_projects/rust$ cargo install espup
    Updating crates.io index
  Installing espup v0.15.1
    Updating crates.io index
     Locking 327 packages to latest compatible versions
      Adding indicatif v0.17.11 (available: v0.18.0)
      Adding zip v3.0.0 (available: v4.3.0)
   Compiling libc v0.2.174
   Compiling proc-macro2 v1.0.95
   Compiling unicode-ident v1.0.18
   Compiling cfg-if v1.0.1
   Compiling shlex v1.3.0
   Compiling pkg-config v0.3.32
   Compiling memchr v2.7.5
   Compiling once_cell v1.21.3
   Compiling smallvec v1.15.1
   Compiling bytes v1.10.1
   Compiling autocfg v1.5.0
   Compiling pin-project-lite v0.2.16
   Compiling parking_lot_core v0.9.11
   Compiling log v0.4.27
   Compiling scopeguard v1.2.0
   Compiling stable_deref_trait v1.2.0
   Compiling itoa v1.0.15
   Compiling futures-core v0.3.31
   Compiling version_check v0.9.5
   Compiling typenum v1.18.0
   Compiling fnv v1.0.7
   Compiling bitflags v2.9.1
   Compiling vcpkg v0.2.15
   Compiling futures-sink v0.3.31
   Compiling zerocopy v0.8.26
   Compiling httparse v1.10.1
   Compiling percent-encoding v2.3.1
   Compiling litemap v0.8.0
   Compiling adler2 v2.0.1
   Compiling subtle v2.6.1
   Compiling writeable v0.6.1
   Compiling getrandom v0.3.3
   Compiling slab v0.4.10
   Compiling futures-io v0.3.31
   Compiling equivalent v1.0.2
   Compiling pin-utils v0.1.0
   Compiling icu_properties_data v2.0.1
   Compiling icu_normalizer_data v2.0.0
   Compiling serde v1.0.219
   Compiling unicode-width v0.2.1
   Compiling hashbrown v0.15.4
[... 233 lines cut ...]
42 |     pub static ref PROCESS_BARS: indicatif::MultiProgress = indicatif::MultiProgress::new();
   |                                  --------- one version of crate `indicatif` used here, as a direct dependency of the current crate
   = help: you can use `cargo tree` to explore your dependency tree
note: associated function defined here
  --> /home/erhannis/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/indicatif-log-bridge-0.2.3/src/lib.rs:67:12
   |
67 |     pub fn new(bar: MultiProgress, log: L) -> Self {
   |            ^^^

For more information about this error, try `rustc --explain E0308`.
error: could not compile `espup` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
error: failed to compile `espup v0.15.1`, intermediate artifacts can be found at `/tmp/cargo-installgCMCQj`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
```

## Environment

- OS: Ubuntu 25.04
- espup version: attempting to install 0.15.1
- cargo version: cargo 1.88.0 (873a06493 2025-05-10)

## Additional context

I don't know how I'm the only person seeing this error, but I'm seeing it.

Erhannis, 2025-07-12:
I tried on a different computer, same error.

Ubuntu 22.04.5 LTS
cargo 1.88.0 (873a06493 2025-05-10)

Looks like indicatif updated and it's not uniformly taken effect?

SergioGasquez, 2025-07-12:
Can you try using `cargo install espup --locked`?

MihaMarkic, 2025-07-13:
@djugei Could it be that some crate references that cargo version 0.18?

lenntt, 2025-07-17:
I'm on macos and ran into the same issue, created [esp-rs/book#206](https://github.com/esp-rs/book/pull/206)

Borg001, 2025-07-27:
I faced the same issue under Windows.

pacak, 2025-07-28:
`espup` depends on `indicatif` as well as `indicatif-log-bridge` - see `Cargo.toml`

```toml
indicatif            = "0.17.11"
indicatif-log-bridge = "0.2.3"
```

`indicatif-log-bridge` declares it's dependency "as anything above 0.17.5" - a very bold claim :)

```toml
indicatif = ">=0.17.5"
```

`Cargo.lock` was created before `indicatif-0.18` release, so with the `--locked` it works just fine. Without it - `cargo` starts resolving dependencies giving `i-l-b` newest supported version. Problem with that is that `indicatif` version leaks into the API and wants to have whatever instances of the right version. `espup` provides instances from `0.17` - `rustc` considers them incompatible.